cezanne/usbip-win #6: windows 10 as USBIP client

dhyanisumit12, 2018-12-19:
I am configuring my windows 10 as client using this repository. And Raspberry Pi 3 ( Linux raspberrypi 4.4.50-v7+ ) as server using steps given in [link](https://3mdeb.com/os-dev/linux-rpi-and-usb-over-ip-updated/).
USBIP version of server is 2.0
I am able to list the usb drive shared by server on windows. But unable to connect to it. 
In device manager it is showing an unknown device named 'USB Device over IP'.
[image]

Is this ab bug of this repository as you mentioned in last line of README that it did not work?

cezuni, 2018-12-19:
> Is this ab bug of this repository as you mentioned in last line of README that it did not work?
Yes..

I had focused to windows usbip server and something in vhci driver went wrong. 
I'll check out until this year ends.

dhyanisumit12, 2018-12-19:
Are there compatibility issues with USBIP between linux and windows version. As in linux its showing version : usbip (usbip-utils 2.0)
But the library you are providing is showing : 1.0.0

cezuni, 2018-12-19:
#3 may answer your question. Protocol negotiation will fail between different versions.
But usbip-utils 2.0 you mentioned seems to be an only autoconf  package version.
It's not related to usbip protocol.
This repo has 0x111 USBIP_VERSION, which is a up-to-date version. (Kernel 4.20.x has the same USBIP version)

yelfarri, 2019-01-03:
Thank you for your answer. I understand that more testing is needed but I m trying to help in any way possible. I can test some devices if you don't mind :). So from my understanding your Bluetooth dongle in 
 working windows? 
I have tested your code on different mice and keyboards I have the same problem as described in the one here #6. Seems that issue with the Vhci_pdo.c and/or vhci_PnP.c . I have done some attempt find and fix the issue but without any luck.  

I used Kernel debugging to get access log output from VHCI it seems that the  BUS_PNP =  STATUS_NOT_SUPPORTED in four cases. Are they cause for the issue ?

> 00005098	4676.33251953	usbip_vhci: PDO: minor: IRP_MN_QUERY_INTERFACE, IRP: 0xFFFFD68AE85A8980
> 00005099	4676.33251953	usbip_vhci: Query GUID: EC92E9B897A7C44D884684D041707446
> 00005100	4676.33251953	usbip_vhci: Query unknown interface GUID:
> 00005101	4676.33251953	usbip_vhci: Bus_PnP: Leave: STATUS_NOT_SUPPORTED
> 
> 00005123	4676.33349609	usbip_vhci: PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFD68AE85A8980
> 00005124	4676.33349609	usbip_vhci: QueryId Type: BusQueryContainerID
> 00005125	4676.33349609	usbip_vhci: Bus_PnP: Leave: STATUS_NOT_SUPPORTED
> 
> 00005127	4676.33349609	usbip_vhci: PDO: minor: IRP_MN_QUERY_INTERFACE, IRP: 0xFFFFD68AE85A8980
> 00005128	4676.33349609	usbip_vhci: Query GUID: 0E1B2170FB0ADB47AFC1410BF842497A
> 00005129	4676.33349609	usbip_vhci: Query unknown interface GUID:
> 00005130	4676.33349609	usbip_vhci: Bus_PnP: Leave: STATUS_NOT_SUPPORTED
> 
> 00005142	4676.33398438	usbip_vhci: Bus_PnP: Enter
> 00005143	4676.33398438	usbip_vhci: PDO: minor: IRP_MN_QUERY_RESOURCES, IRP: 0xFFFFD68AE85A8980
> 00005144	4676.33398438	usbip_vhci: Bus_PnP: Leave: STATUS_NOT_SUPPORTED
> 00005145	4676.33447266	usbip_vhci: Bus_PnP: Enter

cezuni, 2019-01-03:
I don't mind but your help makes me be hasty. ^^;; Many thanks to you. 

Serious error or warning logs have (E) or (W) prefix.
Did your logs have no such prefix?

If up-to-date vhci codes still have "USB Device Over IP" in device manager,
your mentioned STATUS_NOT_SUPPORTED may be the cause for this issue.

yelfarri, 2019-01-04:
Thank you i m happy that your working on this! :)  

No there is no serious error or warning with prefix (E) or (W) in my log.  The only issue is that the "USB Device Over IP" is classified by windows under other devices.  The status of this device drivers is " device are not installed"(see 1st picture in this post). When checking a bit closer in events tab (Usb Device Over IP Properties) we see that the device has not aquired any configurations. That confirms my suspension that that  vhci_pdo.c does not initialize the new device correctly. Based on the log i trace the message back to the following  functions in code
 

The status STATUS_NOT_SUPPORTED can be tracked back to the file vhci_pdo.c spesificly line 947 and 
``` c
//Line 1255 in vhci_pdo.c
case IRP_MN_QUERY_INTERFACE:
		//
		// This request enables a driver to export a direct-call
		// interface to other drivers. A bus driver that exports
		// an interface must handle this request for its child
		// devices (child PDOs).
		//
		status = Bus_PDO_QueryInterface(DeviceData, Irp);
		break;

```
```c
//Line 1194 in vhci_pdo.c
case IRP_MN_QUERY_ID:
		DBGI(DBG_PNP, "QueryId Type: %s\n", dbg_bus_query_id_type(IrpStack->Parameters.QueryId.IdType));

		status = Bus_PDO_QueryDeviceId(DeviceData, Irp);
		break;

```
```c
//Line 1209 in vhci_pdo.c
case IRP_MN_QUERY_RESOURCES:

		status = Bus_PDO_QueryResources(DeviceData, Irp);

		break;

```

As mentioned in the earlier post here i still see the same
[image]

 When i right click to check the events of the USB Device over IP i see  the following. There is no configuration for the devices.

[image]

wrenchonline, 2019-01-05:
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA8003A52C70
usbip_vhci:(QQ)QueryId Type: BusQueryContainerID
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_RESOURCE_REQUIREMENTS, IRP: 0xFFFFFA8003A52C70
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_BUS_INFORMATION, IRP: 0xFFFFFA8003A52C70
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_RESOURCES, IRP: 0xFFFFFA8003A52C70
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_DEVICE_ENUMERATED, IRP: 0xFFFFFA8003A52C70
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_Ioctl: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_Read: Enter
usbip_vhci:(QQ)Bus_Read: Leave: STATUS_PENDING
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)FDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP:0xFFFFFA8003AB0010
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_INTERFACE, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)Query GUID: 0E1B2170FB0ADB47AFC1410BF842497A
usbip_vhci:(QQ)Query unknown interface GUID:
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)QueryDeviceRelation Type: EjectionRelations
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)QueryDeviceRelation Type: RemovalRelations
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)QueryDeviceRelation Type: PowerRelations
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)QueryDeviceRelation Type: Unknown device relation code: 6
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)FDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP:0xFFFFFA8003AB0010
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_INTERFACE, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)Query GUID: 0E1B2170FB0ADB47AFC1410BF842497A
usbip_vhci:(QQ)Query unknown interface GUID:
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)QueryDeviceRelation Type: EjectionRelations
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)QueryDeviceRelation Type: RemovalRelations
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFFA8004380180
usbip_vhci:(QQ)QueryDeviceRelation Type: PowerRelations
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFFA8004380180
[... 96 lines cut ...]
usbip_vhci:(QQ)QueryDeviceRelation Type: PowerRelations
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFFA80042AE010
usbip_vhci:(QQ)QueryDeviceRelation Type: Unknown device relation code: 6
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP: 0xFFFFFA80042AE010
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)FDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP:0xFFFFFA8003AB0010
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP: 0xFFFFFA80042AE010
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)FDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP:0xFFFFFA8003AB0010
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP: 0xFFFFFA80042AE010
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)FDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP:0xFFFFFA8003AB0010
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
No there is no severe error or warning with a prefix (E) or (W) in my log.I change another one without prefix QQ

wrenchonline, 2019-01-05:
I found an error in Bus_PDO_QueryDeviceId and did not get the device ID

cezuni, 2019-01-05:
Below is my log when a bluetooth dongle in linux is successfully attached from win10. 
What's different?
```
usbip_vhci: ioctl code: IOCTL_USBIP_VHCI_PLUGIN_HARDWARE
usbip_vhci: Exposing PDO: addr: 1, vendor:product: 0b05:1715
usbip_vhci: fdodata->NextLowerDriver = 0xFFFFDE8D20590E40
usbip_vhci: pdo 0xFFFFDE8D22B31350, extension 0xFFFFDE8D22B314A0
usbip_vhci: Bus_PnP: Enter
usbip_vhci: FDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP:0xFFFFDE8D208B2E10
usbip_vhci:  QueryDeviceRelation Type: BusRelations
usbip_vhci: # of PDOS: present: 1, reported: 1
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFDE8D20D68880
usbip_vhci: QueryId Type: BusQueryDeviceID
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP: 0xFFFFDE8D20D68880
usbip_vhci: Bus_PnP: Enter
usbip_vhci: FDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP:0xFFFFDE8D208B2E10
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_DEVICE_TEXT, IRP: 0xFFFFDE8D20D68880
usbip_vhci:  DeviceTextDescription :USB Device Over IP
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_DEVICE_TEXT, IRP: 0xFFFFDE8D20D68880
usbip_vhci:  DeviceTextLocationInformation :on USB/IP VHCI
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFDE8D20D68880
usbip_vhci: QueryId Type: BusQueryInstanceID
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFDE8D20D68880
usbip_vhci: QueryId Type: BusQueryHardwareIDs
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFDE8D20D68880
usbip_vhci: QueryId Type: BusQueryCompatibleIDs
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFDE8D20D68880
usbip_vhci: QueryId Type: BusQueryContainerID
usbip_vhci: Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci: Bus_Ioctl: Leave: STATUS_SUCCESS
usbip_vhci: Bus_Read: Enter
usbip_vhci: Bus_Read: Leave: STATUS_PENDING
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_INTERFACE, IRP: 0xFFFFDE8D20D68880
usbip_vhci: Query GUID: 0E1B2170FB0ADB47AFC1410BF842497A
usbip_vhci: Query unknown interface GUID:
usbip_vhci: Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_BUS_INFORMATION, IRP: 0xFFFFDE8D20D68880
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_RESOURCE_REQUIREMENTS, IRP: 0xFFFFDE8D2477F010
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_RESOURCES, IRP: 0xFFFFDE8D2477F010
usbip_vhci: Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_DEVICE_ENUMERATED, IRP: 0xFFFFDE8D2477F010
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFDE8D22BC6CC0
usbip_vhci: QueryId Type: BusQueryDeviceID
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
[... 50 lines cut ...]
usbip_vhci: PDO: minor: IRP_MN_QUERY_INTERFACE, IRP: 0xFFFFDE8D22BC6CC0
usbip_vhci: Query GUID: 136AA9B1E03D74459B01C08FEAB318D6
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: IsDeviceHighSpeed called, it is 2
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_LEGACY_BUS_INFORMATION, IRP: 0xFFFFDE8D22BF1E10
usbip_vhci:(WW) not handled: IRP_MN_QUERY_LEGACY_BUS_INFORMATION
usbip_vhci: Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_QUERY_RESOURCE_REQUIREMENTS, IRP: 0xFFFFDE8D22BF1E10
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_FILTER_RESOURCE_REQUIREMENTS, IRP: 0xFFFFDE8D22BF1E10
usbip_vhci:(WW) not handled: IRP_MN_FILTER_RESOURCE_REQUIREMENTS
usbip_vhci: Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci: Bus_PnP: Enter
usbip_vhci: PDO: minor: IRP_MN_START_DEVICE, IRP: 0xFFFFDE8D22BF1E10
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: Bus_Internal_Ioctl: Enter
usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB
usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE
usbip_vhci: store_urbr: urbr: [seq:1]
usbip_vhci: store_urbr_submit: urbr: [seq:1]
usbip_vhci: process_urb_req_submit: urbr: [seq:1], func:URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE
```

wrenchonline, 2019-01-05:
usbip_vhci:(QQ)QueryDeviceRelation Type: Unknown device relation code: 6,Whether this affects the device manager display usb flash drive device information,I find this type:6 seems to affect Bus_PDO_QueryDeviceRelations return very much.

wrenchonline, 2019-01-06:
Hi,cezuni.After nearly two hours of debugging, I found that the two irps that output the most in the log are IRP_MN_QUERY_DEVICE_RELATIONS and IRP_MN_QUERY_CAPABILITIES, but I don't know what they are doing for this purpose, I just detect something strange (after harvesting the last irp IRP_MN_QUERY_CAPABILITIES).
Kd > dx-id 0,0,fffffa8003bceb30._DEVICE_POWER_STATE (*) [7]) 0 xfffff88004aa2980))
(* ((usbip_vhci!_DEVICE_POWER_STATE (*)[7])0xfffff88004aa2980)) [Type: _DEVICE_POWER_STATE [7]
[0] : PowerDeviceUnspecified (0) [Type: _DEVICE_POWER_STATE]
[1] : PowerDeviceD0 (1) [Type: _DEVICE_POWER_STATE]
[2] : PowerDeviceD3 (4) [Type: _DEVICE_POWER_STATE]
[3] : PowerDeviceD3 (4) [Type: _DEVICE_POWER_STATE]
[4] : PowerDeviceD3 (4) [Type: _DEVICE_POWER_STATE]
[5] : PowerDeviceD3 (4) [Type: _DEVICE_POWER_STATE]
[6] : PowerDeviceD3 (4) [Type: _DEVICE_POWER_STATE]
This is the state of the parent device
Kd > dx-id 0,0,fffffa8003bceb30._DEVICE_POWER_STATE (*) [7]) 0 xfffff88004aa2980))
(* ((usbip_vhci!_DEVICE_POWER_STATE (*)[7])0xfffff88004aa2980)) [Type: _DEVICE_POWER_STATE [7]
[0] : PowerDeviceUnspecified (0) [Type: _DEVICE_POWER_STATE]
[1] : PowerDeviceD0 (1) [Type: _DEVICE_POWER_STATE]
[2] : PowerDeviceD3 (2) [Type: _DEVICE_POWER_STATE]
[3] : PowerDeviceD3 (4) [Type: _DEVICE_POWER_STATE]
[4] : PowerDeviceD3 (4) [Type: _DEVICE_POWER_STATE]
[5] : PowerDeviceD3 (4) [Type: _DEVICE_POWER_STATE]
[6] : PowerDeviceD3 (4) [Type: _DEVICE_POWER_STATE]
I don't know if this affects the driver identification installation

cezuni, 2019-01-06:
@wrenchonline: I'm not sure but problems at IRP_MN_QUERY_DEVICE_RELATIONS or IRP_MN_QUERY_CAPABILITIES may not be related to this issue.
Can you share your logs or check if following lines are appeared?
```
usbip_vhci: PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFDE8D22BC6CC0
usbip_vhci: QueryId Type: BusQueryHardwareIDs
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
``` 
and
```
usbip_vhci: PDO: minor: IRP_MN_QUERY_INTERFACE, IRP: 0xFFFFDE8D22BC6CC0
usbip_vhci: Query GUID: 136AA9B1E03D74459B01C08FEAB318D6
usbip_vhci: Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci: IsDeviceHighSpeed called, it is 2
usbip_vhci: Bus_PnP: Enter
```

wrenchonline, 2019-01-06:
This is all of my log output and you can find it only QueryId Type: BusQueryHardwareIDs is successful and IsDeviceHighSpeed is not found related to USB_BUSIF_USBDI_VERSION_1 and USB_BUSIF_USBDI_VERSION_0 calls
`usbip_vhci:(QQ)Bus_Ioctl: Enter
usbip_vhci:(QQ)ioctl code: IOCTL_USBIP_VHCI_GET_PORTS_STATUS
usbip_vhci:(QQ)get ports status
usbip_vhci:(QQ)Bus_Ioctl: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_Ioctl: Enter
usbip_vhci:(QQ)ioctl code: IOCTL_USBIP_VHCI_PLUGIN_HARDWARE
usbip_vhci:(QQ)Exposing PDO: addr: 2, vendor:product: 9bb5:0001
usbip_vhci:(QQ)fdodata->NextLowerDriver = 0xFFFFFA8001920BB0
usbip_vhci:(QQ)pdo 0xFFFFFA8001A36BF0, extension 0xFFFFFA8001A36D40
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)FDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP:0xFFFFFA8001AD69D0
usbip_vhci:(QQ)	QueryDeviceRelation Type: BusRelations
usbip_vhci:(QQ)# of PDOS: present: 2, reported: 2
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)QueryId Type: BusQueryDeviceID
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)FDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP:0xFFFFFA8001AD69D0
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_TEXT, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)	DeviceTextDescription :USB Device Over IP
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_TEXT, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)	DeviceTextLocationInformation :on USB/IP VHCI
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)QueryId Type: BusQueryInstanceID
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)QueryId Type: BusQueryHardwareIDs
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)QueryId Type: BusQueryCompatibleIDs
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)QueryId Type: BusQueryContainerID
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_RESOURCE_REQUIREMENTS, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_BUS_INFORMATION, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_RESOURCES, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_DEVICE_ENUMERATED, IRP: 0xFFFFFA800403E670
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_Ioctl: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_Read: Enter
usbip_vhci:(QQ)Bus_Read: Leave: STATUS_PENDING
`

wrenchonline, 2019-01-07:
@cezuni  I tried guid tracing and found the Bus_PDO_QueryInterface code segment if (!IsEqualGUID(interfaceType, (PVOID) &USB_BUS_INTERFACE_USBDI_GUID)){} is always returned here, and my debug output always says interfaceType is
For the first time,
InterfaceType = 0 xfffff800 ` 04004598 {70211 b0e afb - 470db - AFC1-410 bf842497a}
The second time
InterfaceType = 0 xfffff800 ` 04004598 {70211 b0e afb - 470db - AFC1-410 bf842497a}
after
InterfaceType = 0 xfffff800 ` 04004598 {70211 b0e afb - 470db - AFC1-410 bf842497a}
Obviously not down here
The switch (version) {
Case USB_BUSIF_USBDI_VERSION_1:
...

cezuni, 2019-01-07:
With my usb stick, I have the same result with logs from @wrenchonline.
The reason why USB stick is not recognized is that 
PNP manager does not invoke IRP_MN_QUERY_INTERFACE with USB_BUS_INTERFACE_USBDI_GUID.
And such a no action of PNP manager may be due to wrong compatible IDs of USBIP device
which are all zeros.

yelfarri, 2019-01-07:
@wrenchonline  and @cezuni I think the issues start from the usbip.exe file. When debuging usbip i see that the issues start in the userspace. 

User ip>usbip.exe --debug attach -r <ip-address> -b 1-1
For the sake of testing where can i set a default value to check if that is the cause of the issues ? is udev struct used to pass the values to other functions ?

> usbip: debug: c:\..\usbip_vhci.c:57:[usbip_vhci_driver_open] device path: \\?\root#unknown#0000#{d35f7840-6a0c-11d2-b841-00c04fad5171}
> usbip: debug: c:\..\usbip\usbip_attach.c:55:[import_device] got free port 1
> usbip: debug: c:\..\usbip\usbip_vhci.c:121:[usbip_vhci_attach_device] plugin.devid: 65538
> 
> usbip: debug: c:\..\usbip\usbip_vhci.c:122:[usbip_vhci_attach_device] plugin.vendor: 2362
> 
> usbip: debug: c:\..\usbip\usbip_vhci.c:123:[usbip_vhci_attach_device] plugin.product: 9488
> 
> usbip: debug: c:\..\usbip_vhci.c:124:[usbip_vhci_attach_device] plugin.version: 256
> 
> usbip: debug: c:\..\usbip\usbip_vhci.c:125:[usbip_vhci_attach_device] plugin.speed: 
> 
> usbip: debug: c:\..\usbip\usbip_vhci.c:127:[usbip_vhci_attach_device] plugin.inum: 
> 
> usbip: debug: c:\..\usbip\usbip_vhci.c:128:[usbip_vhci_attach_device] plugin.class:
> 
> usbip: debug: c:\..\usbip\usbip_vhci.c:129:[usbip_vhci_attach_device] plugin.subclass:
> 
> usbip: debug: c:\..\usbip\usbip_vhci.c:130:[usbip_vhci_attach_device] plugin.protocol:
> 
> usbip: debug: c:\..\usbip\usbip_vhci.c:131:[usbip_vhci_attach_device] plugin.addr: 
>

cezuni, 2019-01-08:
@yelfarri You can test with a following workaround. The values are from my USB stick.
```
diff --git a/userspace/src/usbip/usbip_attach.c b/userspace/src/usbip/usbip_attach.c
index f7f51ff..942d2fe 100644
--- a/userspace/src/usbip/usbip_attach.c
+++ b/userspace/src/usbip/usbip_attach.c
@@ -115,6 +115,11 @@ static int query_import_device(SOCKET sockfd, const char *busid, HANDLE *phdev)
                return 1;
        }
 
+       /* manual setting from the values of lsusb -v in linux box */
+        reply.udev.bDeviceClass = 8;
+        reply.udev.bDeviceSubClass = 6;
+        reply.udev.bDeviceProtocol = 80;
+
        /* import a device */
        return import_device(sockfd, &reply.udev, phdev);
 }
```

cezuni, 2019-01-08:
9d72a1a9e43df6b7415f490d89aa318c6af5db8d makes a device to be recognized.
But my usb stick does not seem to be working. :cry:

wrenchonline, 2019-01-08:
my usb stick：
Bus 001 Device 016: ID 0951:1603 Kingston Technology DataTraveler 1GB/2GB Pen Drive
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0951 Kingston Technology
  idProduct          0x1603 DataTraveler 1GB/2GB Pen Drive
  bcdDevice            2.00
  iManufacturer           1 Kingston
  iProduct                2 DataTraveler 2.0
  iSerial                 3 832000000000000045B46338
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)

yelfarri, 2019-01-08:
@wrenchonline   that is odd, can you test with a mouse or keyboard ? this my  do my device are is mouse thus i have the follow configurations. Do you have the same issue ?
``` c
/*3 - Human Interface Device */
BDeviceClass = 3;
/* 1 - Boot Interface Subclass */
BDeviceSubClass = 1;
/* 2 - Mouse */
BDeviceProtocol = 2;
```

@wrenchonlin for usbip in server mode Linux does not assign values for BDeviceClass , BDeviceSubClass, and  BDeviceprotocol . It is the client that signs those values.   When running Linux-host to Linux-client, you can notice the same behavior where the client sets those values. 

From @Oxalin usbip I understood that earlier Linux-host set values than send them over to the client. I have tested that approach, and it worked for a short period but......  both windows and Linux becomes unstable (BSOD@windows and Linux crashes).

cezuni, 2019-01-08:
> I don't know what the actual information on the device is, but they're zero on Linux itself
BDeviceClass = 0;
BDeviceSubClass = 0;
BDeviceProtocol = 0;

@wrenchonline Many devices have zeros in device descriptor.(I got to know through this issue)
Those zero class/protocol means they should be assigned from USB interface descriptor.

> @wrenchonlin for usbip in server mode Linux does not assign values for BDeviceClass , BDeviceSubClass, and BDeviceprotocol . It is the client that signs those values. When running Linux-host to Linux-client, you can notice the same behavior where the client sets those values.

AFAIK, original Linux code resolves those values via additional listing command when importing.
But recent linux kernel(over 4.13 ??) does not seem to provide interface info when a device is not configured. So listing command does not help.
  
9d72a1a9e43df6b7415f490d89aa318c6af5db8d code assigns class/subclass/protocol values from direct usbip protocol.

cezuni, 2019-01-08:
@yelfarri I think that there's no problem in usbip protocol related code.
16777216 in network byte order means 1 in host byte order.
It is just a problem in usbip userland printing routine.     
> Does seems that the pdu header is getting wrong value. This values should be between 1-4 do you have the same issue ?
>
>> usbip: info: from_swapped value = 0
>> usbip: info: swap_usbip_header_endian cmd value = 16777216
>> usbip: error: unknown command in pdu header: 16777216

> In my case the device is regonized as a HID device. There is some improvement that good :).

Improved but windows vhci driver still is not working. I'm working. 😅

yelfarri, 2019-01-09:
@cezuniII have some good news and some bad news. Today i have been doing some testing on the system. I decided that I will reinstall the driver and test again. I m using a USB mouse in Linux host with windows client. And it worked!! It is stilling working after 3 hours running. 
The bad news I have no idea why!

wrenchonline, 2019-01-09:
@yelfarri Ha,ha.This is a little bit the same problem I raised last time in another project. I think how to scientifically debug is to know the problem.

cezuni, 2019-01-09:
@yelfarri , @wrenchonline : You mean that it takes too long to work? 3 hours or 20 minutes.
That may be another issue. My usb stick and mouse over usbip work after 20~40 seconds.
However, my bluetooth dongle has a problem. Its failure is due to that a USB internal IRP is cancelled when usbip command exceeds 500msec. I suspect that this cancellation is done by a upper function driver.

yelfarri, 2019-01-09:
@cezuni my bad it works at once and I had connection running for 3 hours and everything was ok. I will add logs tomorrow :) if u need to take a look at them. 

I was going to test a usb storage tomorrow.

yelfarri, 2019-01-09:
Yeah my mouse works after seconds too. I will take a look one storage tomorrow. Can you provide  the device descriptor you are testing ? I will try to Find something similar to test

yelfarri, 2019-01-10:
> @yelfarri Ha,ha.This is a little bit the same problem I raised last time in another project. I think how to scientifically debug is to know the problem.
> In fact, [Oxalin/usbip_windows#12](https://github.com/Oxalin/usbip_windows/issues/12). All my devices will work on his project, but some will have to wait too long

@wrenchonline  what devices do you need do you have a list that we might need to work our way through?

@cezuniI will create a test plan that will go through a certain type of devices classes so we will try to expose any type of bugs or issues in the system.

maurice-stegmann, 2019-01-11:
Hello, 
what am I doing wrong or what is causing this error? 
Today I've compiled the Project without any errors and installed the driver. 

As I try to attach a device I'm getting following errors:
  *error: recv op_common
  *error: failed to read: err:6

Attaching the same device from a linux machine works fine.

You can see a screenshot from my output below:
[image]

cezuni, 2019-01-11:
@maurice-stegmann : What is a version of linux box? Can you share a linux kernel log?

cezuni, 2019-01-13:
@wrenchonline : I think that usb speed mismatch is not the problem.
Your queue full error may imply that a usbip server does not respond to commands sent by a client.
> They can't quit normally.my usb stick 3.1 . The computer jack is 2.0. Is it related to usb speed driver configuration?

You can increase queue length from below code but it will not resolve the problem.
https://github.com/cezuni/usbip-win/blob/27ec58531963eeca4d679a7ad70e705f00bdb4e4/userspace/lib/usbip_forward.c#L251

What is a compatible id of usbip device which can be obtained from detailed tab of property sheet ? 
I want to check that class/subclass/protocol are still all zero's even if you apply up-to-date repo.

wrenchonline, 2019-01-13:
@cezuni  As you said, I set

>  #define OUT_Q_LEN 2048

It works much better.A few usb pop tests were good.But it will still prompt out queue full.
I look at the log at the beginning there are too many commands.

> usbip_vhci:(QQ)process_urb_res: urbr:[seq:6290], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB
usbip_vhci:(QQ)Bus_Write: Leave: STATUS_SUCCESS

Before that, I tested that the bus device driver could not be recognized when starting up, and sometimes class/subclass/protocol are still all zero's.usb driver not working.(I test this more often)

cezuni, 2019-01-13:
@wrenchonline : Well, if larger queue length helps, there's a significant speed gap between usbip server and client. 2048 queue size would be sufficiently large but waiting for a free usbip command seq slot instead of returning with error will be a more desirable approach. This can be a new issue. 
> It works much better.A few usb pop tests were good.But it will still prompt out queue full.

Is the meaning of "pop" in pop tests "unplugging" a USB device?

wrenchonline, 2019-01-13:
@cezuni  I set

> #define OUT_Q_LEN 8192 

Usb works without prompts.Unplug usb test is normal.Does it have anything to do with Internet speed?I tested it on two VMware OS.

wrenchonline, 2019-01-14:
Some usb devices failed to load drivers.
> usbip_vhci:(QQ)Bus_Ioctl: Enter
usbip_vhci:(QQ)ioctl code: IOCTL_USBIP_VHCI_GET_PORTS_STATUS
usbip_vhci:(QQ)get ports status
usbip_vhci:(QQ)Bus_Ioctl: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_Ioctl: Enter
usbip_vhci:(QQ)ioctl code: IOCTL_USBIP_VHCI_PLUGIN_HARDWARE
usbip_vhci:(QQ)Exposing PDO: addr: 1, vendor:product: 058f:6387
usbip_vhci:(QQ)fdodata->NextLowerDriver = 0xFFFFFA800190D060
usbip_vhci:(QQ)pdo 0xFFFFFA8004467670, extension 0xFFFFFA80044677C0
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)FDO: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP:0xFFFFFA80034D9550
usbip_vhci:(QQ)	QueryDeviceRelation Type: BusRelations
usbip_vhci:(QQ)# of PDOS: present: 1, reported: 1
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)QueryId Type: BusQueryDeviceID
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)FDO: minor: IRP_MN_QUERY_CAPABILITIES, IRP:0xFFFFFA80034D9550
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_TEXT, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)	DeviceTextDescription :USB Device Over IP
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_DEVICE_TEXT, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)	DeviceTextLocationInformation :on USB/IP VHCI
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)QueryId Type: BusQueryInstanceID
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)QueryId Type: BusQueryHardwareIDs
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)QueryId Type: BusQueryCompatibleIDs
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_ID, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)QueryId Type: BusQueryContainerID
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_RESOURCE_REQUIREMENTS, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_BUS_INFORMATION, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_RESOURCES, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_DEVICE_ENUMERATED, IRP: 0xFFFFFA8003D26DC0
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_Ioctl: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_Read: Enter
usbip_vhci:(QQ)Bus_Read: Leave: STATUS_PENDING
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_LEGACY_BUS_INFORMATION, IRP: 0xFFFFFA80034E6A40
usbip_vhci:(WW)not handled: IRP_MN_QUERY_LEGACY_BUS_INFORMATION
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_NOT_SUPPORTED
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_QUERY_RESOURCE_REQUIREMENTS, IRP: 0xFFFFFA80034E6A40
[... 139 lines cut ...]
usbip_vhci:(QQ)pipe num 1:
MaximumPacketSize: 512
EndpointAddress: 0x82
Interval: 0
PipeType: 2
PiPeHandle: 0x00020082
MaximumTransferSize -1
PipeFlags 0x00000000
usbip_vhci:(QQ)Bus_Internal_Ioctl: Enter
usbip_vhci:(QQ)ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB
usbip_vhci:(QQ)process_irp_urb_req: function: URB_FUNCTION_SELECT_INTERFACE
usbip_vhci:(QQ)store_urbr: urbr: [seq:9]
usbip_vhci:(QQ)store_urbr_submit: urbr: [seq:9]
usbip_vhci:(QQ)process_urb_req_submit: urbr: [seq:9], func:URB_FUNCTION_SELECT_INTERFACE
usbip_vhci:(QQ)submit_urbr: urb requested: status:STATUS_PENDING
usbip_vhci:(QQ)Bus_Internal_Ioctl: Leave: STATUS_PENDING
usbip_vhci:(QQ)Bus_Write: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_Read: Enter
usbip_vhci:(QQ)Bus_Read: Leave: STATUS_PENDING
usbip_vhci:(QQ)Bus_Write: Enter
usbip_vhci:(QQ)process_urb_res: urbr:[seq:9], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB
usbip_vhci:(QQ)Bus_Write: Leave: STATUS_SUCCESS
usbip_vhci:(QQ)Bus_PnP: Enter
usbip_vhci:(QQ)PDO: minor: IRP_MN_REMOVE_DEVICE, IRP: 0xFFFFFA80034D9550
usbip_vhci:(QQ)Bus_PnP: Leave: STATUS_SUCCESS

yelfarri, 2019-01-14:
@maurice-stegmann  I need to ask first did you do the following steps? 
Are you sure that the driver is been installed properly ?
Did you put windows in test signing mode ?
run usbip.exe from the folder that contains the following files. 
first follow the steps under seems that the usbip can not find the file name name usb.ids. Than try to attach without using --debug.
```
Copy usbip.exe, usbip_vhci.sys, usbip_vhci.inf, usbip_vhci.cer, usbip_vhci.cat, usb.ids, usbip_stub.sys, usbip_stub.inx into a folder in target machine

You can find usbip.exe, usbip_vhci.sys, usbipd.exe, usbip_stub.sys  usbip_vhci.cer, usbip_vhci.inf in output folder.
usbip_vhci.cat can be found from usbip_vhci subfolder of output folder
userspace/usb.ids
driver/stub/usbip_stub.inx
You can find usbip.exein output folder.

```

maurice-stegmann, 2019-01-14:
@yelfarri : I've installed the driver by adding a legacy device in device manager. I've also copied checked wether the correct files are in the same folder as usbip.exe. Still getting the same error...

Is it possible that I did something wrong during building?

yelfarri, 2019-01-14:
@maurice-stegmann From linux side do you do the follow? 

sudo modprobe vhci-hcd
 driver >sudo modprobe usbip-core
Start usbip deamon                        >sudo usbipd -d
Bind usb device that will be used   >sudo usbip bind -b Device ID

maurice-stegmann, 2019-01-14:
@yelfarri : 

> check the time stamp one your generated files?

Do you mean the date of the files in the usbip directory on windows client machine?
I've tried Building this morning again so the time of the outputted files is right:

[image]

Binding on Linux machine worked properly worked properly. 
```
 @raspberrypi:~ $ sudo usbip list -l
 [...]
 - busid 1-1.2 (058f:6387)
   Alcor Micro Corp. : Flash Drive (058f:6387)
 [...]
pi@raspberrypi:~ $ sudo usbip bind -b 1-1.2
usbip: info: bind device on busid 1-1.2: complete
```

yelfarri, 2019-01-14:
@maurice-stegman 
Can you test something other than usb flash drive, such as a mouse ? does it work for you ?
for testing one again can you  show me how driver looks like on device manager ?
[image]

maurice-stegmann, 2019-01-14:
@yelfarri : I've tested a usb mouse and usbip on the client side behaved strange...
As I listed the devices my Mouse has shown up as expected:
```
C:\usbip>usbip.exe list -r 172.22.0.212
Exportable USB devices
======================
 - 172.22.0.212
      1-1.3: Microsoft Corp. : Comfort Mouse 4500 (045e:076c)
           : /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3
           : (Defined at Interface level) (00/00/00)
```
After I tried to attach the devices I've got no console output and wasn't able to quit the process hitting ctrl + c or closing the console window or quitting usbip by task manager. 

In Device Manager USB over IP is listed as system device. Heres a Screenshot of the details: 
[image]

[image]

I'm so sorry for bothering...

yelfarri, 2019-01-14:
@maurice-stegmann  no worries mate you are not bothering me. This good we need to find any bugs related to the driver :). Does mouse shows up under hid devices ?

yelfarri, 2019-01-14:
@maurice-stegmann Everything is looking good for far....Can you  create a new rule for windows firewall where the TCP port 3240 should be allowed  for inbound and outbound traffic

maurice-stegmann, 2019-01-15:
@yelfarri : The firewall on my windows machine was disabled during testing so I don't think that firewall is causing issues.

maurice-stegmann, 2019-01-15:
@yelfarri : This morning, testing attaching the  pendrive I've tried before:

>  @raspberrypi:~ $ sudo usbip list -l
>  [...]
>  - busid 1-1.2 (058f:6387)
>    Alcor Micro Corp. : Flash Drive (058f:6387)
>  [...]
> ```

resulted in:
[image]

[image]

yelfarri, 2019-01-16:
@maurice-stegmann  sorry for the slow replay I was busy moving to a new apartment and as well trying to find the issue. I installed a clean windows and did the same steps to install the driver. I have got the same issue as you did. I fixed it by doing the following. 

Delete your old driver 

[image]

Verify that your driver is remove. 
Navigate to C:\Windows\System32\drivers and search for usbip_vhci.sys.
delete usbip_vhci.sys
Restart your windows 10...

 run  cmd as administrator past those two commands.
> bcdedit.exe -set loadoptions DISABLE_INTEGRITY_CHECK
> bcdedit.exe -set TESTSIGNING ON

Than restart....

Install USBIP test certificate
Install driver/usbip_test.pfx(password: usbip)
Certificate should be installed into "**Trusted Root Certification Authority"** and **"Trusted Publishers"** on **local machine**(not current user)

**Copy usbip.exe, usbip_vhci.sys, usbip_vhci.inf, usbip_vhci.cer, usbip_vhci.cat into a folder in target machine. **The files should be in the same folder****

Install USBIP vhci driver

Than restart....
Use usbip.exe

yelfarri, 2019-01-16:
@maurice-stegmann i have tested a  mouse, keyboard and joystick. Seems that issue happen when the driver is not installed properly. that the only think i notice. But i really have no idea why it happens in your pc.  you get the follow  water mark on the bottom right corner of your windows ?

 
[image]

maurice-stegmann, 2019-01-17:
@yelfarri Yes, I get this watermark. 
Where you able to use mouse, keyboard and joystick, after you attached them? On my machine, the mouse driver loaded properly but I just wasn't able to use it (click or move coursor).

cezuni, 2019-01-17:
@maurice-stegmann : If a mouse pointer is not moved on usbip client, something goes wrong.
Your problem does not seem to be a just mis-configuration. 
> On my machine, the mouse driver loaded properly but I just wasn't able to use it (click or move coursor).

Windows and linux kernel logs would be helpful to find the problem. Or usbip forwarding debugging logs. You can just define DEBUG_PDU in following code.
https://github.com/cezuni/usbip-win/blob/27ec58531963eeca4d679a7ad70e705f00bdb4e4/userspace/lib/usbip_forward.c#L33

yelfarri, 2019-01-17:
@maurice-stegmann at least we went through every configuration I could think of. Yeah, I manage to use them as if they were connected to the client pc.  Can you provide the debugging logs as @cezuni suggested?

maurice-stegmann, 2019-01-17:
@cezuni excuse me, how do i define DEBUG_PDU? I've removed the # wich resulted in build errors

yelfarri, 2019-01-17:
@maurice-stegmann just add 

>  #define DEBUG_PDU //<<<< add this
> #ifdef DEBUG_PDU

hubbybob2015, 2019-01-22:
> @yelfarri Ha,ha.This is a little bit the same problem I raised last time in another project. I think how to scientifically debug is to know the problem.
> In fact, [...]

this  project   Oxalin/usbip_windows#12，I would like to know if the usb camera can work in windows client successsfully?

yelfarri, 2019-01-22:
@hubbybob2015 Sorry but I have no idea...   Maybe @wrenchonline   can answer that

keirlessmi, 2019-05-18:
hello,
plz where i can find usbip_vhci.sys, usbip_vhci.inf,
thank you

keirlessmi, 2019-05-18:
hello
how can i  generate the project to get the files
thank you

cezuni, 2019-05-20:
@keirlessmi : There's builtin binaries. Check [release page](https://github.com/cezuni/usbip-win/releases). And [Readme.md](https://github.com/cezuni/usbip-win/blob/master/README.md) is the best for usbip-win build guide. 😄

keirlessmi, 2019-05-20:
@cezuni thank you for the answer

svirot, 2020-04-07:
Hello,
Similar problem (probably) whilst trying to use a PL2303 (USB to Serial converter) from a Linux server to a Windows client. In a nutshell, it gives the same "Code 10" error with the proprietary driver and comparing the USB messages from Wireshark, the first difference is a lack of "GET STATUS Request" (i.e. URB_FUNCTION_GET_STATUS_FROM_DEVICE) when using usbip. I have more details if it can help.
Given the 7a92e8e commit essentially simulates the status request/reply for the driver side, the PL2303 hardware probably needs that status request to progress in its initialisation process. This may be confirmed by the next command in the stream, a "SET CONFIGURATION" to which the device (through usbip) replies with blanked data (of the same size than the request buffer).
I guess adding the feature properly would need an update to the server side as well, meaning patching the Linux kernel which would make it tricky.
Please do not hesitate to ask if there is anything I can help with.
Thanks for the great with you put into this project.

cezanne, 2020-04-08:
@svirot : 
> Given the 7a92e8e commit essentially simulates the status request/reply for the driver side, the PL2303 hardware probably needs that status request to progress in its initialisation process

That's a good point. :+1:  Why don't you create another issue ?

[5 more replies]