cezanne/usbip-win #7: Vendor specific device class

cezuni, 2019-01-14:
As @wrenchonline reported (https://github.com/cezuni/usbip-win/issues/6#issuecomment-453907301), USB device with USB class 255(vendor-specific) is not working.
Can it be resolved similarly such as a device with class 0 ?

wrenchonline, 2019-01-14:
Two questions（1）In fact, Class 255, which cannot be parsed, Nor can it be parsed by official drivers.Usbip does not parse class 255.（2）In addition, I found that with your usbip user space code, part of the class 0 usb device will receive IRP_MN_REMOVE_DEVICE IRP, After causing the identification of "big Usb storage device".Usb not working.This is rarely the case with class 0 devices.

cezuni, 2019-01-14:
@wrenchonline : (1) Yeah, 255 class had not been treated.. But this case can be easily improved. (2) You mentioned IRP_MN_REMOVE_DEVICE happens. Honestly, I have no idea for that case. More thoughts needed?

wrenchonline, 2019-01-15:
After debugging, I found that the function process_urb_res_submit went through the URB_FUNCTION_SELECT_CONFIGURATION -> select_config->IOCTL_INTERNAL_USB_SUBMIT_URB .system to issue an issue before removing the device irp.

> MaximumPacketSize: 0
EndpointAddress: 0x00
Interval: 0
PipeType: 0
PiPeHandle: 0x00000000
MaximumTransferSize -1
PipeFlags 0x00000000
usbip_vhci:(QQ)pipe num 0:
MaximumPacketSize: 512
EndpointAddress: 0x01
Interval: 0
PipeType: 2
PiPeHandle: 0x00020001
MaximumTransferSize -1
PipeFlags 0x00000000
usbip_vhci:(QQ)pipe num 1:
MaximumPacketSize: 0
EndpointAddress: 0x00
Interval: 0
PipeType: 0
PiPeHandle: 0x00000000
MaximumTransferSize -1
PipeFlags 0x00000000
usbip_vhci:(QQ)pipe num 1:
MaximumPacketSize: 512
EndpointAddress: 0x82
Interval: 0
PipeType: 2
PiPeHandle: 0x00020082
MaximumTransferSize -1
PipeFlags 0x00000000

Is that normal?
[image]
[image]

cezuni, 2019-01-15:
> @wrenchonline : (1) Yeah, 255 class had not been treated.. But this case can be easily improved. (2) You mentioned IRP_MN_REMOVE_DEVICE happens. Honestly, I have no idea for that case. More thoughts needed?

After more thought, IRP_MN_REMOVE_DEVICE may be issued by an upper USB class driver. This problem was addressed at https://github.com/cezuni/usbip-win/issues/6#issuecomment-452890754 .

cezuni, 2019-01-15:
@wrenchonline : In my opinion, IOCTL_INTERNAL_USB_GET_TOPOLOGY_ADDRESS is not related to device removal. As noted before, I suspect that USB command timeout (500msec) makes PNP manager to issue IRP_MN_REMOVE_DEVICE. By the way, do your logs have any cancellation of an issued IRP ? The reason why i ask is that IRP cancellation is preceded before removing a device.

wrenchonline, 2019-01-15:
You are saying that the usbip command failed to respond to the message, resulting in a lack of information.
The official drive succeeds because it is asynchronous and full of messages beforehand.

wrenchonline, 2019-01-15:
@cezuni I try the submit_urbr function to get the next section of the cancel irp.No, just lots of STATUS_PENDING.

wrenchonline, 2019-01-15:
When I look at the failed device, it doesn't call process_urb_res_submit's URB_FUNCTION_SELECT_INTERFACE.Only successful devices invoke it
[image]

wrenchonline, 2019-01-16:
@cezuni I found out the problem!In the function: process_urb_res_submit()
       ` /*if (hdr->u.ret_submit.status != 0) {
		urb->UrbHeader.Status = to_usbd_status(hdr->u.ret_submit.status);
		return STATUS_UNSUCCESSFUL;
	}*/`
Remove it and there is no IRP_MN_REMOVE_DEVICE.After waiting for 20 minutes, my usb flash drive can be loaded and working, which is the same as the official one.Because I found a new problem, processing data in the kernel log seq1-75 queue took me 20 minutes!There are canceled irps between 1-75.It seems like there were a lot of unprocessed operations that took that long before that.

cezuni, 2019-01-16:
@wrenchonline : If you comment out following lines, vhci cannot handle a real faulty error.

https://github.com/cezuni/usbip-win/blob/27ec58531963eeca4d679a7ad70e705f00bdb4e4/driver/vhci/vhci_write.c#L239-L242

> It seems the return STATUS_UNSUCCESSFUL;Is the upper driver issued IRP_MN_REMOVE_DEVICE key!

I finally have realized why IRP cancellation was not found when your device was removed. In your case, IRP actually failed!! Yeah, usbip-win logging is too bad to miss an important event.    
I think that dropping STATUS_UNSUCCESSFUL cannot be a desirable solution. How about trying to find the reason why usbip command fails at linux box. Linux kernel logs may provide a useful information.

wrenchonline, 2019-01-16:
@cezuni I found that the usb device not only loaded the "high-capacity storage device" driver (success), but also loaded the "disk device" driver (failure).
[image]

cezuni, 2019-01-16:
@wrenchonline : Your hub on linux doesn't seem to work correctly. "device descriptor read/64, error -110" is generated by drivers/usb/core/hub.c in linux kernel. -110 error is ETIMEDOUT. 
I recommend you to check a physical or virtual hub.

wrenchonline, 2019-01-16:
@cezuni  This is the log I uncommented.If not removed, there is only one Linux kernel log
[image]

wrenchonline, 2019-01-16:
@cezuni  I think win and Linux kernel layer are different, some irp is useless operation, such as both sides of the disk drive

cezuni, 2019-01-16:
@wrenchonline : I'm happy to see your smile. Maybe, returning STATUS_SUCCESS is more reasonable. I 'll check it out more.

cezuni, 2019-02-08:
@wrenchonline : Is your vendor specific class device working correctly if a recent vhci driver is applied ?

closed 2019-07-05 by cezuni