cezanne/usbip-win #8: usbip_vhci driver install failed unusually on windows7

luochaoying, 2019-01-15:
hello,cezuni
  There is report an error after  i download the source codes and compiled usbip-win, installed the USBIP test certificate and USBIP vhci driver subsequently. but the USB/IP VHCI display a abnormity icon at the device manager. i used the the windows OS version is win7,is it results in this issue of USBIP vhci driver? 
Looking forward to your reply,thanks!

My Build Tools is as below:
Visual Studio 2017 Community
Windows SDK 10.0.17134
Windows Driver Kit 10.0.17134
Build is tested on Win7

[image]
[image]
[image]

luochaoying, 2019-01-15:
C:\usbip-win-master\Debug\x64>usbip --debug attach -r 192.168.133.34 -b 2-1
usbip: debug: c:\usbip-win-master\userspace\src\usbip\usbip.c:133:[run_command]
running command: `attach'

Displaying the Installed Devices

====>> found the USB/IP VHCI
usbip: error: open vhci driver
usbip: error: failed to read: err:6

C:\usbip-win-master\Debug\x64>

cezuni, 2019-01-15:
@luochaoying : Not tested on Windows 7. Can you try the vhci driver at release page ? 
(https://github.com/cezuni/usbip-win/releases/download/0.0.1/usbip_vhci.sys)
You can just copy it after disabling VHCI device.

yelfarri, 2019-01-16:
@luochaoying This might be related to issue seen here. can you test method i test on https://github.com/cezuni/usbip-win/issues/6#issuecomment-454760066 this worked for me. Does it work for you?

yelfarri, 2019-01-17:
@luochaoying  I don't know:). Can you check a mouse or keyboard first does it work? After that can you try to the web camera? You might be the first to have tested this device class :). If it does not work can you provide debugging logs and information?

luochaoying, 2019-01-17:
cezuni and yelfarri
i'm try,that mouse,USB flash drive,them are work well,but the UVC camera driver always have exception,can't be installed correctly  in Device Manager of the win10.

yelfarri, 2019-01-17:
@luochaoying  Does mouse work on windows 7 ?

hubbybob2015, 2019-01-22:
@luochaoying the usb camera driver cannot be installed correctly ,the Devce Manager shows "USB Device over IP" for warning ,it 's a bug? And when I attcched my USB Audio in the win10 usbip client caused bluescreen,it's a bug? But the mouse,keyboard and mass storage can work! @cezuni @yelfarri

jkhunter, 2019-02-07:
Hi, I have the issue with the vhci driver, too. I already check that the certificate is stored on local machine, and on both entries it says, that i own the private key of the certificate.
I followed the instructions on https://github.com/cezuni/usbip-win/issues/6#issuecomment-454760066 but I still have error code 39, and cannot use the driver.

yelfarri, 2019-02-08:
@jkhunter  How do you install your driver ? do you use device manager -> Add legacy hardware ? I have not tested on windows 7. I want to help :) but it is hard understand the source of the problem based on comments :(.  It seems that people have issues with windows 7. is it possible to do a skype meeting ? If yes drop me an email. We can plan meet where we can find the issue and how we can fix it.

tatokis, 2019-02-14:
Since the issue I created seemed to be marked as a duplicate of this, I felt that it would be helpful to link https://github.com/cezuni/usbip-win/issues/12#issuecomment-463534695 here too.
The post contains binaries that may work for more people using Windows 7.

I have *not* tested them on any other computers, and I can *not* guarantee that they will work for anyone else. I also take no responsibility if something goes wrong.
You have to install the included .cer file into "Trusted Root Certification Authority" on local machine before installing the driver.

jkhunter, 2019-02-14:
Thank you. This new binary works for me, too.
Now, I can able to attach a device to win7 and it shows up in device manager. 

When I want to start using the attached device, I'm getting an new error: "Failed t record. out queue full." But I think this is a different story and I should investigate to it by myself.

Oxalin, 2019-03-08:
@luochaoying Would you be able to test tatokis' solution, which is to change the target OS before building the driver? Please let me know if this works for you so we may close this bug or continue investigating it.

luochaoying, 2019-03-17:
OK.thanks.

closed 2019-07-10 by cezuni

terryshenzhen, 2020-09-05:
> @luochaoying : Not tested on Windows 7. Can you try the vhci driver at release page ?
> (https://github.com/cezuni/usbip-win/releases/download/0.0.1/usbip_vhci.sys)
> You can just copy it after disabling VHCI device.

My enviroment is win10, encounter this issue too:
usbip: error: open vhci driver
usbip: error: failed to read: err:6

I use win10 as client, and run usbip.exe attach -r 192.168.1.232 -b 1-1    then meet it;
usbip.exe list -r 192.168.1.232   ---> ok
uninstall usbip from the device manager and reinstall (https://github.com/cezuni/usbip-win/releases/download/0.0.1/usbip_vhci.sys),  meet the same error.

terryshenzhen, 2020-09-05:
> OK.thanks.

hi My brother, I encounter the same issue on win10, my OS system is Chinese too.---我也是中文的系统。
How did you resolve it at the finally?

terryshenzhen, 2020-09-05:
@luochaoying @yelfarri @cezuni Hi My friends, I can use usbip-win now. let me conclusion:
my enviroment is win10, encounter this issue:
usbip: error: open vhci driver
usbip: error: failed to read: err:6----------->but my error is : usbip: error: read_devbuf: failed to read: err:6

Reason: vhci driver is not correctly installed.

in the device manager, usbip-win VHCI root is yellow, next step: remove this yellow.  
BCDEDIT -SET LOADOPTIONS DISABLE_INTEGRITY_CHECKS

BCDEDIT -SET TESTSIGNING ON 
(@yelfarri 's comment has some error about the first command)

when do this, I must forbid the security options in BIOS or this cmd will no use. refer (https://blog.csdn.net/u012891962/article/details/96701710)

then reinstall, yellow is gone.

reinstall are:
usbip-win-master----Debug---x64----- usbip-vhci.cer
Install driver/usbip_test.pfx(password: usbip)
usbip.exe  install

attach is:
usbip.exe attach -r 192.168.1.232 -b 1-1

then it works well, I can see usb disk on the linux board via winsows 10. copy , past are OK.
------------------------
Thank you all, I learn more from you!

CpServiceSpb, 2022-09-27:
Which is the latest driver version working under Windows  7 ?
I have error 38 as well.