cezanne/usbip-win #9: uvc (USB video device class) camera/webcam

yelfarri, 2019-01-17:
@luochaoying  i'm try,that mouse,USB flash drive,them are work well,but the UVC camera driver always have exception,can't be installed correctly in Device Manager of the win10.

hubbybob2015, 2019-01-22:
@yelfarri  the usb camera driver cannot be installed correctly ,the  Devcei Manager shows "USB Device over IP" for warning ,it 's a bug? And when I attcched my USB Audio in the win10 usbip client caused bluescreen,it's a bug?

yelfarri, 2019-01-22:
> @yelfarri the usb camera driver cannot be installed correctly ,the Devcei Manager shows "USB Device over IP" for warning ,it 's a bug? And when I attcched my USB Audio in the win10 usbip client caused bluescreen,it's a bug?

According to @luochaoying  yes it is. I have no access to a usb audio or UVC camera. I did not experience a bluescreen can you share more details on what USB audio device your using, and some bugs reports? :)

hradec, 2019-01-23:
I just installed usbip-win on a windows 10 home running inside a KVM. 

Everything went well, and I was able to attach to a `Microsoft Corp. : LifeCam Cinema (045e:075d)` on my linux machine without a problem... no error messages, nothing! 

but, this is how the Lifecam shows in device manager:
[image]

Although it's not `USB Device over IP`, it seems the webcam was passed on to windows as a input device, not a media device. 

in my linux machine, this is how usbip lists the device: 
[image]

I'm not sure where the problem lies, but I thought it was weird that usbip on linux recognized the device as `Miscellaneous Device / ? / Interface Association (ef/02/01)`... but maybe that's just how media devices are recognized by usbip? 

anyhow... I'm running the usbip released from cezuni/usbip-win depot, btw... I've downloaded the binaries! 

any ideas?

cezuni, 2019-01-23:
> I'm not sure where the problem lies, but I thought it was weird that usbip on linux recognized the device as Miscellaneous Device / ? / Interface Association (ef/02/01)... but maybe that's just how media devices are recognized by usbip... or maybe the linux usbipd server miss interpreted it and its the reason windows got it as a input device?

@hradec : I think that the problem lies in usbip-win code itself. 😢 
I suspects that windows VHCI driver does not handle an usbip command in iso transfer mode.

> anyhow... I'm running the usbip released from cezuni/usbip-win depot, btw... I've downloaded the binaries!

It's good news that uploaded binaries are working correctly. In fact, I had not checked them.

> @cezuni: if you wan't I could redirect my linux usbipd server port to my webserver IP address so you can try it yourself with my MS Lifecam webcam... let me known! :)

That's a good idea 😆 but sometimes resetting usbip on linux is required. Also linux logs.
Fortunately, I found my mac notebook has a usb camera and barely do a work.

cezuni, 2019-01-25:
Recent commit(cf70c869d0ad559db1dda739e665df708c7dc9a6) does not solve this issue. 😢 
After some efforts, I have found that my borrowed USB cam(MS lifecam) is a sort of [IAD device](https://docs.microsoft.com/ko-kr/windows-hardware/drivers/usbcon/usb-interface-association-descriptor), which has EF/01/02(class/subclass/protocol) in device descriptor. [MSDN](https://technet.microsoft.com/en-us/ff537107(v=vs.90)) tells me that a hardware id and compatible id of such a device should be composed in different way. 

This issue seems to need more commits.

Oxalin, 2019-01-28:
I do have a ID 046d:0826 Logitech, Inc. HD Webcam C525 at hand. I could give it a go. I've seen some bug reports under SourceForge about webcams not being supported correctly. Here is on such example: https://sourceforge.net/p/usbip/discussion/418507/thread/d1256c5b/

As you said @cezuni , it was pointing to isochronous transfer.

cezuni, 2019-01-28:
@Oxalin: Currently, Windows kernel detects my MS camera correctly by adjusting class/subclass/protocol fields according to IAD. But it's not working yet. Your cam is a sort of IAD ?

wrenchonline, 2019-01-29:
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/support-for-interface-collections There are too many compatibility ids and registry controls.

cezuni, 2019-01-29:
@wrenchonline : In my opinion, 4 ways in [MSDN](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/support-for-interface-collections ) may help if a driver has no control over compatible id. But we have a vhci and can let usbccgp.sys enumerate a composite device by returning USB\COMPOSITE as a compatible id. Recent commit 393ac6ac336198a8706724687c4a3ca9565b265f adds USB\COMPOSITE to compatible ids if a device is IAD type. Now my cam is detected correctly, *but* it's still not working. 😢

wrenchonline, 2019-01-30:
https://community.osr.com/discussion/290999/usb-3-0-camera-streaming-issue Do you check your usb device parameter ?Unfortunately, I do not have the equipment to provide testing and help.

cezuni, 2019-02-08:
I have checked that MS lifeCam is working with 0.0.4 release though camera app screen is not perfect. Poor camera quality may be due to usbip-win performance problem.

No visible screen in my previous comment was caused by too many debug messages. After removing those nuisance logs, camera app shows captured images.
> Now camera app seems to work with my MS lifeCam. However, the app does not show a visible screen even if it runs over 10 minutes. From kernel logs, vhci driver simply does isochronous transfers repeatedly. I'm not sure that this is due to slow network/performance or another bugs.

closed 2019-03-18 by cezuni