cezanne/usbip-win #10: Driver Signing on Windows 10

yelfarri, 2019-01-23:
In the case of device drivers, signing is even required by certain versions of Windows in certain situations. Currently, the USBIP requires windows 10 to run in test mode to be able to install the driver.  Does anyone from here know how this driver can be signed/approved by Microsoft? Does Microsoft provide an open source license?

From my understanding, Microsoft announced on 2015-04-01 that all new Windows 10 kernel mode drivers must be submitted to and digitally signed by the Windows Hardware Developer Center Dashboard portal, which is a web service provided by Microsoft.  According to the link under there should be two different routes for getting the driver signed "Hardware compatibility" and "Attestation". 

**The Hardware Compatibility route** is the old route where your driver is subjected to a series of tests designed  by Microsoft to ensure its quality.  Then Microsoft will supposedly sign your driver properly for all the versions of Windows you are interested in, and you can have a single driver that works on all those versions.   
          
**The attestation route** is the new route where your driver does not have to pass any tests, but the resulting driver only works on Windows 10.             

The portal will only accept driver submissions from you if you sign them with an Extended Validation (EV) certificate, which is typically more expensive than a normal certificate.

  [https://www.osr.com/blog/2015/07/24/questions-answers-windows-10-driver-signing/](url)
@cezuni this more a case of thinking about the future and get inputs from people.

cezuni, 2019-01-23:
@yelfarri : Thanks, I'll try to read a link. (I slightly edited your comment because link is not working) 
> https://www.osr.com/blog/2015/07/24/questions-answers-windows-10-driver-signing/
@cezuni this more a case of thinking about the future and getting some inputs from people.

yelfarri, 2019-01-24:
> @yelfarri : Thanks, I'll try to read a link. (I slightly edited your comment because link is not working)
> 
> > https://www.osr.com/blog/2015/07/24/questions-answers-windows-10-driver-signing/
> > @cezuni this more a case of thinking about the future and getting some inputs from people.

Thanks for invite :D, sure if i find something about signing i will write it here to keep the thread public and accessible for every body :).

yelfarri, 2019-02-19:
@cezuni **The attestation route** will be the most possible for our case. As Microsoft announced that Windows 7 reached the end of life and support will end 14 January 2020. I can argue from our case I find the case of attestation the most favorable. The **Hardware Compatibility route** and **attestation route** both require a set of steps to be fulfilled but as I said in the early post the later does not require to pass any tests from  Microsoft. 

The attestation route requirements are described well in the [following article](http://wrogn.com/tag/driver-signing/) but since I know you don't have a lot of time I will write a quick summary :). 

The New requirements from microsoft means the following 

- The “old” way meant you — as the software developer — use your certificate to sign your software. Since your certificate is cross-signed by a certification authority (CA) that is (in the end) trusted by Microsoft this ensures that no one could tamper with the file and that Microsoft trusts that you are who you claim you are.
- The “new” way means that you submit your software to Microsoft and they add their certificate, provided all the requirements are met.
- For that, you need an EV code signing certificate, there is no way around it. This basically means, it’s more expensive and it comes on an USB hardware token (so you cannot copy it).

The link describeds the steps needed for achieve signed driver from microsoft

- Getting an Extended Validation(EV) certificate this can be purched from https://www.globalsign.com/en/ the article says that the certificate will costed (2016-09-09) about  709€ for 3 years.
-  Getting the Microsoft Signature need a  Microsoft sysdev account to submitt  the required files to microsfts for a secound signing the process is explained in details in article linked 

@cezuni  Can we have a skype chat about this ? I might have some ideas how to do this but i need your permission

cezuni, 2019-02-19:
@yelfarri : To be honest, usbip-win driver is not stable for signing. Although usbip-win drivers still have many flaws, signed driver would be nice. I will agree to your idea if you do not send me bills for EV. 😄I'm ready to listen your idea with written language. (Forgive me to refuse skype chat due to my hell conversation)

yelfarri, 2019-02-19:
@cezuni No worries we can skip the skype part. I understand that the driver is not yet ready to be signed yet. The way I see it we need to figure out a way how to get verification.  Like you said EV bill and the fact that Microsoft requires an organization to submit a driver, not individuals.  

The issues on hand
1. How to pay for EV bill? I have been in discussion with some individuals/companies who are interested in backing this project to obtain an EV certificate. I refer them to your GitHub where they liked what they saw
   
-  Our conditions that this should stay open source and should not be altered based on their interests
-  Their conditions that all the donate on the condition that the money will result into a signed driver

2.  Personally, I would love to contribute personality to the project. Can we set a donation page to collect money to fund EV certificate? So any fan/user can help out in making this possible. With that, we might need to make members to oversees that the contributions are going to the goal of making this open project succeed. using  https://www.patreon.com/

3.  We are not an organization thus we cannot submit the driver to Microsoft(if I m wrong please correct me).... How can we solve this?
Should we become an organization to be able to control the signature of the driver? What do you think about this? Do you have any ideas?

I really want this USBIP windows to survive the time and the only way to do so is by building a community that is actively using this solution.

cezuni, 2019-02-20:
@yelfarri : I *deeply* understand your concerns. However, my bottom line opinion is that usbip-win requires a development more than a certificate. 😄 

My answers to your issues:
1. Welcome!! Nothing to be addressed more. However, using the company's certificate will be more feasible instead of creating a new one. 
2. I think that a donation page requires more quality for usbip-win. Should we take a deep breadth?
3. It's not easy to build up a organization for this project. I heard that there exists a specialized group to support open source project such as [letsencrypt](https://letsencrypt.org/).

yelfarri, 2019-03-04:
@cezuni I agree that more development is needed :). 
1. I see I will tell them that
2. sure than that something we can work. Do you have any idea on areas you will like to improve ? Can we make an overview over functionality that needs to be improved ? no one knows the code better than you ;)
3. I just learned something new today :). I will point those companies to letsencrypt. Is there an equivalent organization for windows products ?

cezuni, 2019-03-04:
@yelfarri : I mean that usbip-win is too unstable to get donation, at least right now. 😢 Maybe we need bug fixes instead of improvement. 
For 3, it's sad that letsencrypt does not provide a EV certificate.

doganmurat, 2019-03-04:
Hi @cezuni  and @yelfarri,

I am also interested in this project and my company has an EV Signing Code cert.
We can always sign them for testing.

But I think as @cezuni said it is too early for this kind of things.
This project is at initial phase and I think there will be lots of phases.

My company is looking for an open source Remote USB solution & so I am watching this project.
 Donation always could be on the desk.

Regards,

yelfarri, 2019-03-05:
@doganmurat and @cezuni  yeah agree that the code needs multiple phases to be considered for signing. My goal is to start a discussion about the process early. By doing so, we can capture the interests of people who want to pitch in with ideas :). It is good that there are more people interested in this. Where I work and where my friend's work we are interested in finding a way to use @cezuni 's driver.  Where I work, we do have an EV signing code cert as well. I can sign a driver for my self, but I think it is fair that companies who use @cezuni work at least give support to the project as engineering hours(debugging, development...) or donations. In that way,  we give back to the community.

zhLoser01, 2019-05-06:
@yelfarri My question is not caused by the driver signature. 
[image]

yelfarri, 2019-05-09:
@zhLoser01  I see that @cezuni answer you in #32. Can we keep these issues only about driver signing :)

cezuni, 2019-07-26:
@yelfarri : How about closing this old issue? If needed, reopen it or start with a new issue page.

Ilia1, 2019-08-12:
https://www.globalsign.com/en/ssl/ssl-open-source/ for open projects the certificate is free

Ilia1, 2019-08-12:
https://sourceforge.net/projects/signed-drivers/ It seems here can help with driver signing

cezuni, 2019-08-16:
@Ilia1 : Thanks for information. I have dropped [the request](https://sourceforge.net/p/signed-drivers/discussion/general/thread/587929f47e/?limit=25#d134) in the project site.

xoocoon, 2019-08-21:
Hi all, 
above all: thanks to cezuni for this great piece of work! I'd also love to use it with signed drivers as disabling Secure Boot in UEFI exposes a security risk. I would also be willing to donate some amount to support driver signing, even if it's still in beta state. Maybe having the drivers signed by the ReactOS people would be an option? This has been mentioned already by somebody else. The predecessor Windows client from 2011 was also signed by ReactOS.
Best regards, xoocoon

cezuni, 2019-08-22:
@xoocoon : Alas, [ReactOS](https://reactos.org/wiki/Driver_Signing) seems to no longer support driver signing. Anyway, I remember you when a donation is required someday. 😄

closed 2020-03-14 by cezanne