cezanne/usbip-win #12: usbip_vhci not loading on Windows 7

tatokis, 2019-02-08:
I installed the `usbip-win 0.0.4` pre-release on Windows 7 (7601), and even after manually generating a signed cat file, windows refuses to load the driver with `Code 39`.

Event viewer says 
```
Application popup:  : \SystemRoot\system32\DRIVERS\usbip_vhci.sys cannot find ntoskrnl.exe KeInitializeSpinLock
```
which looks like Win7 compatibility was broken.

I tried compiling the driver with the old build system and the Win7 DDK but there were quite a few issues and I could not get it to compile successfully.

I do not have access to a proper Windows 10 dev installation at the moment to have a look, so it would be appreciated if this could be fixed.

cezuni, 2019-02-08:
This issue seems to be duplicate of #8. Anyway, it is almost obvious that usbip-win is not working on win7. But in my opinion, win7 support cannot be a high priority. Win10 is already a major version and there are so many malfunctioning and unstable features even on win10. 😢

tatokis, 2019-02-14:
Thank you for the response.
I took the time to install VS2017 and the Win 10 WDK.

I changed the project's Target OS Version in VS to Windows 7, and then built the driver.
[image]

Windows 7 was then able to successfully load my newly built driver.
[image]

I have attached x64 debug binaries built from https://github.com/cezuni/usbip-win/commit/94b35a0916962874f062076176ec597a612aaa77 for anyone wanting to try to run this.
[usbipwinx64.zip](https://github.com/cezuni/usbip-win/files/2863924/usbipwinx64.zip)

closed 2019-02-14 by tatokis

Oxalin, 2019-03-08:
@tatokis Adding you workaround to the README.md.

Eric-N-Be, 2019-07-30:
thanks for you investigation, it's solve my drivers install to a windows 2012 r2 (client side), i can continue my test