cezanne/usbip-win #13: usbip server on Win

igoryonya, 2019-02-10:
Your readme explains installation of server on linux, but I thought, that your port is for windows.
Can I use usbip server on windows?

cezuni, 2019-02-10:
> Your readme explains installation of server on linux, but I thought, that your port is for windows.

INSTALL section in readme consists of 2 parts: USBIP server & client.

> Can I use usbip server on windows?

Sure, you can. usbip server on windows was the first motivation to start this repo.

igoryonya, 2019-02-11:
Do I have to compile it with VC++, or can I compile it with GCC or Ming?

cezuni, 2019-02-11:
I had compiled with Visual studio 2017. Detailed build environment is presented in readme. Other build tools are not tried yet. You can be a pioneer. :)

closed 2019-07-05 by cezuni