cezanne/usbip-win #14: Bluescreen when sending a constant stream of URBs

tatokis, 2019-02-18:
I have an application that sends a lot of URB Interrupts in a very short period of time, and after a while, usbip starts unlinking them.

With the new unlinking code introduced by https://github.com/cezuni/usbip-win/commit/10d26c6e3f00fe716c462349b103ddf9809c74ba in vhci, a bugcheck is triggered after a few URBs get unlinked.

WinDbg shows that the issue occurs here https://github.com/cezuni/usbip-win/blob/94b35a0916962874f062076176ec597a612aaa77/driver/vhci/usbreq.c#L114-L115

The following patch prevents the segfault, however I haven't made a PR because I am not sure if it is the right approach to follow.
Additionally, while the patch solves the bluescreen, the usbip client then throws a few `failed to record. out queue full` errors, and the connection seems to freeze. I can still detach and reattach the device though.

```diff
diff --git a/driver/vhci/usbreq.c b/driver/vhci/usbreq.c
index db5bb0c..261ee39 100644
--- a/driver/vhci/usbreq.c
+++ b/driver/vhci/usbreq.c
@@ -111,8 +111,10 @@ remove_cancelled_urbr(pusbip_vpdo_dev_t vpdo, PIRP irp)
 
 	urbr = find_urbr_with_irp(vpdo, irp);
 	if (urbr != NULL) {
-		RemoveEntryList(&urbr->list_state);
-		RemoveEntryList(&urbr->list_all);
+		if ((&urbr->list_state)->Blink != NULL && (&urbr->list_state)->Flink != NULL)
+			RemoveEntryList(&urbr->list_state);
+		if((&urbr->list_all)->Blink != NULL && (&urbr->list_all)->Flink != NULL)
+			RemoveEntryList(&urbr->list_all);
 		if (vpdo->urbr_sent_partial == urbr) {
 			vpdo->urbr_sent_partial = NULL;
 			vpdo->len_sent_partial = 0;

```

I am not sure how to further debug this issue. Any help would be appreciated. If the patch above is acceptable, then I can open a PR to get it merged.

cezuni, 2019-02-19:
@tatokis : Your report may imply that removing urbr does not guarantee mutual exclusion and incur memory corruption. If list related fields of urbr are NULL, the memory has been already broken. So I greatly appreciate your patch but it seems to be not needed in normal case.

One of unpleasant things is fixed out queue length in userspace forward routine. :cry:  Anyway, your application manage to work by increasing OUT_Q_LEN.  
https://github.com/cezuni/usbip-win/blob/94b35a0916962874f062076176ec597a612aaa77/userspace/lib/usbip_forward.c#L251

tatokis, 2019-02-26:
I will definitely try that, thank you. That said, it still shouldn't trigger a bluescreen, and it doesn't seem to happen without that commit.

That aside, this sounds more like a "double free" issue (RemoveEntryList getting called twice on the same pointer), rather than memory corruption.
I believe if there was corruption, the fields wouldn't be null, but some random values, and there would also possibly be corruption that wouldn't let me view the struct fields inside WinDbg.

cezuni, 2019-02-26:
@tatokis : After inspecting a code, list_state NULL case was found. So list_state should be initialized when urbr is created. Could you try with the following diff? 
```
diff --git a/driver/vhci/usbreq.c b/driver/vhci/usbreq.c
index db5bb0c..508f187 100644
--- a/driver/vhci/usbreq.c
+++ b/driver/vhci/usbreq.c
@@ -161,6 +161,7 @@ create_urbr(pusbip_vpdo_dev_t vpdo, PIRP irp, unsigned long seq_num_unlink)
        urbr->vpdo = vpdo;
        urbr->irp = irp;
        urbr->seq_num_unlink = seq_num_unlink;
+       InitializeListHead(&urbr->list_state);
        return urbr;
 }
 
```

tatokis, 2019-02-26:
That fixes the issue, and the system no longer crashes on unlink. Thank you :)

commit 5bf18d1c (2019-02-27): Fix BSOD caused when a partially sent urbr is unlinked

This commit will resolve #14.
List entries of a newly created urbr had not been initialized.
They was expected to be setup when interted into a pending or
a sent list. However, there's a missing point. For safety,
list entries of a new urbr will be initialized at creation
time. In addition, free_urbr() is added for sanity check of a
being destroyed urbr.
closed 2019-02-26 by tatokis