cezanne/usbip-win #15: Disconnecting USBIP

yelfarri, 2019-02-20:
You stop the Linux USBIP host should it stop the windows client? By using the wire shark I noticed that in fact a Finish flag is been sent from the host to the windows client. The client does not do anything with flag. should it disconnect as well in windows?
Did anyone look at this ?

 Let Windows_client_ip be xxx.xxx.xxx.xxx
USBIP_Linux$ sudo tcpdump -i wlo1 host xxx.xxx.xxx.xxx
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlo1, link-type EN10MB (Ethernet), capture size 262144 bytes
15:04:08.797585 IP ZBook-15.3240 > xxx.xxx.xxx.xxx.50285: Flags [P.], seq 3259045765:3259045820, ack 2855156516, win 229, length 55
15:04:08.803557 IP xxx.xxx.xxx.xxx.50285 > ZBook-15.3240: Flags [P.], seq 1:49, ack 55, win 252, length 48
15:04:08.805504 IP ZBook-15.3240 > xxx.xxx.xxx.xxx.50285: Flags [P.], seq 55:110, ack 49, win 229, length 55
15:04:08.810277 IP xxx.xxx.xxx.xxx.50285 > ZBook-15.3240: Flags [P.], seq 49:97, ack 110, win 252, length 48
15:04:08.853446 IP ZBook-15.3240 > xxx.xxx.xxx.xxx.50285: Flags [.], ack 97, win 229, length 0

//Here a finish flag is been sent to windows client and the windows client answer with an acknowledgement
15:04:24.699197 IP ZBook-15.3240 > xxx.xxx.xxx.xxx.50285: Flags [F.], seq 110, ack 97, win 229, length 0
15:04:24.705279 IP xxx.xxx.xxx.xxx.50285 > ZBook-15.3240: Flags [.], ack 111, win 252, length 0

cezuni, 2019-02-20:
@yelfarri : Does a usbip client not clean up a usbip session if a server disconnects? I have not checked closely (or remembered). Your tcpdump confirms that a disconnection event is delivered to client side at TCP level.

tatokis, 2019-02-26:
I've been having the same problem, and commit https://github.com/cezuni/usbip-win/commit/aac9309a807c42e285424bebdd013ad5357ca098 fixes this for me.

yelfarri, 2019-03-04:
@cezuni not the client did not clean up if server disconnects, i had limited amount of time on hand lately sorry for the slow replay. I will look at the fixes done commited by @tatokis  to see if it solves my problem.
Thank you  both :)
 

thank you guys i will test your fix.

Oxalin, 2019-04-19:
@yelfarri is this fixed for you?

yelfarri, 2019-04-28:
@Oxalin @tatokis   and @cezuni thank you. I confirm that the bug is fixed. Now when disconnecting the usbip service the windows side terminates as well.

closed 2019-04-28 by yelfarri