cezanne/usbip-win #16: usbip client on windows7

zjl-1991, 2019-02-28:
usbip: error:  failed to record. out queue full.
but It can be used normally.

cezuni, 2019-02-28:
@zjl-1991 : If you see the error message, some usb responses may be dropped. It's weird that USB device still worked correctly even with such an error. What kind of USB device did you test? And could you test by increasing a queue length? Please refer to https://github.com/cezuni/usbip-win/issues/14#issuecomment-464990256

zjl-1991, 2019-03-03:
I use U-disk for testing. 
I tried to increase the queue length, but the usbip error still exist.

cezuni, 2019-03-03:
@zjl-1991 : USB disk had been tested successfully by many testers including me. Did you build vhci yourself or use binaries on [release page](https://github.com/cezuni/usbip-win/releases) ?

Oxalin, 2019-03-08:
@zjl-1991 Just to be sure so we can try to reproduce your setup, are you talking about a USB rotating disk, a USB flash disk or are you talking of a smartphone on which you turn on the UDISK feature? I'm asking because I've seen all these definitions for U-disk.

zjl-1991, 2019-03-08:
I tested it with a USB flash disk,  the vhci  I used was compiled on Win7.

Oxalin, 2019-03-09:
@zjl-1991 Can you give us the USB flash disk model please? I'd like to know if this is the same as the one reported in issue #20 : Kingston Technology : DataTraveler G4

cezuni, 2019-07-05:
@zjl-1991 : Have you ever checked a recent usbip-win?

cezuni, 2019-07-10:
I'll close, in hopes that this issue is resolved.

closed 2019-07-10 by cezuni