cezanne/usbip-win #17: X86 architecture is not supported

cantops, 2019-03-02:
Will there be any support plan in the future?

cantops, 2019-03-02:
[image]

cezuni, 2019-03-02:
@cantops : Currently there's no plan to support x86. 😢 Is your screenshot related to x86 environment ?

cantops, 2019-03-04:
@cezuni Yes, the driver files compiled with win32 can be installed correctly, but the above error occurred when running usbip！

Oxalin, 2019-03-04:
0x6f8 means ERROR_INVALID_USER_BUFFER

I'm pretty sure it is caused by a bad type under 32 bit when getting/setting the buffer size. I'll have a look at it in the next few days.

@cezuni sorry for being away these last few weeks, all the family got sick one after the other.

Oxalin, 2019-03-05:
@cezuni I may be wrong because I'm looking at it on the fly during my lunchtime, but I think the problem comes from using the wrong structure type in usbip-win/userspace/lib/usbip_setupdi.c, at line 115:
pdev_interface_detail->cbSize = sizeof(PSP_DEVICE_INTERFACE_DETAIL_DATA);

I think this should either be (they should be equivalent):
pdev_interface_detail->cbSize = sizeof(SP_DEVICE_INTERFACE_DETAIL_DATA);
or
pdev_interface_detail->cbSize = sizeof(*pdev_interface_detail);

What do you think of it?

cezuni, 2019-03-06:
@Oxalin : You're correct.

Oxalin, 2019-03-06:
@cantops Could you test this the referenced build commit? Hopefully, this will fix the error. From there, we should be able to determine if this is working on 32 bit systems or if there is more work to do.

I'll need sometime to install a 32 bit OS in a VM or similar, we could test it regularly from there.

cantops, 2019-03-08:
@Oxalin okay

cantops, 2019-03-08:
@Oxalin This build commit is correct and the problem has been solved.

closed 2019-03-08 by Oxalin