cezanne/usbip-win #20: usbip: error: failed to record. out queue full

cantops, 2019-03-08:
This error occurs when attach usb disk scrolls, but does not affect actual use。
https://github.com/cezuni/usbip-win/issues/16 does not solve the problem
[image]

Oxalin, 2019-03-08:
I'm pretty sure this is the same bug as #16 . Can you give us more info: Windows version and can you be more specific on your USB disk.

cantops, 2019-03-08:
@Oxalin 
usbip server: usbip (usbip-utils 2.0) on ubuntu, kernel version：4.18.0-15-generic
usbip drivers & client: The latest code pulled on git
windows: windows 10 x64 10.0.17134.1
usb disk: Kingston Technology : DataTraveler G4 (0951:1666)

ttuibk, 2019-04-12:
I have the same issue.  
usbip server:  usbip-server package 2.0-5 on OpenWrt-8devices Designated Driver v2.10 r49377
usbip drivers & client: The latest code pulled on git
Windows 10 x64 1809.
the usb device is an FTDI USB to serial converter used to drive an RS422 bus.

perlland, 2019-05-02:
Please, somebody has any solution for this issue?

cezuni, 2019-05-05:
@perlland @ttuibk : Default out queue length is 256. In normal case it is a sufficient number. But a usb disk with many files sometimes incurs queue full. So could you try with an empty USB disk? If you success, increasing a queue length may help as shown in https://github.com/cezuni/usbip-win/issues/14#issuecomment-464990256. Of course, current implementation should be improved to support dynamic queue length. 😢

cantops, 2019-05-08:
@Oxalin @cezuni 
Record_outq_seqnum and is_outq_seqnum should be 1:1, right?Is_outq_seqnum writes the bytes to the queue, and is_outq_seqnum is executed to clean up the queue after the acknowledgement is sent,  it will overflow if the queue is not cleaned up in time.

cantops, 2019-05-09:
[image]
I have tried to empty the USB disk and also tried the ukey. There is no overflow prompt. Increasing the queue length only prolongs the overflow time.
According to this guess, I made a simple print of the record_outq_seqnum and is_outq_seqnum calls. According to the print result, the same data is called twice to record_outq_seqnum (different number of loops), while is_outq_seqnum is called only once. Is this the main cause of overflow?

cezuni, 2019-05-09:
@cantops : 
> I have tried empty USB disk and ukey, both of which have overflow prompts. Increasing the queue length only prolongs the overflow time, so I think it should be caused by not cleaning up the queue properly.

I had suspected that your server did not respond while a client pushed usbip requests. 

> According to the print result, the same data is called twice to record_outq_seqnum

Well, it looks weird that record_outq_seqnum() resides in get_xfer_len(). If get_xfer_len() is called twice, record_outq_seqnum() is also called two times. So more inspections should be needed. Could you add more debugging logs for get_xfer_len()?

koudis, 2019-06-20:
I track the "usbip: error: failed to record. out queue full" baceuse I need USBIP on my server farm (so thanks for nice solution :) ).

The problem is that for some seqnums the record_outq_seqnum is called twice. Probably :).

Patch (with log entries) and log.txt attached.
After patch apply, I call ".\usbip.exe attach -r 192.168.0.129 -b 1-2 > log.txt" in powershell in admin mode.

[usbip-win-log.zip](https://github.com/cezuni/usbip-win/files/3311832/usbip-win-log.zip)

cezuni, 2019-06-22:
Thanks @koudis & @cantops . As you pointed out, record_outq_seqnum() can be called multiple times. This means that the same seq numbers are marked at out queue and result in queue full.

I create [patch_out_queue_full_fix.txt](https://github.com/cezuni/usbip-win/files/3316923/patch_out_queue_full_fix.txt) to resolve this issue. This patch is just a work-around, which avoids marking seq number if it is duplicated in record_outq_seqnum(). More desirable solution will be a hash table for bookkeeping outbound messages.

I have checked that a patched usbip works good for my usb bluetooth dongle. If anyone confirms, I'll merge the patch into master branch.

koudis, 2019-06-23:
@cezuni 
### test
I tested it on my Flash Storage. It seems to work without problems.

### Hash Table

Is there any Hash table implementation in the project? (I cannot find any useful for this purpose).
Or the Hash Table must be implemented first?

cezuni, 2019-06-23:
@koudis :
> I tested it on my Flash Storage. It seems to work without problems. (from Debian Stretch to Windows 10 Home)

Good. I'll merge the patch.

> Is there any Hash table implementation in the project? (I cannot find any useful for this purpose).
Or the Hash Table must be implemented first?

Hash table should be implemented first. Well, it would be better that simple one is imported into this project.

closed 2019-06-24 by cezuni

koudis, 2019-06-24:
@cezuni 
I think that Hash Table is not necessary because it non-trivialy increases complexity of the impl. (another library, another non trivial code)
Will Has Table be used in another place except to record_outq_seqnum...?

If Yes the solution below is not good :D.

Is there any reason why is the Queue 256 items long? Can we decrease the number of Queue item to 128 (at least)?
If Yes we can create another Queue with length N, where N <= 128. Items in the new Queue will be "unmarked" seqnums. We just move seqnums from one queue to another.
The Complexity of the operation remains.

cezuni, 2019-06-25:
@koudis : 
> Is there any reason why is the Queue 256 items long? Can we decrease the number of Queue item to 128 (at least)?

I remembered that queue implementation and 256 queue length were determined by old repository. Optimal queue length is hard to guess. We just hope that 256 is sufficiently large enough. Smaller number than 256 would be fine in a normal case.

> If Yes we can create another Queue with length N, where N <= 128. Items in the new Queue will be "unmarked" seqnums. We just move seqnums from one queue to another.
The Complexity of the operation remains.

Multiple queue is interesting!! But every queue should be checked when a response message arrives. Besides previously addressed implementations, linked list may be a good candidate.

commit 2e0fae5d (2019-06-24): fix out queue full problem

record_outq_seqnum() might be called multiple times. This case seems to
happen more frequently under slow network. As the same sequence numbers
were flooded into out queue slots, out queue became full.
Quick remedy to this problem avoids marking duplicated sequence numbers.
This commit resolves #20 and #16.