cezanne/usbip-win #22: mass storage device is not working

chenfy1985, 2019-03-08:
USB mouse and keyboard are working well,
but it seems not working with my kingston USB Flash 
the error code is 10

"This device cannot start. Try upgrading the device drivers for this device. (Code 10)"

Oxalin, 2019-03-08:
@chenfy1985 Would you be able to give us a more complete log?

chenfy1985, 2019-03-21:
I have the same problem as https://github.com/cezuni/usbip-win/issues/7#issuecomment-454737045,
I tried to modify the code , but  the device is still not working.

chenfy1985, 2019-04-18:
Reason：
1. the stub driver detect a err code of STALL_PID   
2. vhci_write.c:process_urb_res_submit catch the error and return STATUS_UNSUCCESSFUL
3. then the vhci driver will receive IRP_MN_REMOVE_DEVICE 

My solution:
1. vhci_write.c:process_urb_res_submit ,  return the STALL_PID to urb->UrbHeader.Status
2. then the vhci driver will receive URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL
3. forward it to usb server side

now it seems working well, but bsod occurred at usb server side

cezuni, 2019-04-18:
@chenfy1985 : Current vhci will act as you describe in reason paragraph.
As previously noted( https://github.com/cezuni/usbip-win/issues/7#issuecomment-454801897 ), I am not sure that returning UNSUCCESSFUL is correct. Can you share your diff ?

chenfy1985, 2019-04-18:
1.returning UNSUCCESSFUL is correct and must set urb->UrbHeader.Status = USBD_STATUS_STALL_PID,
then vhci driver can receive the request of URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL 
2. and vhci do not handle URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL , so just forward it 
3. and stub need to call reset_pipe 

 I'll upload diffs later

cezuni, 2019-07-23:
@chenfy1985 : Sorry for too late. I rebased your awesome work and merged into [fix_stall_multi_conf](https://github.com/cezuni/usbip-win/tree/fix_stall_multi_conf) branch. I slightly modified your code for beautifying. I've checked with my three USB devices. You had also fixed the transient code to always set to 1 when selecting configuration. Maybe af07a343c8719a4a32d27baa81fe7bd6745c977c solves multi configuration device problem such as #25, #30. Anyway, thanks!

cezuni, 2019-07-26:
@chenfy1985 : your work has been merged into master branch. check 781e315a142d7e5454875213c422a0cf7df9f86f  fix_stall_multi_conf branch is removed.

wrenchonline, 2019-08-05:
Hello, long time no see，It seems that the device changed a PID during the second startup.

cezuni, 2019-08-05:
@wrenchonline : Good to see you~ 
> It seems that the device changed a PID during the second startup.

Why don't you create a new issue? If the such kind of a problem exists, it does not seem to be related to this issue.

closed 2019-08-08 by cezuni