cezanne/usbip-win #27: run project windows Error

reclock, 2019-04-18:
I run this project:
server: Windows7
usbip.exe bind -b 1-136   //success
usbipd.exe -b -4               //success

client: Windows10
usbip.exe attach -r 192.168.14.129 -b 1-136 //success
[image]

But the USB device I saw in the human interface device was abnormal.
[image]

What's the problem?

cezuni, 2019-04-18:
@reclock : Something goes wrong when a vhci driver sets up a virtual usbip device. But it's impossible to resolve your problem without windows kernel logs. Could you share kernel logs?

reclock, 2019-04-18:
[image]

what`s the problem?  i change one computer

reclock, 2019-04-18:
I use win10 as  server, use win7 as client,   run success, thanks.

reclock, 2019-04-26:
I want to implement a local virtual USB device based on this project. I don't need to insert a USB device into the server. Do you know of any other similar open source projects? thanks.

cezuni, 2019-04-26:
@reclock : How about libusb-win32?
Anyway, your issue has been fulfilled. I'll close it.

closed 2019-04-26 by cezuni