cezanne/usbip-win #30: Xbox one controller (wired or wireless) doesn't work

Bikkies, 2019-04-21:
I've been able to successfully get other devices like webcams and flash drives happening over usbip, but can't manage to get an Xbox One controller working at all.

I'm using an Orange Pi Zero with Armbian as my host, and it can successfully bind either the controller, or the USB wireless adapter.

Then the problems start with the Windows 10 side of things. With the wireless adapter I will either get a blue screen with kernel security check failure, or I will get an error of "failed to read: err:6". I think one time it didn't give me either of those, but nothing showed up in device manager, so I don't think it actually worked.

With the controller connected via a usb cable to the host, it seems to bind and attach fine, but then device manager gives me this:
[image]

Are there any known issues with trying to get Xbox One controllers happening? Anything else I can try to get it working?

cezuni, 2019-04-22:
@Bikkies : This issue looks similar to #25. If your web cam and flash drive had worked successfully, armbian host is not the issue. In #25, I had suspected that a usbip host had a mismatched version but my guess may be wrong. At present state, VHCI driver does not seem to handle the controller. Can you share your Win10 kernel logs? 

Well, in my personal curiosity, is your interesting usbip configuration for wireless Xbox one controller?

Oxalin, 2019-05-01:
I'll ask around if someone has a Xbox one controller that I could barrow for testing purpose.

cezuni, 2019-07-26:
Duplicate with #25. I'll close it. We can discuss in #25.

closed 2019-07-26 by cezuni