cezanne/usbip-win #31: Question: Is there a command to show imported devices?

builker, 2019-04-30:
On Ubuntu 18.04:
```
$ usbip help
usage: usbip [--debug] [--log] [--tcp-port PORT] [version]
             [help] <command> <args>

  attach     Attach a remote USB device
  detach     Detach a remote USB device
  list       List exportable or local USB devices
  bind       Bind device to usbip-host.ko
  unbind     Unbind device from usbip-host.ko
  port       Show imported USB devices

$ sudo usbip port
Imported USB devices
====================
Port 00: <Port in Use> at Low Speed(1.5Mbps)
       Dell Computer Corp. : Mouse (413c:3200)
       3-1 -> usbip://192.168.1.1:3240/1-1.2
           -> remote bus/dev 001/004
```
Is there a similar command like `port`?

cezuni, 2019-04-30:
@builker : Sadly, NO. usbip-win userspace code should catch up up-to-date linux usbip tool.

Oxalin, 2019-05-01:
I've been working on it in the last few weeks. I had to take care of my tax report, so this was on hold, but I'll resume my work shortly. The new code is mostly inspired by the Linux USBIP tool. I also have some commits to submit to stabilize usbipd. This should be pushed in the next few days.

builker, 2019-05-03:
Thank you, @Oxalin , for your efforts!

Oxalin, 2019-07-03:
Sorry, I've been away from this project for a few weeks. I was not feeling well and had no energy to work on this. Back at work. I'll keep you updated.

Oxalin, 2019-08-05:
@cezuni Yes. I'm still figuring how to get some details, but it's progressing. I'll continue my work on it this week (that's why I was cleaning some commits that I pushed over last weekend). I may need a hand though, I'll let you know.

juju812, 2020-09-22:
Hi, @Oxalin , is there any update for this issue? 
We've tried to detach dedicated USB device based on bus id, but without `port` command we can't establish mapping between bus id and port.

cezanne, 2020-12-28:
Port command is finally implemented by 3045429235467f2785a9036d5a6b2437dac0680f.

closed 2020-12-28 by cezanne