cezanne/usbip-win #32: usbip.exe bind -b x-xx

zhLoser01, 2019-04-30:
win7×64
usbip:error:cannot load certificate:subject: USBIP Test

zhLoser01, 2019-04-30:
- 
[image]

-

zhLoser01, 2019-04-30:
[image]

cezuni, 2019-04-30:
@zhLoser01 : This error means that the certificate cannot be found. Did you install the certificate on local machine registry? (not current user registry) If not, please refer to [this link](https://docs.microsoft.com/ko-kr/windows-hardware/drivers/install/using-certmgr-to-install-test-certificates-on-a-test-computer).

zhLoser01, 2019-04-30:
[image]
I can find the header file

zhLoser01, 2019-04-30:
ok

zhLoser01, 2019-05-05:
[image]

zhLoser01, 2019-05-06:
@cezuni I compiled it in my vs2017+win10 environment, and then ported it to the win7 environment, reporting this error "usbip:error:SignerSignEx failed.hResult #80092006"
"usbip:error:failed to bind device"

cezuni, 2019-05-07:
@zhLoser01 : As shown in readme, USBIP Test certificate should be installed into 2 certicate stores: "Trusted Root Certification Authority" and "Trusted Publishers" on local machine (not current user). You can check with certlm.msc. I had checked that bind process generates "error: cannot load certificate: USBIP Test" if a certificate is not installed into "Trusted Root Certification Authority".  When a test certificate is not installed into "Trusted Publishers", "error: failed to update driver: e0000242" message is shown. I have no idea about your case. 😢

zhLoser01, 2019-05-16:
@cezuni My camera and headphones are not recognized by the driver in windows7

cezuni, 2019-07-10:
@zhLoser01 : I'm too late?? :smile: 
Please tell your story about signing problem.. Is it solved?

 > My camera and headphones are not recognized by the driver in windows7

Maybe #7 will be a solution.

cezuni, 2019-07-24:
Close the old issue. If you( @zhLoser01 ) need, reopen it.

closed 2019-07-24 by cezuni

andrewzli, 2020-07-22:
if anyone does have this issue in the future, try running cmd as administrator, that worked for me!