cezanne/usbip-win #33: Handle USBIP Client interrupts

Oxalin, 2019-05-06:
According to the following bug report, an attached device will not be freed if the connection between a client and a server is abruptly interrupted.

From here: https://sourceforge.net/p/usbip/discussion/418508/thread/34da2931c0/?limit=25#13b9

Under usbip_network.c > usbip_net_tcp_connect(), there is a TODO note stating "/* TODO: write code for heartbeat */".

yelfarri, 2019-05-09:
@Oxalin the keepalive (heartbeat)  was completed I see the keepalive message when using Wireshark. Isn't keepAlive a part of the TCPIP and not unique to the open socket? 

[image]

Oxalin, 2019-05-09:
It can be set system wide and per application / socket. From my quick overlook at usbipd, the application is not setting any customized Keep-Alive options, thus relying on the system's default values.

yelfarri, 2019-05-09:
I understand but i m takling about usbip client side. I Found in code that usbip has a keepalive function call. If I remember correctly this call does not set time and intervals. It might be using system defaul values

Oxalin, 2019-05-17:
Yes, indeed. setsockopt() could be called to set the IPPROTO_TCP socket options (as described here https://docs.microsoft.com/en-us/windows/desktop/WinSock/ipproto-tcp-socket-options).However, many options are only available under Windows 10 (1709 and up), even though there are commonly available under Linux...

Thus, I would be tempted to add a heartbeat (keepalive) at the application level. This way, we will be able to support a decent heartbeat in any Windows version. That's why the usbip_net_set_keepalive() doesn't set any IPPROTO_TCP socket options: it was supposed to be added under usbip_net_tcp_connect(). See line 259.

cezuni, 2019-07-26:
@Oxalin : I agree that application-level heartbeat can help to detect abortive disconnection more quickly. But it's impossible without usbip protocol modification. I think that it's a too big task to resolve at usbip-win project level.

cezanne, 2021-01-13:
Socket-specific KeepAlive is implemented by be88d56c4d740d453436777622defe0583098910. Application-level heartbeat might be not useful unless a linux implementation follows.

closed 2021-01-13 by cezanne