cezanne/usbip-win #34: Hang on attach

M0CRT, 2019-05-06:
Hi.  First, thank you for this great work.  I've been working on this all weekend and today to get this working; really appreciate you help.

Running the 2.0 USBIP tools from Debian 9 (not the buggy version).  All works great re the setup on the Debian host and I can share the devices.

From Windows I can see the devices with the correct USB.IDS.  Looks fine.

Latest 0.4 version and have installed the driver correctly after cert install and dropping the Windows 10 x64 host into 'Test' Mode.  Driver installs with no errors.

On attach I get pop-ups saying that USBIP is being setup and then it mentions that it's all ready.  I don't get a command prompt back and I have no messages relating the setup of the actual device nor any new device appearing in the Device Manager.

Checking on the USBIP on the Debian host, I see working responses from the server with 'received request' 0x8005 (5) on the listing remotely and the connection request which responds with a 0x8003(5) complete.

Any ideas?  I'm going mad! ;-)

Mo

cezuni, 2019-05-07:
> On attach I get pop-ups saying that USBIP is being setup and then it mentions that it's all ready. I don't get a command prompt back and I have no messages relating the setup of the actual device nor any new device appearing in the Device Manager.

Unlike linux version, attach of usbip-win is blocking because processing protocol is done in userspace without a dedicated daemon. This behavior should be improved by using kernel daemon. Anyway, there seems to be an error if no new device is appeared in device manager. 

> Checking on the USBIP on the Debian host, I see working responses from the server with 'received request' 0x8005 (5) on the listing remotely and the connection request which responds with a 0x8003(5) complete.

usbipd output looks good.

M0CRT, 2019-05-07:
Cezuni, thanks for the reply.  Good to know the USBIPD looks ok.  Is there something I should be trying?

Thanks

Mo

cezuni, 2019-05-08:
@M0CRT : What is your USB device? And can you obtain and share linux and windows kernel logs? Enabling userland log is also possible with a slight modification. Please refer  https://github.com/cezuni/usbip-win/issues/6#issuecomment-455079893

cezuni, 2019-07-24:
It's a too old issue. Close it.
@M0CRT : If  you need, please reopen it.

closed 2019-07-24 by cezuni