cezanne/usbip-win #35: TCP/IP keepAlive time and interval

yelfarri, 2019-05-09:
From the USBIP application, I see that keep alive is implemented. When testing with wireshark I found out that KeepAlive packets send after 2 hours than an interval of 1 second. From my understanding, this is a default configuration in windows 10. One can modify the keepAlive default values by modifying/adding two new parameters to Tcpip.

Open "Registry editor" navigate to the following directory
Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\Tcpip\Parameters

Add  DWORD(32bits) value with name KeepAliveInterval set desired in value data (milliseconds)
Add  DWORD(32bits) value with name KeepAliveTime set desired in value data (milliseconds)

My question is it possible to set keepalive time and interval from the application ?

Oxalin, 2019-05-09:
Indeed, it is using Windows' default keepalive. I've been looking at it.

Yes, it is possible to configure the keepalive parameters in the app. I'll be testing some changes in the next few days. I'll keep this post updated.

Oxalin, 2019-05-09:
By the way, this is a duplicate of #33 I opened a few days.

closed 2019-05-09 by Oxalin