cezanne/usbip-win #37: interface claim failed: LIBUSB_ERROR_NOT_FOUND

CrisBSD, 2019-05-23:
Hi, I'm testing a fingerprint reader (U.are.U® 4500 Fingerprint Reader). I have as client a linux machine with kernel 4.15, but when I try to open the device I get: 

journal: interface claim failed: LIBUSB_ERROR_NOT_FOUND

Am I forgetting some configuration?

cezuni, 2019-05-27:
@CrisBSD : Have you ever installed [libusb-win](https://github.com/libusb/libusb/wiki/Windows) ? usbip-win does not present such an error.

Oxalin, 2019-07-03:
@CrisBSD Am I understanding correctly that you are experiencing this error on a Linux machine? If so, you are not reporting your bug at the right place. Otherwise, please be more explicit on your setup.

cezuni, 2019-07-10:
@CrisBSD : Please reopen it when you need.  Close the old issue silently.

closed 2019-07-10 by cezuni