cezanne/usbip-win #38: usb camera headset The driver can't be installed

zhLoser01, 2019-05-27:


cezuni, 2019-06-06:
@zhLoser01 : Can you provide more information ?

zhLoser01, 2019-06-10:
@cezuni  Win7 client camera and headset can not be used,Can't find camera and headset driver in windows7 client

zhLoser01, 2019-06-10:
However, for devices with many endpoints, such as usb camera, usb headset, etc., although it can bind and usbip.exe -a ip 3-6, but the driver under Windows can not be installed, you can see under the device manager. Usb Device Over IP" is an exclamation point. I don't know how to drive such devices

zhLoser01, 2019-06-10:
@cezuni

zhLoser01, 2019-06-10:
[image]

zhLoser01, 2019-06-10:
this is connecting to a webcam device, error.

zhLoser01, 2019-06-11:
[image]
Currently, this hardware device is not connected to the computer. (Code 45)

To correct this problem, reconnect this hardware device to your computer.
@cezuni

cezuni, 2019-06-22:
@zhLoser01 : Your error in screenshot seems to be re-attaching. The first usbip attach command was working correctly even though you typed Ctrl+C . You should note that attach command in usbip-win executes continuously until a user stops explicitly. This is an usbip-win limitation. Linux usbip attach command completes shortly and gives a prompt. Anyway, re-attaching failure should be checked by me or any volunteer.

cezuni, 2019-07-26:
I checked that reattaching works. Close this issue.

closed 2019-07-26 by cezuni