cezanne/usbip-win #40: Build with Release configuration

koudis, 2019-06-20:
Hello all,

I use USBIP client on windows. It works quite fine (except some errors with queue overflow etc.).

I Try to compile USBIP by Release configuration. It fails. Mainly because mishmash between debug and release header/libraries.

My question is - is this behaviour intentionall ot it§s a bug. If it's a Bug, may I create PR with repairs? Or it's the subject of further/actual development.

In case of PR - from which branch I can create a PR?

cezuni, 2019-06-22:
@koudis : Release configuration has not been a main concern. But you are welcome to improve it. master branch would be good for PR.

koudis, 2019-06-23:
ok, thanks for reply. If the Release configuration is not a main concern I will not create PR for this now (it just complicate development).

But I try to create PR with Toolset update. Just for "make all dependencies actual".

cezuni, 2019-07-10:
Current usbip-win is able to build with release configuration. (refer #42)

closed 2019-07-10 by cezuni