cezanne/usbip-win #46: Libusb Compatibility (Windows 7)

amylizzle, 2019-07-07:
I have a USB Composite Device connected on a Linux box and using USBIP to connect it to a Win7 machine. Everything seems to working as intended, except the libusb's `libusb_get_device_list() ` is unable to detect the presence of the USB device. 

The research that I've done suggests that might be because the VHCI driver doesn't register a USB Root Hub, and libusb doesn't know to look for usb devices that aren't connected to a root hub.

Is it possible to make a change to the VHCI driver such that it creates a USB Root Hub and attaches the remote devices to that created hub?

cezuni, 2019-07-08:
@amylizzle : I'm not aware of that a usbip-win device is not accessible via libusb(maybe libusb-win?). It would be nice that libusb can detect usbip-win devices.

> Is it possible to make a change to the VHCI driver such that it creates a USB Root Hub and attaches the remote devices to that created hub?

In my opinion, it's a possible story but not a simple job. Are you ready to be a pioneer? 😄

amylizzle, 2019-07-08:
I was afraid you'd ask that :P I already took a look at the docs and managed to do little but confuse myself, but if I do get it going I'll post a pull request.

amylizzle, 2019-07-08:
I have literally no idea what I'm doing. I'm not even 100% sure I understand how data gets from the network to the driver. I'm afraid this is probably going to have to be someone else's project :(

cezuni, 2019-07-08:
@amylizzle : 
> I'm not even 100% sure I understand how data gets from the network to the driver.

Do you make a windows kernel driver to receive/send data directly from network ? Currently, a vhci/stub receives data by file I/O. Network stuff is done by userland. It's very an inefficient architecture and one of the features I want to improve.

amylizzle, 2019-07-08:
Well, it's good to know I was at least on the right track for figuring that part. Userspace and network stuff is definitely more my wheelhouse, so I could take a look. Driver dev is entirely new to me though.

cezuni, 2019-07-26:
@amylizzle : Do you have a plan to work on libusb support ? I had looked over libusb code but it's too complicated. :cry: Libusb support is not easy one.

If no one tries to follow up this issue, I'll close it. Although it is to be closed, it is marked as "future work" label and thus, any volunteer can easily reopen it.

closed 2019-08-16 by cezuni