cezanne/usbip-win #48: USBIP Windows - install VHCI driver programatically

koudis, 2019-07-10:
I want to install VHCI driver programatically. I try many ways but none works.

One usable:
https://stackoverflow.com/questions/11659223/install-the-libusb-driver-using-the-inf-file-with-c-sharp

I recognized that the problem is not in driver but with "How to install devices programatically".

How can I install USBIP VHCI driver programatically on windows machine? Is there any clean ways?

Thanks a lot for any advise.

cezuni, 2019-07-11:
@koudis: It's wonderful that vhci is ready for use with a simple usbip command. Current stub driver is installed by usbip.exe.
BTW, SetupCopyOEMInfA cannot install a legacy device?

koudis, 2019-07-12:
@cezuni thanks, I will look at it :).

koudis, 2019-07-12:
@cezuni 
SetupCopyOEMInfA install driver but not device. The device is not present in Device Manager and usbip attach command does not work.

I call SetupCopyOEMInfA exactly as in StackOverflow link. Function returns true but if I want to attach device by "usbip attach" comman i get
    usbip: error: open vhci driver
    usbip: error: failed to read: err:6

But if I install driver manually "usbip attach" works fine. I Guess that the OEM function installs only driver but not device.

koudis, 2019-07-13:
@cezuni 
I tray this
https://stackoverflow.com/questions/53864109/windows-driver-install-legacy-hardware-with-command-line-tool

I try every section I found in the inf file. But It does not work...

koudis, 2019-07-13:
I guess that there are some missing sections in .inf file.

It has no priority but It will be really nice install driver without problems  from C/C++ or C#.

The Device Manager probably parse the inf file and go thru all sections, make some heuristics and determine Device info. Then it installs the devicen and driver.
We have no sources for Device Manager so it's really hard to track exact steps which Device Manager do while installs Device.

cezuni, 2019-07-14:
@koudis: I'll try too but I will be busy next week. 😞 My testing will be late..

koudis, 2019-07-14:
@cezuni  thank you :).

I am doing some research around SetupDi functions.
I found https://docs.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupdicreatedeviceinterfacea

But I really do not known what is "interface_class_guid"? I found macros GUID_DEVICEINTERFACE* but I do not known what I may choose.

koudis, 2019-07-14:
I guess that i found the way. I will test it and then post a snippet.

koudis, 2019-07-15:
@cezuni 
Solution with the code:
https://github.com/koudis/usbip-win-vhci-install

Just change "INF_FILE" to absolute path to your vhci .inf file.

cezuni, 2019-07-16:
@koudis: superb!! I'll check your work.
How about integrating the awesome feature into usbip.exe ?

koudis, 2019-07-17:
@cezuni 
ok :).

How the vhci installer is supposed to be integrated into usbipwin? As part of usbip command? As separate utility which installs given inf? Or direct installer for vhci driver (driver packed as resource or present in the same dir as installer)?

I will make small refactoring and try to remove "PlugAndPlay" driver install routine.

koudis, 2019-07-24:
@cezuni 
I vote for "Separate installer which has all resources present in the same dir as installer".

I Create PR with new project for vhci installer. VS Solution has project called "Driver package" which copies all linked drivers to specified directory and permit to compile own C code.

So I create Driver Package project, implement installer and create PR.

(I have not succeed with replacement  of "PlugandPlay" install driver function. I guess that install by PnP is not so bad. Any advise welcome.)

cezuni, 2019-07-24:
> I vote for "Separate installer which has all resources present in the same dir as installer".

I slightly prefer usbip.exe. :) But welcome any approach. 

> (I have not succeed with replacement of "PlugandPlay" install driver function. I guess that install by PnP is not so bad. Any advise welcome.)

What’s the reason to replace UpdateDriverForPlugAndPlayDevices?

koudis, 2019-07-25:
> I slightly prefer usbip.exe. :) But welcome any approach.

Ok, i will try.

prerequisite: Driver files will have to be in the same dir as usbip.exe

> What’s the reason to replace UpdateDriverForPlugAndPlayDevices?

It installs driver for all already inserted/created devices. Not for only specific which I created seconds before. It's not a big problem If the usbip installs driver (because when usbip installs driver It can choose Device Path and HW Id so there will be no collisions)

cezuni, 2019-07-25:
> prerequisite: Driver files will have to be in the same dir as usbip.exe

I also agree that it's the best way for the current state where an installation package is not ready.

koudis, 2019-08-02:
@cezuni I guess that the install procedure can be done in usbip_vhci_driver_open function? If open fails we try to install usbip_vhci driver (according to error code value)

cezuni, 2019-08-02:
@koudis: interesting idea! But explicit installation with an usbip command would be better as a first step. Implicit installation for convenience can be next.

koudis, 2019-09-25:
I am going to work on install feature in start of the next month (2019/10).
I guesst that it's work for a few days, so it can be done in first half of 2019/10

koudis, 2019-12-13:
@cezuni I created #86 . Can you test code in PR?
You need "trust" driver before you can install it by `usbip.exe install`

koudis, 2020-01-04:
@cezuni I closed original PR and create another one

https://github.com/cezanne/usbip-win/pull/90

cezanne, 2020-03-14:
@koudis: This issue had been successfully resolved. 😄

closed 2020-03-14 by cezanne