cezanne/usbip-win #50: USPIP.exe is reporting as infected

elamast, 2019-07-15:
Hello.  The usbip.exe file in the 0.0.5 release is being reported as virus infected by Windows Defender and several others on VirusTotal:  https://www.virustotal.com/gui/file/c39191db245494c59269b5d12e4136bfe9a33b40e9f99d0bc2cb1431d6a01f4e/detection

cezuni, 2019-07-16:
@elamast : Thanks! 
VirusTotal reports that all exe binaries in 0.0.5 and 0.0.3 have been infected. I'll check my windows dev box.

cezuni, 2019-07-16:
@elamast : Executable built by VS is often incorrectly detected as virus. I had found some [stackoverflow links](https://stackoverflow.com/questions/38573708/vs2015-executable-become-virus-with-potential-solution-but-dont-know-why). Is this our case?

cezuni, 2019-07-26:
@elamast : I checked that recent release binaries(v0.0.6) are clean. As all usbip-win release binaries have been built in the same machine, my machine seems to be not infected with some viruses. As commented previously, I suspect that virus alert was caused by an invalid virus detection logic.

cezuni, 2019-07-31:
@elmast: I'll close it. Please reopen if you want.

mahmoudajawad, 2020-04-02:
@cezuni, the last published binaries are reported as infected with a trojan:
```Trojan:Win32/Wacatac.D!ml```

cezanne, 2020-04-02:
@mahmoudajawad : 

> the last published binaries are reported as infected with a trojan:
Trojan:Win32/Wacatac.D!ml

Which file? Binaries in v0.0.9 are checked as no problem from the upper web site.

mahmoudajawad, 2020-04-03:
Here's the report from Windows Defender:
[image]

It's wortth noting that Edge (chromium) refused to download the files initially as they were reported suspicious by SmartScreen. But, I ignored at forced downloaded the files. However, Windows Defender instantly removed them from my Downloads folder.

cezanne, 2020-04-04:
@mahmoudajawad : weird.. I have checked a downloaded usbip.exe with a recent virus checker but there's no virus reported.

mahmoudajawad, 2020-04-04:
We are speaking about last published binaries, right?

elamast, 2020-04-04:
Have you run it through VirusTotal?

Sent from MailDroid

mahmoudajawad, 2020-04-04:
No. As I mentioned, it was deleted by Windows Defender automatically without even me scanning the files manually.

cezanne, 2020-04-04:
> We are speaking about last published binaries, right?

@mahmoudajawad : Yes. https://github.com/cezanne/usbip-win/releases/download/v0.0.9/usbip.exe

This is [virus total result](https://www.virustotal.com/gui/url/2cff0d6e86241cb230b0e830e999f2e222835dd6cd8789702a0b6b54bdca6a0f/detection) for usbip.exe as @elamast suggested.

mahmoudajawad, 2020-04-05:
Hmm. This must've been some false positive. I just tried downloading the files again using Firefox this time and things went fine. It's not clear to me at all why Windows Defender decided the files were infected with some trojan first time. Thanks for your time on this, @cezanne

closed 2020-04-05 by cezanne

Jackfritt, 2022-09-29:
Latest Release usbip-win-0.3.6-dev.zip also get reported by virustotal. 
I think its a false positive. What do you think?

https://www.virustotal.com/gui/file/489daccb3ddffb1ec83ab6e0a4f56a8b13a9af34ca4f9bf445357053aad46dc2

https://www.virustotal.com/gui/file/5a57753f17bfc6436708fbaac22b1e324f178b25b505111819989eefe936d281/behavior
1 match for rule Failed Code Integrity Checks by Thomas Patzke from Sigma Integrated 
1 match for rule Use Remove-Item to Delete File by frack113 from Sigma Integrated Rule 

https://www.virustotal.com/gui/file/8d0fd393dc9e8aaa1722c51190ebb872089867b0c34aa6b86dfcc69af87f153a/details
A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.