cezanne/usbip-win #52: Virtual Com port fail. This device cannot start. (Code 10)

MarkOstis, 2019-07-23:
Hi Guys. I am trying to emulate the STMicroelectronics Virtual Com Port via the usbip vhci. 
As result I have net issue:
[image]

The USBIP/VHCI driver was installed properly:
[image]

Here is a log from the usbip.exe:
```
D:\Projects\USBIP\usbip-win\Debug\output_copy3>usbip.exe --debug attach -r localhost -b 1-1
usbip: debug: d:\projects\usbip\usbip-win\userspace\src\usbip\usbip.c:133:[run_command] running command: `attach'

usbip: debug: d:\projects\usbip\usbip-win\userspace\src\usbip\usbip_vhci.c:58:[usbip_vhci_driver_open] device path: \\?\root#unknown#0000#{d35f7840-6a0c-11d2-b841-00c04fad5171}
usbip: debug: d:\projects\usbip\usbip-win\userspace\src\usbip\usbip_attach.c:57:[import_device] got free port 1
```

As a device that emulate device I use project https://github.com/lcgamboa/USBIP-Virtual-USB-Device. I run cdc-acm and get this:

```
cdc started....
Connection address:127.0.0.1
Header Packet
command: 0x8003
attach device
------------------------------------------------
handles requests
usbip cmd 1
usbip seqnum 2147483647
usbip devid 65538
usbip direction 1
usbip ep 0
usbip flags 0
usbip number of packets 0
usbip interval 0
usbip setup 9225060895304978944
usbip buffer lenght  18
#control requests
8006000200001200
  UC Request Type 128
  UC Request 6
  UC Value  2[0]
  UCIndex  0-0
  UC Length 18
handle_get_descriptor 2 [0]
Configuration
------------------------------------------------
handles requests
usbip cmd 1
usbip seqnum 1
usbip devid 65538
usbip direction 1
usbip ep 0
usbip flags 0
usbip number of packets 0
usbip interval 0
usbip setup 9225060891010011648
usbip buffer lenght  18
#control requests
8006000100001200
  UC Request Type 128
  UC Request 6
  UC Value  1[0]
  UCIndex  0-0
  UC Length 18
handle_get_descriptor 1 [0]
Device
------------------------------------------------
handles requests
usbip cmd 1
usbip seqnum 2
usbip devid 65538
usbip direction 1
usbip ep 0
usbip flags 0
usbip number of packets 0
usbip interval 0
usbip setup 9225060895304976640
usbip buffer lenght  9
#control requests
8006000200000900
  UC Request Type 128
  UC Request 6
  UC Value  2[0]
  UCIndex  0-0
  UC Length 9
handle_get_descriptor 2 [0]
Configuration
------------------------------------------------
handles requests
usbip cmd 1
usbip seqnum 3
usbip devid 65538
usbip direction 1
usbip ep 0
usbip flags 0
usbip number of packets 0
usbip interval 0
usbip setup 9225060895304991488
usbip buffer lenght  67
#control requests
8006000200004300
  UC Request Type 128
  UC Request 6
  UC Value  2[0]
  UCIndex  0-0
  UC Length 67
handle_get_descriptor 2 [0]
Configuration
------------------------------------------------
handles requests
usbip cmd 1
usbip seqnum 4
usbip devid 65538
usbip direction 1
usbip ep 0
usbip flags 0
usbip number of packets 0
usbip interval 0
usbip setup 9225060895304976641
usbip buffer lenght  265
#control requests
8006000200000901
  UC Request Type 128
  UC Request 6
  UC Value  2[0]
  UCIndex  0-0
  UC Length 265
handle_get_descriptor 2 [0]
Configuration
------------------------------------------------
handles requests
usbip cmd 1
usbip seqnum 5
usbip devid 65538
usbip direction 0
usbip ep 0
usbip flags 1
usbip number of packets 0
usbip interval 0
usbip setup 2534374302023680
usbip buffer lenght  0
#control requests
0009010000000000
  UC Request Type 0
  UC Request 9
  UC Value  0[1]
  UCIndex  0-0
  UC Length 0
handle_set_configuration 0[1]
------------------------------------------------
handles requests
usbip cmd 1
usbip seqnum 6
usbip devid 65538
usbip direction 0
usbip ep 0
usbip flags 1
usbip number of packets 0
usbip interval 0
usbip setup 75153818798522368
usbip buffer lenght  0
#control requests
010B000001000000
  UC Request Type 1
  UC Request 11
  UC Value  0[0]
  UCIndex  0-1
  UC Length 0
SET_INTERFACE
------------------------------------------------
handles requests

```
Please help.

cezuni, 2019-07-23:
@MarkOstis : What version did you test ? More detailed logs are required to resolve the issue. Please read the tail part of README.md. I have just added the information about how to generate usbip-win logs.

MarkOstis, 2019-07-24:
> @MarkOstis : What version did you test ? More detailed logs are required to resolve the issue. Please read the tail part of README.md. I have just added the information about how to generate usbip-win logs.

I am using commit 59c09290.
OS: Windows 10 Version 1809 (OS Build 17763.557)
The project was built with the VS 2017 (v141), Windows SDK Version: 10.0.17763.0.

Here is the usbip.exe log file:
[debug_pdu.log](https://github.com/cezuni/usbip-win/files/3425431/debug_pdu.log)

Note, I also tried to emulate hid-mouse and hid-keyboard devices. The device attached/detached successfully without errors. 

One more thing, I was not able to sign USB/IP VHCI driver. To be able to install the driver I temporary disabled the driver signature checks in Windows 10.

cezuni, 2019-07-24:
> I am using commit 59c0929.

Please try more recent one. I just have upload [new binary release](https://github.com/cezuni/usbip-win/releases/tag/v0.0.6). Please check..

> debug_pdu.log

In this case, debug_pdu.log does not say useful things. Can you share windows kernel log?

> One more thing, I was not able to sign USB/IP VHCI driver. To be able to install the driver I temporary disabled the driver signature checks in Windows 10.

Driver signing is explained in README and many stories will be found in comments of closed issues.

MarkOstis, 2019-07-25:
I installed new release 0.0.6 version. The behavior the same. I got Code 10.

Here is the Kernel log file:
[KernelLog_StartEmulate.LOG](https://github.com/cezuni/usbip-win/files/3430309/KernelLog_StartEmulate.LOG)

And usbip debug log file:
[debug_pdu.log](https://github.com/cezuni/usbip-win/files/3430319/debug_pdu.log)

cezuni, 2019-07-25:
@MarkOstis : Can you share STMicroelectronics drivers? I tried to reproduce your case but my result logs are different from yours.  Maybe STMicroelectronics driver must seem to be installed.

MarkOstis, 2019-07-25:
@cezuni : Here is my STMicroelectronics driver:
[stm32_vcp_driver.zip](https://github.com/cezuni/usbip-win/files/3431296/stm32_vcp_driver.zip)

[image]

As I mentioned before, I use USBIP-Virtual-USB-Device and try to emulate cdc-acm.c device. It seems like it is a Virtual comport. I rewrote the cdc-acm.c file to emulate STMicroelectronics Virtual com port. Here is my version of USBIP-Virtual-USB-Device cdc-acm.c file:
[cdc-acm.txt](https://github.com/cezuni/usbip-win/files/3431331/cdc-acm.txt)

MarkOstis, 2019-07-25:
@cezuni : If need I can share full USBIP-Virtual-USB-Device project.

cezuni, 2019-07-25:
@MarkOstis : I have same results with you after installing STMicroelectronics driver and replacing with your cdc-acm.c. Sadly, there's no faulty log about usbip-win. 
In my opinion, the reason why STMicroelectronics device goes into wrong state is that STMicroelectronics driver have removed the device. (guessed from IRP_MN_REMOVE_DEVICE in kernel logs) VHCI seems to be not guilty. :)  I recommend you to check cdc-acm.c source.

MarkOstis, 2019-07-25:
I was looking thought my Kernel log file and found next:
00000141	0.06881560	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER

Could it be the root of this issue?

cezuni, 2019-07-25:
> I was looking thought my Kernel log file and found next:
00000141	0.06881560	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_INVALID_PARAMETER
> 
> Could it be the root of this issue?

I don't think so. That warning is not important because next usb commands proceed successfully.

MarkOstis, 2019-07-25:
@cezuni : Thank you for your help. I will try to fix cdc-acm file and respond to this issue.

halimselim, 2019-07-27:
We have an OMRON PLC and getting samre error code 10 This device cannot start
When PLC connected directly to usb port(without usbip) it works well on win10 with the [usb drivers](http://halimoglu.com/dosya/CXOneDrivers2017.zip) 

Other basic usb devices(usb drive,barcode scanner, serial converter) also work without problem through usbip.

Could you please check and confirm you get the same error ? 
usbip.exe list -r 79.137.39.83 (This is an orangepi linux server, usb bus id 7-1 is binded)

If needed we can give ssh root access to orangepi.
We can also pepare a test environment with a virtual windows pc directly connected to omron plc through usb.(to show it is working well without usbip). You may use this windows pc to compile and debug code.

We are a new startup based in Turkey. If we could solve this problem we will be happy to contribute to this project in some way. For example we have bought the usb token certificate from GlobalSign to sign the driver. We can share the signed driver with public. So people don't need to do bcdedit etc.
Could this be harmful to us ? What is your opinion? 

Thank you

cezuni, 2019-07-27:
@halimselim : What version did you test with ? More detailed logs would be very helpful. Please check README to obtain logs.

cezuni, 2019-07-27:
@halimselim : It would be more desirable that your issue is discussed in other page even though plc failed similar to this issue.  I'll create a new issue for your case.

closed 2020-03-14 by cezanne