cezanne/usbip-win #53: Devices with multiple configurations can't be configured correctly

rpasek, 2019-07-24:
Hello!
I'm working on a device which has multiple configurations. I've noticed that when attaching the device in Linux I see:
```
[   78.134224] usb 1-1: new high-speed USB device number 2 using vhci_hcd
[   78.284314] usb 1-1: SetAddress Request (2) to port 0
[   78.321383] usb 1-1: can't set config #1, error -1
```

Looking in the Windows kernel logs I see:
`usbip_stub:(EE) select_usb_conf: non-existent configuration descriptor: index: 1`

When running uspipd --debug in Windows I see:
```
usbip: info: exportable devices: 1
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:90:[dump_usb_device] path                 = \\?\usb#vid_f17b&pid_1111#ed648296380e#{fb265267-c609-41e6-8eca-a20d92a833e6}
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:91:[dump_usb_device] busid                = 1-12
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:97:[dump_usb_device] Device(C/SC/P)       = (Defined at Interface level) (00/00/00)
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:99:[dump_usb_device] bcdDevice            = 0
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:104:[dump_usb_device] Vendor/Product       = unknown vendor : unknown product (f17b:1111)
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:106:[dump_usb_device] bNumConfigurations   = 0
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:107:[dump_usb_device] bNumInterfaces       = 0
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:109:[dump_usb_device] speed                = High Speed(480Mbps)
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:111:[dump_usb_device] busnum               = 1
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\lib\usbip_common.c:112:[dump_usb_device] devnum               = c
usbip: info: request 0x8005: complete
usbip: info: connection from 172.17.221.248:59058
usbip: info: received request: 0x8003
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\src\usbipd\usbipd_import.c:18:[forwarder_stub] stub forwarding started
usbip: debug: C:\Users\rpasek\Documents\usbip-win\userspace\src\usbipd\usbipd_import.c:108:[recv_request_import] import request busid 1-12: complete
```

The error seems to make sense considering what is reported in uspipd. It seems that one of the descriptors is not being read properly or the data from the descriptors isn't being propagated correctly. Do you have any hints on why bNumConfigurations and bNumInterfaces appear as 0?

Thanks,
Richard

cezuni, 2019-07-24:
@rpasek : What version did you try testing with?  I created a new branch recently. Refer to https://github.com/cezuni/usbip-win/issues/22#issuecomment-514182116 .

> Do you have any hints on why bNumConfigurations and bNumInterfaces appear as 0?

This is no  problem. All things are zero displayed because linux does not hand over no  information.
usbip.exe message should be adequately modified.

rpasek, 2019-07-25:
Thanks for the info! I tried rebasing my branch with origin/fix_stall_multi_conf and I didn't see a difference. I'll continue to work with this rebased version with the hope I can find why this is happening:

`usbip_stub:(EE) select_usb_conf: non-existent configuration descriptor: index: 1`

rpasek, 2019-07-30:
I made a PR which seems to fix this problem https://github.com/cezuni/usbip-win/pull/58

cezuni, 2019-07-31:
@rpasek : I merged your #58 into master branch 35b0c116734957a6d0a4cde217b09a45a0593bd8 . Thanks for your great work..

closed 2019-07-31 by rpasek