cezanne/usbip-win #54: CMake Configuration for USBIP-Win

koudis, 2019-07-24:
As discussed in #23 It can be helpfull to move/add usbip-win to CMake.

I opened this Issue to gather more opinions and for discussion.

**CMake dvantages for me**

- much more use-case possibilities than Solution
- more stable --> 10 years old CMake files work without problem
- More readable (many tutorials, simple documentation, ...)

**Problems**
CMake works fine for standard user-space applications.
Driver development is quite complicated. There are several quite good CMake modules for Windows kernel drivers but It's not widely tested.

cezuni, 2019-07-24:
That's  great!!

> As discussed in #23 It can be helpfull to move/add usbip-win to CMake.

I hope that build process with CMake is compatible with an existing one at  least when CMake is firstly introduced.

koudis, 2019-07-25:
> I hope that build process with CMake is compatible with an existing one at least when CMake is firstly introduced.

This is not problem. CMake and .sln can coexist until/when CMake become stable (Or we just choose the better environment. In the end it can be a .sln as is now)

cezanne, 2021-01-16:
@koudis : I'll close it. We need discuss again someday.

closed 2021-01-16 by cezanne