cezanne/usbip-win #56: VHCI failure for ORMON PLC

cezuni, 2019-07-27:
VHCI fails for ORMON PLC. Detailed description is here: https://github.com/cezuni/usbip-win/issues/52#issuecomment-515667578

cezuni, 2019-07-27:
@halimselim : I tried your usbip server several times. All results were same that vhci failed after selecting configuration. In the window kernel log, IRP_MN_REMOVE_DEVICE was shown after selecting configuration. Maybe ORMON plc driver removed it. There were no warning or error in logs of usbip-win.

By the way, your server cannot be attached any more because your device was not detached by mistake before my client was to be reset. If a usbip connection is lost without detach, a stub server should seem to be reset manually.

halimselim, 2019-07-28:
Thank you for testing, I rebinded the device. What do you suggest me to do ? Device driver works fine when connected to usb directly, but fails with usbip. And usbip shows no error as you say. Is there anything we could do ?

cezuni, 2019-07-28:
I try again to attach to your server.. and never forget to detach it.

We should check if usb compatible ids are same for both cases(via usbip-win or directly). Can you compare usb identifiers? For more details, please refer to
https://github.com/cezuni/usbip-win/issues/25#issuecomment-508264500 .

cezuni, 2019-07-28:
Here's are logs from my testing: [usbip_log.txt](https://github.com/cezuni/usbip-win/files/3438884/usbip_log.txt), [debug_pdu.log](https://github.com/cezuni/usbip-win/files/3438885/debug_pdu.log)

Your server returned -EPIPE(-32) for selecting configuration. Your server log may contain useful information. Can you share it?

halimselim, 2019-07-28:
Following log is from /var/log/syslog

Jul 26 11:52:52   localhost usbipd: usbipd: info: connection from ::1:38528
--
Jul 26 11:52:53 localhost usbipd: usbipd:   info: received request: 0x8005(6)
Jul 26 11:52:53 localhost usbipd: usbipd:   info: exportable devices: 0
Jul 26 11:52:53 localhost usbipd: usbipd:   info: request 0x8005(6): complete
Jul 26 11:53:24 localhost kernel:   [5088395.878108] usbip-host 7-1: usbip-host: register new device (bus 7 dev   13)
Jul 26 11:53:31 localhost usbipd: usbipd:   info: connection from ::1:38530
Jul 26 11:53:31 localhost usbipd: usbipd:   info: received request: 0x8005(6)
Jul 26 11:53:31 localhost usbipd: usbipd:   info: exportable devices: 1
Jul 26 11:53:31 localhost usbipd: usbipd:   info: request 0x8005(6): complete

Jul 27 07:58:16 localhost usbipd: usbipd:   info: connection from ::1:38532
Jul 27 07:58:16 localhost usbipd: usbipd:   info: received request: 0x8005(6)
Jul 27 07:58:16 localhost usbipd: usbipd:   info: exportable devices: 1
Jul 27 07:58:16 localhost usbipd: usbipd:   info: request 0x8005(6): complete

Jul 27 08:47:16 localhost usbipd: usbipd:   info: connection from ::1:38534
Jul 27 08:47:16 localhost usbipd: usbipd:   info: received request: 0x8005(6)
Jul 27 08:47:16 localhost usbipd: usbipd:   info: exportable devices: 1
Jul 27 08:47:16 localhost usbipd: usbipd:   info: request 0x8005(6): complete
Jul 27 08:48:17 localhost usbipd: usbipd:   info: connection from ::1:38536
Jul 27 08:48:17 localhost usbipd: usbipd:   info: received request: 0x8003(6)
Jul 27 08:48:17 localhost kernel:   [5163690.776534] usbip-host 7-1: stub up
Jul 27 08:48:17 localhost usbipd: usbipd:   info: found requested device: 7-1
Jul 27 08:48:17 localhost usbipd: usbip:   info: connect: 7-1
Jul 27 08:48:17 localhost usbipd: usbipd:   info: request 0x8003(6): complete
Jul 27 08:48:20 localhost kernel:   [5163693.688694] usbip-host 7-1: endpoint 0 is stalled
Jul 27 08:49:42 localhost usbipd: usbipd:   info: connection from ::1:38538
Jul 27 08:49:42 localhost usbipd: usbipd:   info: received request: 0x8003(6)
Jul 27 08:49:42 localhost usbipd: usbipd:   info: found requested device: 7-1
Jul 27 08:49:42 localhost usbipd: usbipd:   info: request 0x8003(6): failed
Jul 27 08:50:30 localhost kernel:   [5163823.680455] usbip-host 7-1: recv a header, 0
Jul 27 08:50:30 localhost kernel:   [5163823.686826] usb 7-1: can't set config #1, error -32
Jul 27 08:50:34 localhost kernel:   [5163827.677638] usbip-host 7-1: usbip-host: register new device (bus 7 dev   13)
Jul 27 08:52:02 localhost usbipd: usbipd:   info: connection from ::1:38540
Jul 27 08:52:02 localhost kernel:   [5163915.373991] usbip-host 7-1: stub up
Jul 27 08:52:02 localhost usbipd: usbipd:   info: received request: 0x8003(6)
Jul 27 08:52:02 localhost usbipd: usbipd:   info: found requested device: 7-1
Jul 27 08:52:02 localhost usbipd: usbip:   info: connect: 7-1
Jul 27 08:52:02 localhost usbipd: usbipd:   info: request 0x8003(6): complete
Jul 27 08:52:05 localhost kernel:   [5163918.278763] usbip-host 7-1: can't set config #1, error -32
Jul 27 08:52:05 localhost kernel:   [5163918.282671] usbip-host 7-1: endpoint 0 is stalled
Jul 27 08:52:44 localhost kernel:   [5163957.533928] usbip-host 7-1: recv a header, 0
Jul 27 08:52:44 localhost kernel:   [5163957.720178] usbip-host 7-1: reset full-speed USB device number 13 using   ohci-platform
Jul 27 08:52:44 localhost kernel:   [5163957.949767] usbip-host 7-1: device reset
Jul 27 08:54:15 localhost usbipd: usbipd:   info: connection from ::1:38542
Jul 27 08:54:15 localhost usbipd: usbipd:   info: received request: 0x8005(6)
Jul 27 08:54:15 localhost usbipd: usbipd:   info: exportable devices: 1
Jul 27 08:54:15 localhost usbipd: usbipd:   info: request 0x8005(6): complete

Jul 27 21:51:38 localhost usbipd: usbipd:   info: connection from ::1:38544
Jul 27 21:51:38 localhost kernel:   [5210692.625214] usbip-host 7-1: stub up
Jul 27 21:51:38 localhost usbipd: usbipd:   info: received request: 0x8003(6)
Jul 27 21:51:38 localhost usbipd: usbipd:   info: found requested device: 7-1
Jul 27 21:51:38 localhost usbipd: usbip:   info: connect: 7-1
Jul 27 21:51:38 localhost usbipd: usbipd:   info: request 0x8003(6): complete
Jul 27 21:51:41 localhost kernel:   [5210695.003503] usbip-host 7-1: endpoint 0 is stalled
Jul 27 21:52:57 localhost kernel:   [5210771.522235] usbip-host 7-1: recv a header, 0
Jul 27 21:52:57 localhost kernel:   [5210771.713710] usbip-host 7-1: reset full-speed USB device number 13 using   ohci-platform
Jul 27 21:52:58 localhost kernel:   [5210771.945663] usbip-host 7-1: device reset
Jul 27 22:05:43 localhost systemd[1]:   Starting Daily apt download activities...
Jul 27 22:05:43 localhost systemd[1]:   Started Daily apt download activities.
Jul 27 22:29:53 localhost usbipd: usbipd:   info: connection from ::1:38546
Jul 27 22:29:54 localhost kernel:   [5212988.102983] usbip-host 7-1: stub up
Jul 27 22:29:54 localhost usbipd: usbipd:   info: received request: 0x8003(6)
Jul 27 22:29:54 localhost usbipd: usbipd:   info: found requested device: 7-1
Jul 27 22:29:54 localhost usbipd: usbip:   info: connect: 7-1
Jul 27 22:29:54 localhost usbipd: usbipd:   info: request 0x8003(6): complete
[... 41 lines cut ...]
Jul 28 04:55:23 localhost usbipd: usbipd:   info: connection from ::1:38560
Jul 28 04:55:23 localhost usbipd: usbipd:   info: received request: 0x8005(6)
Jul 28 04:55:23 localhost usbipd: usbipd:   info: exportable devices: 1
[... cut ...]
Jul 28 05:50:00 localhost usbipd: usbip:   info: connect: 7-1
Jul 28 05:50:00 localhost usbipd: usbipd:   info: request 0x8003(6): complete
Jul 28 05:50:03 localhost kernel:   [5239398.056382] usbip-host 7-1: endpoint 0 is stalled
Jul 28 05:51:35 localhost kernel:   [5239489.446938] usbip-host 7-1: recv a header, 0
Jul 28 05:51:35 localhost kernel:   [5239489.631859] usbip-host 7-1: reset full-speed USB device number 13 using   ohci-platform
Jul 28 05:51:35 localhost kernel:   [5239489.863467] usbip-host 7-1: device reset

cezuni, 2019-07-28:
@halimselim : The server log also said that selecting conf failed with -EPIPE. Control port may be stalled. I need to check further that vhci handles correctly the result code. 
BTW, file attachment would be better than inserting log entries as a comment text. This page will be too long if all logs are inserted as comments.  😄

cezuni, 2019-07-29:
@halimselim : Is PLC_NS driver correct ? The driver removes the usbip device after selecting configuration even if a vhci forcefully returns success code.

halimselim, 2019-07-29:
If you connect to my local pc, I will plug in omron plc to the physical usb port. Then you will be able to compare the difference. Driver loads successfully

cezuni, 2019-07-29:
@halimselim : I do not suspect the correctness of a PLC driver. I asked you that driver files to be installed are located in CXOne-Drivers2017/drivers/USB/8/PLC_NS/64bit folder of driver zip.

halimselim, 2019-07-29:
Yes that's correct. I reinstalled to confirm.
[windriver.txt](https://github.com/cezuni/usbip-win/files/3441054/windriver.txt)

cezuni, 2019-07-29:
@halimselim : When a plc is directly attached without usbip-win, what are driver file details and compatible ids of the device? They can be found from device properties in the device manager. 
Also I want to know "lsusb -v" results of the linux server.

halimselim, 2019-07-29:
Here is lsusb -v result: [lsusb_v.txt](https://github.com/cezuni/usbip-win/files/3444575/lsusb_v.txt)
I will go to office 8 hours later, only then I can directly attach to get those details.

halimselim, 2019-07-30:
[omrondriver.txt](https://github.com/cezuni/usbip-win/files/3445567/omrondriver.txt)
and many more details are present
[omrondriver](https://user-images.githubusercontent.com/23665903/62108270-53b7fb80-b2b2-11e9-8d4c-c63e40827b23.png)

cezuni, 2019-07-30:
@halimselim : Sadly, compatible ids are same between a real device and a virtual device via usbip-win. I had asked also driver file details. Which drivers are loaded for a plc device ?  That information resides in a driver tab of a properties window.

halimselim, 2019-07-30:
[image]

cezuni, 2019-07-30:
@halimselim : Loaded drivers are also same.. 
Currently, I can't figure out the reason why a plc driver removes a virtual plc device if the device is created via usbip-win. Can you capture usb packets when a plc device is directly attached ? I had used Wireshark to capture.

halimselim, 2019-07-30:
[ws2.zip](https://github.com/cezuni/usbip-win/files/3446350/ws2.zip)

cezuni, 2019-07-30:
@halimselim : Was ws2.pcapng in the zip file generated with a plc driver directly attached without usbip-win? But the wireshark log you've sent is exactly same with usbip-win log.  Weird!!

halimselim, 2019-07-30:
Yes it is the log of direct attach

halimselim, 2019-07-30:
I know that eltima software works well with this plc. When it is installed it creates its own branch under Device Manager tree. But usbip is listed under System Devices branch. Could it be related ?

cezuni, 2019-07-30:
@halimselim : 
> I know that eltima software works well with this plc. When it is installed it creates its own branch (Eltima Usb Stub) under Device Manager tree. But usbip is listed under System Devices branch. Could it be related ?

This means that device classes of both case are different: Eltima Usb Stub(custom device class) and System(usbip-win). In my opinion, such a difference does not matter. By the way, is it USB class(not system class) for usbip-win and directly attached case ? 

Now, I should look into the beginning parts of usbip-win logs such as IRP_MN_QUERY_XXX related stuff. In fact, many bus queries have not been handled properly in usbip-win because there's no problem with unhandled queries. It can be a long journey. It may be better to use eltima.. 😢

halimselim, 2019-07-30:
Yes Omron PLC appears under "Universal Serial Bus Controllers" in both cases. We will be watching the developments, thanks for your time.

closed 2020-03-14 by cezanne