cezanne/usbip-win #60: usbip error:failed to update driver:103

zhLoser01, 2019-08-06:
I use win7 as the server side.
[image]

cezuni, 2019-08-07:
@zhLoser01 : The error(0x103, ERROR_NO_MORE_ITEMS) in your screenshot may indicate that your system has a more specific driver than usbip_stub.sys in a temporarily generated inf. Take a look at [UpdateDriverForPlugAndPlayDevices API in msdn](https://docs.microsoft.com/en-us/windows/win32/api/newdev/nf-newdev-updatedriverforplugandplaydevicesa). According to the document, the following code may require additional INSTALLFLAG_FORCE  flag. Can you try?  
https://github.com/cezuni/usbip-win/blob/f87ff13c0bef32062a016d6e6d0a3b8f3619ce56/userspace/lib/usbip_stub.c#L176

zhLoser01, 2019-08-07:
[image]
What should I do?

cezuni, 2019-08-07:
@zhLoser01 : modify the source as follows:
 "INSTALLFLAG_NONINTERACTIVE" -> "INSTALLFLAG_NONINTERACTIVE | INSTALLFLAG_FORCE"

zhLoser01, 2019-08-07:
[image]
[image]

zhLoser01, 2019-08-07:
error 103

cezuni, 2019-08-07:
@zhLoser01 : Which drivers are currently loaded for the DataTraveler G4 device ? You can manually replace the driver with an usbip stub driver via device manager.

zhLoser01, 2019-08-07:
[image]
I manually added this driver

zhLoser01, 2019-08-07:
[image]
I have manually updated the driver

cezuni, 2019-08-07:
@zhLoser01 : If you can update the driver manually, there's no reason to fail with usbip bind. Can you try with usbip bind again after removing the device?

cezuni, 2019-08-07:
@zhLoser01 : I recommend you to build usbip.exe with no INSTALLFLAG_NONINTERACTIVE. If that flag is dropped, some useful window may appear.

zhLoser01, 2019-08-07:
[image]
This file is generated by you, how can I generate it?

cezuni, 2019-08-07:
@zhLoser01 : It's a template file for a stub driver inf. You can get it from [source tree](https://github.com/cezuni/usbip-win/blob/master/driver/stub/usbip_stub.inx) or [release page](https://github.com/cezuni/usbip-win/releases/download/v0.0.5/usbip_stub.inx).

zhLoser01, 2019-08-07:
Yes, I use the source code you posted, but you are 64-bit, my server is 32-bit.

cezuni, 2019-08-07:
@zhLoser01 : Well, usbip-win is not ready for 32bit. But you can easily edit the file for 32bit. Maybe  all most things to do is to replace amd64 -> x86 ??

zhLoser01, 2019-08-07:
[image]
Yes, I am compiling with x86, the compiler can pass, but I don't know why the binding failed. The error code is 103.

cezuni, 2019-08-07:
@zhLoser01 : usbip_stub.inx should be properly modified for 32bit(x86). Did you edit that file?

zhLoser01, 2019-08-07:
no

cezuni, 2019-08-07:
> no

You should. If you have trouble for editing, I'll do that later..

zhLoser01, 2019-08-07:
[image]
I don't know how to compile this file.

cezuni, 2019-08-07:
@zhLoser01 : usbip_stub.inx is just a simple text file. usbip.exe reads that file and automatically generates a stub driver installation file(inf). No need to compile.

zhLoser01, 2019-08-07:
[image]
Ok, compiling this usbip_stub with x86 is also successful.

zhLoser01, 2019-08-07:
[image]
[image]
I commented out this paragraph. Is this a hint that the binding is successful?

cezuni, 2019-08-07:
@zhLoser01 : No, it's not working. You just need a modified usbip_stub.inx for 32bit.

zhLoser01, 2019-08-07:
[image]
I don't know how to change this to 32-bit.

cezuni, 2019-08-07:
@zhLoser01 : I'm not sure but all you have to do is to replace amd64 with x86 for usbip_stub.inx file. Here is [good reference](https://docs.microsoft.com/ko-kr/windows-hardware/drivers/install/inf-manufacturer-section).

zhLoser01, 2019-08-07:
So, this usbip_stub.inx file needs to be written manually, it is not automatically generated？

zhLoser01, 2019-08-08:
[image]
@cezuni What is the cause of this error?

cezuni, 2019-08-08:
> So, this usbip_stub.inx file needs to be written manually, it is not automatically generated？

No, it's not automatically generated. You can edit the file manually.

The server screenshot tells that the 1-71 device is not correctly bound to a usbip stub. Please retry after you edits the usbip_stub.inx.

Here is a [modified usbip_stub.inx for x86](https://github.com/cezuni/usbip-win/files/3480060/usbip_stub.inx.zip), which is not tested because I have no x86 test machine.

zhLoser01, 2019-08-08:
The server uses 32 bits and the client uses 64 bits. This seems to be no good

cezanne, 2020-03-14:
@zhLoser01 : Close too old issue.

closed 2020-03-14 by cezanne