cezanne/usbip-win #63: Error Installing Driver usbip_vhci.inf

DanielBorgesOliveira, 2019-08-16:
Hi friends,

I am trying to install the drivers and the device manager return the error:

Deriver is no intended for this plataform.

The image show the details.

[image]

The OS is Windows 10 Pro x64.

What can I do to resolv this?

I thank you very much.

cezuni, 2019-08-16:
@DanielBorgesOliveira : Please check the steps in "Windows USB/IP Client" section of README. vhci installation works well. You seemed to miss some installation steps.

DanielBorgesOliveira, 2019-09-03:
Thank for your reply. I solved the issue compiling the driver from source.

closed 2019-09-05 by cezuni