cezanne/usbip-win #64: error attaching on win10 client

evelJ, 2019-08-29:
Hi,

I trying to share a usb device from Ubuntu 16.04 (usbip server) to a windows 10 (1809 17763.678) (usbip client).

Build OK.
Bind on Ubuntu 16.04 server side OK
Certificate installed OK
test signing enabled OK
All files copied OK
VHCI driver install OK
USB/IP VHCI driver visible into device manager and OK
usbip list OK (found wished usb device)

but :
usbip attach -r 192.168.1.2 -b 4.12
usbip: error: recv op_common
usbip: error: read_devbuf: failed to read: err: 0x6

Is this a known issue ?
Have I missed something ?

zhLoser01, 2019-10-15:
Have you solved this problem? I have encountered this problem now.

evelJ, 2019-10-15:
No, I haven't found any solution and I gave up.

But I you found something, I'm still interested !

mankitty, 2020-01-02:
> Hi,
>
> I trying to share a usb device from Ubuntu 16.04 (usbip server) to a windows 10 (1809 17763.678) (usbip client).
>
> Build OK.
> Bind on Ubuntu 16.04 server side OK
> Certificate installed OK
> test [...]

Did you run usbipd？

harshilishere, 2021-01-03:
I have  a usbip server running in raspiberry pi 4 but getting the same error as @evelJ 
Did someone found a possible solution??
Following this tutorial: https://www.youtube.com/watch?v=lsMsl8LEZvs&list=LL&index=1

cezanne, 2021-01-04:
@harshilishere : 
Thanks for introducing a youtube tutorial. It's very useful even to me.

For your issue, how about this comment? https://github.com/cezanne/usbip-win/issues/178#issuecomment-717513396

harshilishere, 2021-01-04:
@cezanne  thank you for a quick response!
I already did that. It is working perfectly now on the linux. The devices are listed on the windows side which are binded in the server. Its just that it gives me error when i try to attach any of them. And I have tried various USB devices. But same thing happens with all of them.

cezanne, 2021-01-04:
@harshilishere : Can you share the logs of usbipd and dmesg of raspberry pi? This kind of issue is related to protocol version mismatch or server configuration fault. If remote listing is ok, we may focus on a server side.

cezanne, 2021-01-16:
I'll close this issue. Please reopen if it is required.

closed 2021-01-16 by cezanne