cezanne/usbip-win #65: dd image to mass storage device results in corrupted partition

rpasek, 2019-09-03:
Hi All,
I've been trying to dd a disk image to a mass storage device however the vast majority of time the image ends up being corrupted. About 1 out of 10 times I'm successfully able to write the disk image with no errors. I've been using https://downloads.raspberrypi.org/raspbian_lite_latest as the source for my disk image for testing as it's not too large. Larger images show the same effect. I'm able to tell that the image was corrupt because I'm unable to mount it. I've been flashing and mounting the images with:

```
sudo dd if=2019-07-10-raspbian-buster-lite.img of=/dev/sdb bs=1M && sync
sudo mount /dev/sdb /media/newhd
```

The error that I see after attempting to mount the image is
`mount: /media/newhd: wrong fs type, bad option, bad superblock on /dev/sdb, missing codepage or helper program, or other error.`

Things that I've tried:
- Looking at the kernel logs on Windows and Linux. Nothing seems out of place.
- Looking at the ethernet packets using Wireshark. Nothing seems out of place.
- Different mass storage devices. All seem to show the same symptoms.
- Different disk images. All seem to show the same symptoms.
- Running dd against a mass storage device directly attached to a Linux machine. The image writes successfully.
- Running usbipd host on a Linux machine. The image still fails to write successfully.

Because using usbip on host and client still results in a failure, this likely indicates that usbip-win is not at fault however it is most certainly affected. With that said, would anyone have any tips on how to debug this kind of issue? I'm somewhat lost on where to go from here. Would anyone be interested in helping debug? Maybe we can push to mainline Linux when we find the problem?

Thanks,
Richard

cezuni, 2019-09-04:
@rpasek : I suspect that a device via usbip cannot manage sync operation well. 

Your test is equivalent to a following scenario which is better to isolate an error. 

- Prepare a file of similar size with a raspberry image, which is filled with a specific byte.
- dd write to a device.
- dd read from a device.
- Is the image read from a device same with the original one?

cezanne, 2020-02-24:
@rpasek : I'll close. Please reopen if you want.

closed 2020-02-24 by cezanne