cezanne/usbip-win #68: Win7 as USBIP Client and Server

XT19958, 2019-09-19:
Hello！I am configuring my windows 7 as client and server using this repository. I bind a USB flash drive on server,client connect server successful. And the device driver runs normally,but the client recognized the USB flash drive as a hard disk，and can't open it ,What is the reason for this?
[image]
[image]

cezuni, 2019-09-25:
@XT19958 : A USB flash drive is one of the working good devices with usbip-win. How about trying another USB device such as a USB mouse? It is also a working good device.

XT19958, 2019-09-26:
@cezuni : Thanks for your reply！My problem will happen only when **Win7** bind a USB flash drive as server（Win10 works good）.

cezanne, 2020-03-14:
@XT19958 : Close too old issue. Please reopen if you want.

closed 2020-03-14 by cezanne