cezanne/usbip-win #69: BSOD when using USB KEY

QingyangChen, 2019-09-25:
excuse me,Can you help me solve this problem?

environment：
client：        win10 18362
server：       win10 18362
usbip-win:    0.0.6

u disk can be used normally.
But when the client uses this usb key, the server will BSOD,

1: **kd> !analyze -v**

**Bugcheck Analysis**

**KERNEL_SECURITY_CHECK_FAILURE (139)**
A kernel component has corrupted a critical data structure.  The corruption
could potentially allow a malicious user to gain control of this machine.
Arguments:
Arg1: 0000000000000003, A LIST_ENTRY has been corrupted (i.e. double remove).
Arg2: fffffa02c3be6b30, Address of the trap frame for the exception that caused the bugcheck
Arg3: fffffa02c3be6a88, Address of the exception record for the exception that caused the bugcheck
Arg4: 0000000000000000, Reserved
......

this is my BOSD dump file.
[092519-70375-01.dmp.txt](https://github.com/cezuni/usbip-win/files/3651353/092519-70375-01.dmp.txt)
thanks!

koudis, 2019-09-25:
Hello,

maybe it's the same problem which is resolved by
https://github.com/cezuni/usbip-win/pull/67

koudis, 2019-09-25:
And some quistions

- Under server you mean USBIP Linux implementation or USBIP Windows implementation? (you host key on Windows with USBIP Windows?)
- The client is Windows or Linux?

QingyangChen, 2019-09-25:
> And some questions
> 
> * Under server you mean USBIP Linux implementation or USBIP Windows implementation? (you host key on Windows with USBIP Windows?)
> * The client is Windows or Linux?

thank you for your reply.
Here is the answer to your question:
- usb key on Windows with USBIP Windows.
- client is Windows10.
- server is Windows10.

QingyangChen, 2019-09-25:
I have modified the code according to #67 
BOSD after recompiling and running
This is the result of windbg analysis:
`BAD_POOL_CALLER (c2)`
`The current thread is making a bad pool request.  Typically this is at a bad IRQL level or double freeing the same allocation, etc.`
`Arguments:`
`Arg1: 000000000000000d, Attempt to release quota on a corrupted pool allocation.`
`Arg2: ffffc50c03936310, Address of pool`
`Arg3: 0000000020ab4433, Pool allocation's tag`
`Arg4: 981276f174f0dba6, Quota process pointer (bad).`
`Debugging Details:`
`------------------`
`*** WARNING: Unable to verify checksum for usbipd.exe`
`KEY_VALUES_STRING: 1`
`PROCESSES_ANALYSIS: 1`
`SERVICE_ANALYSIS: 1`
`STACKHASH_ANALYSIS: 1`
`TIMELINE_ANALYSIS: 1`
`DUMP_CLASS: 1`
`DUMP_QUALIFIER: 0`
`BUILD_VERSION_STRING:  18362.1.amd64fre.19h1_release.190318-1202`
`DUMP_TYPE:  0`
`BUGCHECK_P1: d`
`BUGCHECK_P2: ffffc50c03936310`
`BUGCHECK_P3: 20ab4433`
`BUGCHECK_P4: 981276f174f0dba6`
`FAULTING_IP: `
`nt!IopCompleteRequest+8e`
`fffff805'6da4c4ce 8327cf          and     dword ptr [rdi],0FFFFFFCFh`
`BUGCHECK_STR:  0xc2_d`
`CPU_COUNT: 2`
`CPU_MHZ: 9c0`
`CPU_VENDOR:  GenuineIntel`
`CPU_FAMILY: 6`
`CPU_MODEL: 9e`
`CPU_STEPPING: 9`
`CPU_MICROCODE: 6,9e,9,0 (F,M,S,R)  SIG: 8E'00000000 (cache) 8E'00000000 (init)`
`DEFAULT_BUCKET_ID:  WIN8_DRIVER_FAULT`
`PROCESS_NAME:  usbipd.exe`
`CURRENT_IRQL:  1`
`ANALYSIS_SESSION_HOST:  DESKTOP-8PVQ1JT`
`ANALYSIS_SESSION_TIME:  09-25-2019 17:17:37.0584`
`ANALYSIS_VERSION: 10.0.18362.1 amd64fre`
`IRP_ADDRESS: ffffc50c03cb4478`
`LAST_CONTROL_TRANSFER:  from fffff8056dcaf2a2 to fffff8056dbcef30`
`STACK_TEXT:  `
`ffffa50f'68d74a18 fffff805'6dcaf2a2 : 00000000'0000000d 00000000'00000003 ffffa50f'68d74b80 fffff805'6db23b10 : nt!DbgBreakPointWithStatus`
`ffffa50f'68d74a20 fffff805'6dcae992 : 00000000'00000003 ffffa50f'68d74b80 fffff805'6dbdb760 00000000'000000c2 : nt!KiBugCheckDebugBreak+0x12`
`ffffa50f'68d74a80 fffff805'6dbc71a7 : 00000000'00000000 00000000'00000046 ffffc50b'fb602000 00000000'80000004 : nt!KeBugCheck2+0x952`
`ffffa50f'68d75180 fffff805'6dc0092b : 00000000'000000c2 00000000'0000000d ffffc50c'03936310 00000000'20ab4433 : nt!KeBugCheckEx+0x107`
`ffffa50f'68d751c0 fffff805'6dd740a9 : 00000000'00000000 00000000'00000000 00000000'00000000 01000000'00100000 : nt!ExFreeHeapPool+0x1b4deb`
`ffffa50f'68d752e0 fffff805'6da4c4ce : 00000000'00000000 00000000'00000000 00000000'00000000 00000000'00000000 : nt!ExFreePool+0x9`
`ffffa50f'68d75310 fffff805'6da4dddc : ffffc50c'03cb44f0 ffffa50f'68d75420 00000000'00000001 ffffa50f'68d75530 : nt!IopCompleteRequest+0x8e`
`ffffa50f'68d75400 fffff805'6da4d667 : ffffc50c'06298660 00000000'00000000 ffffa50f'68d75500 ffffc50c'06298601 : nt!IopfCompleteRequest+0x75c`
`ffffa50f'68d75510 fffff805'723346d8 : ffffc50b'ffc42da0 ffffc50b'fb703e20 fffff805'72339e50 00000000'00000000 : nt!IofCompleteRequest+0x17`
`ffffa50f'68d75540 fffff805'72334483 : ffffc50b'ffc42da0 ffffc50c'03cb44e0 ffffc50b'fb703e20 ffffc50c'06298660 : `**`usbip_stub!send_stub_res+0x248 [E:\Ô´Âë\usbip-win-0.0.6\driver\stub\stub_res.c @ 178]`** 
`ffffa50f'68d75590 fffff805'7233424c : ffffc50b'ffc42da0 ffffc50b'fb703e00 ffffc50b'fb70cf90 ffffc50c'039362f0 : usbip_stub!send_irp_sres+0xa3 [E:\Ô´Âë\usbip-win-0.0.6\driver\stub\stub_res.c @ 265] `
`ffffa50f'68d755e0 fffff805'723342c5 : ffffc50b'ffc42da0 ffffc50b'fb703e20 545f5453'00000000 ffffc50b'fb70cf90 : usbip_stub!reply_stub_req+0x7c [E:\Ô´Âë\usbip-win-0.0.6\driver\stub\stub_res.c @ 301] `
`ffffa50f'68d75620 fffff805'7233624d : ffffc50b'ffc42da0 ffffa50f'0000000c ffffc50c'039362f0 ffff0ff7'00000037 : usbip_stub!reply_stub_req_data+0x65 [E:\Ô´Âë\usbip-win-0.0.6\driver\stub\stub_res.c @ 327] `
`ffffa50f'68d75680 fffff805'723364d9 : ffffc50b'ffc42da0 ffffc50c'039362e8 ffffc50c'039362c0 00000000'0000000c : usbip_stub!process_class_request+0x18d [E:\Ô´Âë\usbip-win-0.0.6\driver\stub\stub_write.c @ 181] `
`ffffa50f'68d75710 fffff805'723362bd : ffffc50b'ffc42da0 ffffc50c'039362c0 ffffc50c'06298660 ffffc50c'06298660 : usbip_stub!process_control_transfer+0xa9 [E:\Ô´Âë\usbip-win-0.0.6\driver\stub\stub_write.c @ 206] `
`ffffa50f'68d75760 fffff805'72336ee9 : ffffc50b'ffc42da0 ffffc50c'03cb4010 ffffc50c'039362c0 fffff805'7233eae0 : usbip_stub!process_cmd_submit+0x2d [E:\Ô´Âë\usbip-win-0.0.6\driver\stub\stub_write.c @ 368] `
`ffffa50f'68d757a0 fffff805'72332b2d : ffffc50b'ffc42da0 ffffc50c'03cb4010 fffff805'72339a20 fffff805'7233be98 : usbip_stub!stub_dispatch_write+0x99 [E:\Ô´Âë\usbip-win-0.0.6\driver\stub\stub_write.c @ 440] `
`ffffa50f'68d757e0 fffff805'6da37f39 : ffffc50b'ffc42c50 ffffc50c'03cb4010 0000015d'59550348 00000000'20206f49 : usbip_stub!stub_dispatch+0x10d [E:\Ô´Âë\usbip-win-0.0.6\driver\stub\stub_dispatch.c @ 53] `
`ffffa50f'68d75830 fffff805'6dfee345 : ffffc50c'03cb4010 00000000'00000000 00000000'00000000 ffffc50c'06298660 : nt!IofCallDriver+0x59`
`ffffa50f'68d75870 fffff805'6e04ad46 : ffffd686'00000000 ffffc50b'ffc42c50 ffffc50c'06298660 ffffa50f'68d75b00 : nt!IopSynchronousServiceTail+0x1a5`
`ffffa50f'68d75910 fffff805'6dbd8915 : 00000000'00000130 00000000'00000000 00007ffc'3adc6ec0 00007ff7'6154d280 : nt!NtWriteFile+0x676`
`ffffa50f'68d75a10 00007ffc'3dd3c1d4 : 00007ffc'3adbfcc8 00000000'00000000 00007ffc'3adc6ec0 00000044'055ff298 : nt!KiSystemServiceCopyEnd+0x25`
`00000044'055fee58 00007ffc'3adbfcc8 : 00000000'00000000 00007ffc'3adc6ec0 00000044'055ff298 00007ff7'6154d280 : ntdll!NtWriteFile+0x14`
[... cut ...]
`OSNAME:  Windows 10`
`OSEDITION:  Windows 10 WinNt TerminalServer SingleUserTS`
`OS_LOCALE:  `
`USER_LCID:  0`
`OSBUILD_TIMESTAMP:  unknown_date`
`BUILDDATESTAMP_STR:  190318-1202`
`BUILDLAB_STR:  19h1_release`
`BUILDOSVER_STR:  10.0.18362.1.amd64fre.19h1_release.190318-1202`
`ANALYSIS_SESSION_ELAPSED_TIME:  78f8`
`ANALYSIS_SOURCE:  KM`
`FAILURE_ID_HASH_STRING:  km:0xc2_d_usbip_stub!send_stub_res`
`FAILURE_ID_HASH:  {8233cfcb-ce09-1a62-9d1e-499065033395}`
`Followup:     MachineOwner`
`---------`

QingyangChen, 2019-09-25:
The problem seems to be on the server side，usbip_stub.sys? @koudis

cezuni, 2019-09-25:
@QingyangChen : 
Did you test with v0.0.6 in the release page?
A line number of your windbg output does not match with v0.0.6 sources. 

Anyway, v0.0.6 is a little old version.
I'll upload a new release from up-to-date master branch.

cezuni, 2019-09-25:
@QingyangChen : Could you try again with [v0.0.7](https://github.com/cezuni/usbip-win/releases/tag/v0.0.7)?

QingyangChen, 2019-09-26:
@cezuni 
Yes, I tested the code with v0.0.6
I added some print code, so line number does not match with v0.0.6 sources.

Just now, I tested the code for v0.0.7 without making any changes.
But still BOSD.
This is the result of windbg analysis:
***BAD_POOL_CALLER (c2)***
*The current thread is making a bad pool request.  Typically this is at a bad IRQL level or double freeing the same allocation, etc.*
*Arguments:*
*Arg1: 000000000000000d, Attempt to release quota on a corrupted pool allocation.*
*Arg2: ffff88890ccc7110, Address of pool*
*Arg3: 000000009f36a5bf, Pool allocation's tag*
*Arg4: 463fae8b28d9c34a, Quota process pointer (bad).*
*Debugging Details:*
*------------------*
*KEY_VALUES_STRING: 1*
*PROCESSES_ANALYSIS: 1*
*SERVICE_ANALYSIS: 1*
*STACKHASH_ANALYSIS: 1*
*TIMELINE_ANALYSIS: 1*
*DUMP_CLASS: 1*
*DUMP_QUALIFIER: 400*
*SYSTEM_MANUFACTURER:  TR*
*SYSTEM_PRODUCT_NAME:  911 SE-KN*
*SYSTEM_SKU:  Not Applicable*
*SYSTEM_VERSION:  Not Applicable*
*BIOS_VENDOR:  American Megatrends Inc.*
*BIOS_VERSION:  1.05.13*
*BIOS_DATE:  05/08/2017*
*BASEBOARD_MANUFACTURER:  TR*
*BASEBOARD_PRODUCT:  911 SE-KN*
*BASEBOARD_VERSION:  Not Applicable*
*DUMP_FILE_ATTRIBUTES: 0x8*
*Kernel Generated Triage Dump*
*DUMP_TYPE:  2*
*BUGCHECK_P1: d*
*BUGCHECK_P2: ffff88890ccc7110*
*BUGCHECK_P3: 9f36a5bf*
*BUGCHECK_P4: 463fae8b28d9c34a*
*FAULTING_IP:*
*nt!IopCompleteRequest+8e*
*fffff805'6f4464ce 8327cf          and     dword ptr [rdi],0FFFFFFCFh*
*BUGCHECK_STR:  0xc2_d*
*CPU_COUNT: 4*
*CPU_MHZ: 9c0*
*CPU_VENDOR:  GenuineIntel*
*CPU_FAMILY: 6*
*CPU_MODEL: 9e*
*CPU_STEPPING: 9*
*CPU_MICROCODE: 6,9e,9,0 (F,M,S,R)  SIG: 8E'00000000 (cache) 8E'00000000 (init)*
*BLACKBOXBSD: 1 (!blackboxbsd)*
*BLACKBOXNTFS: 1 (!blackboxntfs)*
*BLACKBOXPNP: 1 (!blackboxpnp)*
*BLACKBOXWINLOGON: 1*
*CUSTOMER_CRASH_COUNT:  1*
*DEFAULT_BUCKET_ID:  WIN8_DRIVER_FAULT*
***PROCESS_NAME:  usbipd.exe***
*CURRENT_IRQL:  1*
*ANALYSIS_SESSION_HOST:  DESKTOP-8PVQ1JT*
*ANALYSIS_SESSION_TIME:  09-26-2019 10:12:13.0362*
*ANALYSIS_VERSION: 10.0.18362.1 amd64fre*
*IRP_ADDRESS: ffff888907d7b938*
*LAST_CONTROL_TRANSFER:  from fffff8056f5fa92b to fffff8056f5c10a0*
*STACK_TEXT:*
*fffff50e'0e6df128 fffff805'6f5fa92b : 00000000'000000c2 00000000'0000000d ffff8889'0ccc7110 00000000'9f36a5bf : nt!KeBugCheckEx*
*fffff50e'0e6df130 fffff805'6f76e0a9 : 00000000'00000000 00000000'00000000 00000000'00000000 01000000'00100000 : nt!ExFreeHeapPool+0x1b4deb*
*fffff50e'0e6df250 fffff805'6f4464ce : 00000040'000000ff ffff78ff'754af849 ffff8889'0b4d8060 ffff8889'0a6e3790 : nt!ExFreePool+0x9*
*fffff50e'0e6df280 fffff805'6f447ddc : ffff8889'07d7b9b0 fffff805'6c10154f 00000000'00000001 ffffffff'ffffffff : nt!IopCompleteRequest+0x8e*
*fffff50e'0e6df370 fffff805'6f447667 : ffff8889'0a6e3790 00000000'00000000 fffff50e'0e6df400 ffff8889'0a6e3701 : nt!IopfCompleteRequest+0x75c*
[... 44 lines cut ...]
*FAILURE_BUCKET_ID:  0xc2_d_usbip_stub!send_stub_res*
*BUCKET_ID:  0xc2_d_usbip_stub!send_stub_res*
*PRIMARY_PROBLEM_CLASS:  0xc2_d_usbip_stub!send_stub_res*
*TARGET_TIME:  2019-09-26T02:08:19.000Z*
*OSBUILD:  18362*
*OSSERVICEPACK:  356*
*SERVICEPACK_NUMBER: 0*
*OS_REVISION: 0*
*SUITE_MASK:  784*
*PRODUCT_TYPE:  1*
*OSPLATFORM_TYPE:  x64*
*OSNAME:  Windows 10*
*OSEDITION:  Windows 10 WinNt TerminalServer SingleUserTS Personal*
*OS_LOCALE:*
*USER_LCID:  0*
*OSBUILD_TIMESTAMP:  unknown_date*
*ANALYSIS_SESSION_ELAPSED_TIME:  9f2*
*ANALYSIS_SOURCE:  KM*
*FAILURE_ID_HASH_STRING:  km:0xc2_d_usbip_stub!send_stub_res*
*FAILURE_ID_HASH:  {8233cfcb-ce09-1a62-9d1e-499065033395}*
*Followup:     MachineOwner*
*---------*

Here is BOSD dump file
[092619-8515-01.dmp.txt](https://github.com/cezuni/usbip-win/files/3655601/092619-8515-01.dmp.txt)

QingyangChen, 2019-09-26:
@cezuni 
The function of the usbip-win is no problem!
I will crash after reading the usb key remotely four or five times.
It may be a problem with the usb key driver compatibility?

cezuni, 2019-09-26:
@QingyangChen : 
> The function of the usbip-win is no problem!
> I will crash after reading the usb key remotely four or five times.

Did you get the same result without usbip-win ? If so, usbip-win is not related to your problem.

QingyangChen, 2019-09-26:
@cezuni 
I use the usb key directly, there is no such problem.
The result given by windbg is variable,
sometimes because of usbip_stub.sys, sometimes because of other system drivers.

cezanne, 2020-03-20:
@QingyangChen : I have updated the stub driver on [stub_fix branch](https://github.com/cezanne/usbip-win/tree/stub_fix). Cloud you retry?

cezanne, 2020-03-30:
@QingyangChen : I finally got a usb key to test. (tendyron usb key) And the device seemed to work with usbip client & server. Anyway, I'm going to close this issue because it lost vitality. Please reopen if you want.

closed 2020-03-30 by cezanne