cezanne/usbip-win #70: Flash drive is not working

kmvi-dev, 2019-09-27:
I am using the last version from the repository https://github.com/cezuni/usbip-win
The server part is deployed on 4.18.0-25 @26~18.04.1-Ubuntu SMP using usbutils from the distribution.
The client part is deployed on the VmWare virtual machine in the Windows 10 Pro (x64), in which debug mode is enabled.
The virtual machine is deployed on a host with a Windows 10 Pro (x64).
After building the client part, the usbip module and the driver were copied to the guest station.
Further, all actions were carried out according to the Readme.

As a result of buid and testing the assembly, it turned out the following:

Flash drive on Linux station [server side usb / ip]
[image]
[image]
[image]
[image]

cezuni, 2019-10-04:
@kmvi-dev : 
In flashdrive.LOG(usbip-win vhci log), there's no suspicious line except for this.
```
00000244	22:57:52.184	usbip_vhci: vpdo: minor: IRP_MN_REMOVE_DEVICE, IRP: 0xFFFFD00800E05010
```

From that line, the vhci device seems to have been removed by an upper driver.
Is there any such a driver in your Windows VM ?

kmvi-dev, 2019-10-05:
@cezuni 
No . I removed all the system USB drivers from the virtual machine before booting Windows.
I want to note that forwarding does not work correctly with some devices from the manufacturer Alcor Micro Corp.
 I found two such devices for the test. A flash drive and a card reader. The rest of the devices that I used in my projects work quite correctly
 ( Thank you for responding ) .

closed 2019-10-28 by kmvi-dev

cezanne, 2019-10-31:
@kmvi-dev : Sorry for my late response. Any progress on this closed issue?

kmvi-dev, 2019-10-31:
@cezuni 
Yes ,the flash drive turned out to be malfunctioning. :).
But there is another problem. This is due to card readers.
If all card reader slots are filled with SD cards, then a card reader is available and SD cards are read.
But if at least one slot is free, then transferring a card reader is not possible.
Card readers in working condition.

kmvi-dev, 2019-10-31:
@cezuni  
Open a new topic on this issue?

cezanne, 2019-10-31:
@kmvi-dev : 
> Open a new topic on this issue?

Yes, it can be a new topic.

kmvi-dev, 2019-10-31:
@cezuni 
Well, today I’ll try to collect debug information and create a new topic for this issue.

kmvi-dev, 2019-11-01:
@cezuni 
I checked the latest version from the repository and did not find any problems with the card reader. Everything works. Thank you for participating in solving problems with my devices.
P.S 
I needed to update the driver version and the last question would not be :)