cezanne/usbip-win #71: Problem with usbip-win (Certificate Issue)

Route66Fan, 2019-09-29:
When I try to run the command usbip.exe bind -b with the busid 1-198, I get the error 
usbip: error: cannot load certificate: subject: USBIP Test
usbip: error: bind_device: error binding device on busid 1-198: err: 0
The device using busid 1-198 is an SDRPlay software defined radio.
I would like to get this working. Thank you.

QingyangChen, 2019-09-30:
You should install the certificate to your local machine .

1. Open mmc as an administrator.
2. File->Add or Remove Snap-ins
[image]

3. Add Certificates , and select computer account
[image]

4. import certificate
[image]

QingyangChen, 2019-09-30:
The author mentioned the precautions in the readme

[image]

Route66Fan, 2019-09-30:
Ok, Thank you.

Route66Fan, 2019-09-30:
Ok. Thank you.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2020-02-25 by cezanne