cezanne/usbip-win #72: vs2017 v141 windows sdk 10.0.18362.0 generate problem

zhLoser01, 2019-10-12:
[image]
[image]

[image]

zhLoser01, 2019-10-12:
Must I use the v142 generation tool?  @cezuni

zhLoser01, 2019-10-14:
[image]

zhLoser01, 2019-10-14:
[image]

zhLoser01, 2019-10-15:
[image]

zhLoser01, 2019-10-15:
[image]

zhLoser01, 2019-10-15:
[image]

zhLoser01, 2019-10-15:
[image]

zhLoser01, 2019-10-16:
Can list usb devices

zhLoser01, 2019-10-16:
[image]

cezanne, 2020-03-14:
@zhLoser01 : Close too old issue. Please reopen if you want.

closed 2020-03-14 by cezanne

orcuan, 2020-05-25:
Any updates on this issue ? It is still relevant for me.

designworxz, 2020-10-28:
Still getting this error with new release
usbip: error: recv op_common
usbip: error: read_devbuff: failed to read: err: 0x6

koudis, 2020-10-28:
Hi,

0x6 is a network error. So you forgot to star usbipd daemon or maybe
firewall?