cezanne/usbip-win #73: Error installing driver

danielkucera, 2019-10-15:
I'm getting following error when trying to install the driver in Win 7 Enterprise x64:
[image]

The folder with driver contains following files:
```
usbip.exe       usbip_test.pfx  usbip_vhci.inf
usbip_stub.sys  usbip_vhci.cat  usbip_vhci.sys
```

Am I doing something wrong?

danielkucera, 2019-10-15:
I also tried to install it manually but it is still not shown in device manager:
```
C:\usbip>pnputil -i -a usbip_vhci.inf
Microsoft PnP Utility

Processing inf :            usbip_vhci.inf
Successfully installed the driver on a device on the system.
Driver package added successfully.
Published name :            oem267.inf

Total attempted:              1
Number successfully imported: 1

C:\usbip>usbip.exe attach -r 10.11.1.31 -b 1-13
usbip: error: walker_devpath: invalid hw id: root\vbdenum

usbip: error: open vhci driver
usbip: error: read_devbuf: failed to read: err: 0x6
```

danielkucera, 2019-10-15:
I was able to install this one https://github.com/barbalion/usbip-win-client/tree/master/old_signed
and the error message changed to:
```
usbip: error: walker_devpath: invalid hw id: root\vbdenum

usbip: error: walker_devpath: invalid hw id: root\usbipenum

usbip: error: open vhci driver
usbip: error: read_devbuf: failed to read: err: 0x6
```

zhLoser01, 2019-10-16:
I also reported this error.

danielkucera, 2019-10-31:
@cezanne you can see in my first post that I tried to install it via dev manager but it shows: folder doesn't contain compatible software driver.
Is it tested with win 7 x64?

cezanne, 2019-10-31:
@danielkucera : 
> Is it tested with win 7 x64?

No. The most or all vhci drivers were built against win 10. You should build it against win7. Refer #12.
Or you can try a pre-built vhci driver. (a little bit old version) In #57, I had uploaded vhci for win7 .

danielkucera, 2019-11-06:
This has worked but sometimes freezes my PC. Thanks.

closed 2019-11-06 by danielkucera