cezanne/usbip-win #74: USBIP-client hang on network disturbance

termski, 2019-10-22:
Have problems with the client and server hangs due to connection/ tcp session lost.

everything works well until there is a network issue, resulting in that the client "hangs" the usbip.exe process and its not possible to "kill" it or the process tree in Task Manager.
only thing that solves it is to restart windows.

is it possible to write some kind of code so it is possible to kill the usbip.exe process? so it does not become a complete zombie process.
I do understand that the actual transfer protocol of USB-ip does not have a kind of connection handler to be able to identify connection loss. but it would be great that the client could be able to be 'killed' and not forcing a reboot of the client.

usb device attached:  usb1 compatible HID mouse.
server: rpi4 Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux (usbip-utils 2.0)
client: win7x64 enterprise,  master branch from here.

lompal, 2019-11-27:
@termski You can use the `usbip.exe detach -p <port>` command to disable the zombie process without an additional reboot of the machine. Port `0` will detach all devices attached at the time of command execution. To detach a particular device, you need to know its port, which is enumerated from `1` and increments as new devices are attached.

Be careful if, for example, you have three devices connected in series, and then you detached the device with index `2`, then the newly connected equipment will first take port `2`, not `4`, as it might seem initially.

mankitty, 2019-12-30:
> @termski :
> 
> > everything works well until there is a network issue, resulting in that the client "hangs" the usbip.exe process and its not possible to "kill" it or the process tree in Task Manager.
> > only thing that solves it is to restart windows.
> 
> I reproduced the problem and confirmed that it's a bug. This bug seems to arise from [789b38a](https://github.com/cezanne/usbip-win/commit/789b38a47f0502c7b4fe785e022d13a3116f06c8).

hello, I have sent an email to you, but there is no reply. I would like to ask, does your project support composite equipment? thank you

koudis, 2019-12-30:
Hello all,

I am working on bugfix in usbip.exe and with composite devices. It seems that ABORT_PIPE, RESET_PIPE_AND_CLEAR_STALL and GET_DESCRIPTOR_FROM_INTERFACE is not implemented correctly. (it gets STALL_PID response sometimes)

Once the configuration of given USB device is not issued for first time the autorepair feature of USB is not able to repair failures due to non consistent implementation of URBs listed above.

I will create PR soon.

mankitty, 2020-01-02:
> Hello all,
>
> I am working on bugfix in usbip.exe and with composite devices. It seems that ABORT_PIPE, RESET_PIPE_AND_CLEAR_STALL and GET_DESCRIPTOR_FROM_INTERFACE is not implemented correctly. (it [...]

If you can solve the problem of composite equipment, it is really amazing. I am a novice in USB, so when I face the process of the entire project, I feel like I ca n’t start. I want to know how you solved this Problematic, for example, code or other aspects. Can we chat by mail if it is convenient? This is my email, 873337054@qq.com, thank you very much.

koudis, 2020-01-04:
@mankitty  I will create PR sonn.

I will be glad if you test my PR once it's created.

cezanne, 2020-04-02:
@termski : Can you try again with a recent version?

cezanne, 2020-04-07:
@termski : Please reopen if you want.

closed 2020-04-07 by cezanne