cezanne/usbip-win #78: Multi devices frequently connect and detach from Server causes BSOD (Client:windows10;Server:debian stretch)

lfld287, 2019-11-11:
Hello,debugview logs :
[crash.txt](https://github.com/cezanne/usbip-win/files/3829967/crash.txt)

lfld287, 2019-11-12:
It seems even one device can also cause a BSOD,debugview logs:

URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER
05465602	27356.53710938	usbip_vhci: store_urbr: urbr: [seq:42]
05465603	27356.53710938	usbip_vhci: store_urbr_submit: urbr: [seq:42]
05465604	27356.53710938	usbip_vhci: store_urbr_submit: urbr: [seq:42], func:URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER
05465605	27356.53710938	usbip_vhci: submit_urbr: urb requested: status:STATUS_PENDING
05465606	27356.53710938	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING
05465607	27356.53710938	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS
05465608	27356.53710938	usbip_vhci: vhci_read: Enter
05465609	27356.53710938	usbip_vhci: vhci_read: Leave: STATUS_PENDING
05465610	27359.58593750	usbip_vhci: vhci_ioctl: Enter
05465611	27359.58593750	usbip_vhci: ioctl code: IOCTL_USBIP_VHCI_UNPLUG_HARDWARE
05465612	27359.58593750	usbip_vhci: Plugging out single device: port: 1
05465613	27359.58593750	usbip_vhci: found device: port: 1
05465614	27359.58593750	usbip_vhci: Plugging out: port: 1
05465615	27359.58593750	usbip_vhci: finish pending irp
05465616	27359.58593750	usbip_vhci: Device 1 plug out finished
05465617	27359.58593750	usbip_vhci: vhci_ioctl: Leave: STATUS_SUCCESS
05465618	27359.58593750	usbip_vhci: vhci_cleanup: Enter
05465619	27359.58593750	usbip_vhci: vhci_cleanup: Leave
05465620	27359.58593750	usbip_vhci: vhci_internal_ioctl: Enter
05465621	27359.58593750	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_GET_PORT_STATUS
05465622	27359.58593750	usbip_vhci:(WW) device is not connected
05465623	27359.58593750	usbip_vhci: vhci_pnp: Enter
05465624	27359.58593750	usbip_vhci: vpdo: minor: IRP_MN_QUERY_CAPABILITIES, IRP: 0xFFFFCC0CE8924010
05465625	27359.58593750	usbip_vhci: vhci_pnp: Enter
05465626	27359.58593750	usbip_vhci: vhub: minor: IRP_MN_QUERY_CAPABILITIES, IRP:0xFFFFCC0CE7889B10
05465627	27359.58593750	usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS
05465628	27359.58593750	usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS
05465629	27359.58789063	usbip_vhci: vhci_pnp: Enter
05465630	27359.58789063	usbip_vhci: vhub: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP:0xFFFFCC0CE7CCCD30
05465631	27359.58789063	usbip_vhci:  QueryDeviceRelation Type: BusRelations
05465632	27359.58789063	usbip_vhci: # of vpdo's: present: 1, reported: 0
05465633	27359.58789063	usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS
05465634	27359.58789063	usbip_vhci: vhci_pnp: Enter
05465635	27359.58789063	usbip_vhci: vpdo: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFCC0CE94421D0
05465636	27359.58789063	usbip_vhci: QueryDeviceRelation Type: RemovalRelations
05465637	27359.58789063	usbip_vhci: vhci_pnp: Leave: STATUS_NOT_SUPPORTED
05465638	27359.58789063	usbip_vhci: vhci_pnp: Enter
05465639	27359.58789063	usbip_vhci: vpdo: minor: IRP_MN_QUERY_DEVICE_RELATIONS, IRP: 0xFFFFCC0CE94421D0
05465640	27359.58789063	usbip_vhci: QueryDeviceRelation Type: EjectionRelations
05465641	27359.58789063	usbip_vhci: vhci_pnp: Leave: STATUS_NOT_SUPPORTED
05465642	27359.58789063	usbip_vhci: vhci_pnp: Enter
05465643	27359.58789063	usbip_vhci: vpdo: minor: IRP_MN_SURPRISE_REMOVAL, IRP: 0xFFFFCC0CE94421D0
05465644	27359.58789063	usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS
05465645	27359.58984375	usbip_vhci: vhci_pnp: Enter
05465646	27359.58984375	usbip_vhci: vpdo: minor: IRP_MN_REMOVE_DEVICE, IRP: 0xFFFFCC0CE94421D0
05465647	27359.58984375	usbip_vhci: Deleting vpdo: 0xFFFFCC0CE75B67E0
05465648	27359.58984375	usbip_vhci: vhci_pnp: Leave: STATUS_SUCCESS

mankitty, 2020-02-21:
> It seems even one device can also cause a BSOD,debugview logs:
>
> URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER
> 05465602 27356.53710938 usbip_vhci: store_urbr: urbr: [seq:42]
> 05465603 27356.53710938 [...]

Hello, I am also troubled by the same problem, I wonder if you solved it?

cezanne, 2020-03-30:
@lfld287 : Would you try again with a up-to-date version?

cezanne, 2020-04-07:
@lfld287 : Please reopen if you want.

closed 2020-04-07 by cezanne