cezanne/usbip-win #79: Is there any way to use the driver without keeping test signing on?

rzvc, 2019-11-23:
The moment you disable testsigning, you can no longer use the driver.

Is there a way to fix this?

koudis, 2019-12-07:
No without valid sign.
Driver must be signed and trusted by the system.

You can buy EV Code Sign certificate and sign driver with that cert.

MartinDrab, 2020-02-09:
> You can buy EV Code Sign certificate and sign driver with that cert.

If you do not mind turning off Secure Boot, no EV certificate is necessary, just standard code signing one (cheaper, easier to obtain).

koudis, 2020-02-11:
@MartinDrab ou really? I did known that.

So if I understand correctly:

- If the Secure Boost is disabled the normal Code Sign cert is enought even for drivers.
- If the Secure Boost is enabled I need EV Coide Sign cert.

Correct?

cezanne, 2020-03-14:
I'll close this stale discussion. Please reopen if it is required.

closed 2020-03-14 by cezanne