cezanne/usbip-win #82: Attaching from win10 client does not work

pavhofman, 2019-12-05:
Hi, please can you attach from win10 64bit?
Command:
usbip.exe attach -r 192.168.105.144 -b 1-1.3

stalls, does not return. Deivce (usb soundcard) appears in windows sound devices. When trying to communicate with an audio application the usbip returns with:

usbip: error: read_devbuf: failed to read: err: 0x57

The same usbipd server works flawless from linux client.

Please can I provide some logs to troubleshoot? Thanks a lot,

Pavel.

koudis, 2019-12-07:
If I understand correctly you have USBIP server on windows machine?

pavhofman, 2019-12-07:
USBIP server is on linux. Another USBIP linux client works perfect with that soundcard (E-MU 0404 USB). USBIP Win10 client exhibits that error. I used the latest pre-compiled release files (0.07 with usbip_vhci.inf from release 0.05 - the latest one which includes the inf file). Unfortunately I have no windows compilation tools available.

Thanks a lot!

koudis, 2019-12-10:
USBIP blocks. On windows It does not return (due to implementation constraints).
Which version of Linux kernel do you use for USBIP server?
Did you try to compile USBIP from master? (Version 0.5.0 is really old)

pavhofman, 2019-12-10:
The server was RPi4 with kernel 5.3 https://github.com/raspberrypi/linux/tree/rpi-5.3.y . On windows 10 I used the pre-compiled release 0.0.7 from this repo https://github.com/cezanne/usbip-win/releases/tag/v0.0.7 . Because the release does not include usbip_vhci.inf, I used usbip_vhci.inf from release 0.0.5 which does include it.

Thank you for help.

koudis, 2019-12-13:
Sorry for late answer.

I'm affraind that i am not able to answer your question. I do not known why the USBIP-win does not work in that case.
Maybe @cezuni help?

koudis, 2019-12-13:
I have an idea:

- disconnect your usb device from server
- connect your device back to server and bind it to usbip by bind command
- on windows client call usbip.exe list -r <ip>
- check that upsip.exe show right names, device IDs and interfaces of your usb device
- if not try attach your usbb device by attach command
- even if attach fail run usbip.exe list -r <ip> and check if the device description provided by usbip-win changed

If usbip-win description of the device (from list command) chenged - it can be a problem with device driver update/install (usbip-win does not correctly update HW info etc...).

cezanne, 2020-03-30:
@pavhofman : Could you retry with [0.0.8](https://github.com/cezanne/usbip-win/releases/tag/v0.0.8)?

pavhofman, 2020-03-30:
Thank you, I will gladly test. Please what about the usbip_vhci.inf file, I would like to use the pre-compiled files from the 0.0.8 package. Thanks.

cezanne, 2020-03-30:
@pavhofman : Oh, I had missed the usbip_vhci.inf. Just now added. Please check again the release page.

pavhofman, 2020-03-30:
Thanks a lot for immediate response, unfortunately I will not be so fast :-)

cezanne, 2020-06-06:
@pavhofman : Close this issue. Please, reopen if you need.

closed 2020-06-06 by cezanne