cezanne/usbip-win #85: Win10 attach doesn't work

MasterLivens, 2019-12-09:
I'm using WIN10 with version 0.0.5( driver install failure with higher version ), but I could not attach it to the system successfully.
[image]

koudis, 2019-12-13:
Hello,

usbip win blocks. It's a feaure :).

Can you try:

- disconnect your usb device from server
- connect your device back to server and bind it to usbip by bind command
- on windows client call usbip.exe list -r
- check that usbip.exe shows right names, device IDs and interfaces of your usb device
- if not try attach your usb device by attach command
- even if attach fail run usbip.exe list -r and check if the device description provided by usbip-win changed

If usbip-win description of the device (from list command) chenged - it can be a problem with device driver update/install (usbip-win does not correctly update HW info etc...).

xzcvczx, 2019-12-20:
you also might want to change the power settings to disable selectively suspend usb devices. i was having no luck with either of the devices i tried to attach until i changed this setting in win10

[image]

koudis, 2020-01-04:
@MasterLivens I made some changes in PR #90 . Maybe it helps.

cezanne, 2020-03-14:
@xzcvczx : 

This project has not treated USB power issue but it's worth to discuss.
  
> i was having no luck with either of the devices i tried to attach until i changed this setting in win10

Did you mean that you were able to attach the device after power setting had been changed?

xzcvczx, 2020-03-14:
> Did you mean that you were able to attach the device after power setting had been changed?

correct

cezanne, 2020-03-15:
@xzcvczx : Let me reproduce your case. Please, give your power setting when attach operation failed.

cezanne, 2020-04-07:
@xzcvczx : It would be todo. Please reopen if you want.

closed 2020-04-07 by cezanne