cezanne/usbip-win #87: USBIP Server dead lock problem

519790441, 2019-12-19:
Version :0.0.7 debug x64，Client and server are using windows.
USBIP server side bind a usb-key. And client attach to server.
The first attach operation is normal.  
Then client detach from server.

Starting with the second attach operation, the program has a 99 percent chance of deadlock at call_usbd.
`
	KeInitializeEvent(&event, NotificationEvent, FALSE);
	irp = IoBuildDeviceIoControlRequest(IOCTL_INTERNAL_USB_SUBMIT_URB, devstub->next_stack_dev, NULL, 0, NULL, 0, TRUE, &event, &io_status);
	if (irp == NULL) {
		DBGE(DBG_GENERAL, "IoBuildDeviceIoControlRequest failed\n");
		return STATUS_NO_MEMORY;
	}

	IoSetCompletionRoutine(irp, on_usbd_complete, &event, TRUE, TRUE, TRUE);

	irpstack = IoGetNextIrpStackLocation(irp);
	irpstack->Parameters.Others.Argument1 = purb;
	irpstack->Parameters.Others.Argument2 = NULL;

	status = IoCallDriver(devstub->next_stack_dev, irp);
	if (status == STATUS_PENDING) {
		KeWaitForSingleObject(&event, Executive, KernelMode, FALSE, NULL);// <--Dead lock at here
		status = io_status.Status;
	}
`
What should I do? I need to debug it.

519790441, 2019-12-30:
Bulk transfer via stub forwarding often leads to problems.

cezanne, 2020-03-17:
@519790441 : Does the same problem occur for devices other than  usb-key? And can I have information about USB-key?

519790441, 2020-03-26:
> @519790441 : Does the same problem occur for devices other than usb-key? And can I have information about USB-key?

This problem occurs only on this USBKey device. Currently I have modified the code and managed to reset the device state when unbind-> bind, which solved the problem
The device is similar to the U-Shield of China Communications Bank

cezanne, 2020-03-30:
@519790441 : I have tested but failed to get a similar result of your report. Both client and server worked even with repetitive attach & detach operations.

> Currently I have modified the code and managed to reset the device state when unbind-> bind, which solved the problem

Can you share your modified codes?

cezanne, 2020-06-06:
@519790441 : Close the issue. Please reopen if you need.

closed 2020-06-06 by cezanne