cezanne/usbip-win #89: problems with composite equipment ?

mankitty, 2019-12-31:
Problems occur when my device (composite device) performs the connection. The following is my environment and logs. I do n’t know if anyone has the same problem as me. Thank you.

1. server
os:ubuntu 16.04
kernel:4.4.98
2. client
windows10

log in windows
00006850	53.93912506	usbip_vhci: store_urbr_submit: urbr: [seq:461], func:URB_FUNCTION_CLASS_INTERFACE	
00006851	53.93912506	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00006852	53.93922043	usbip_vhci: vhci_read: Enter	
00006853	53.93922424	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00006854	53.94559479	usbip_vhci: vhci_write: Enter	
00006855	53.94563675	usbip_vhci: process_urb_res: urbr:[seq:461], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB	
00006856	53.94564438	usbip_vhci: store_urb_data: Function code:27	
00006857	53.94569016	usbip_vhci: vhci_internal_ioctl: Enter	
00006858	53.94569778	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00006859	53.94570160	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CLASS_INTERFACE	
00006860	53.94571686	usbip_vhci: store_urbr: urbr: [seq:462]	
00006861	53.94572067	usbip_vhci: store_urbr_submit: urbr: [seq:462]	
00006862	53.94572830	usbip_vhci: store_urbr_submit: urbr: [seq:462], func:URB_FUNCTION_CLASS_INTERFACE	
00006863	53.94573975	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00006864	53.94575500	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	
00006865	53.94585419	usbip_vhci: vhci_read: Enter	
00006866	53.94586182	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00006867	53.95553970	usbip_vhci: vhci_write: Enter	
00006868	53.95555115	usbip_vhci: process_urb_res: urbr:[seq:462], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB	
00006869	53.95555878	usbip_vhci: store_urb_data: Function code:27	
00006870	53.95557785	usbip_vhci: vhci_internal_ioctl: Enter	
00006871	53.95557785	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	
00006872	53.95558548	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00006873	53.95558929	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CLASS_INTERFACE	
00006874	53.95559692	usbip_vhci: store_urbr: urbr: [seq:463]	
00006875	53.95560455	usbip_vhci: store_urbr_submit: urbr: [seq:463]	
00006876	53.95561600	usbip_vhci: store_urbr_submit: urbr: [seq:463], func:URB_FUNCTION_CLASS_INTERFACE	
00006877	53.95562363	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00006878	53.95573044	usbip_vhci: vhci_read: Enter	
00006879	53.95573807	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00006880	53.96352768	usbip_vhci: vhci_write: Enter	
00006881	53.96353912	usbip_vhci: process_urb_res: urbr:[seq:463], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB	
00006882	53.96354675	usbip_vhci: store_urb_data: Function code:27	
00006883	53.96355820	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	
00006884	53.96356583	usbip_vhci: vhci_internal_ioctl: Enter	
00006885	53.96356583	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00006886	53.96356964	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CLASS_INTERFACE	
00006887	53.96357727	usbip_vhci: store_urbr: urbr: [seq:464]	
00006888	53.96357727	usbip_vhci: store_urbr_submit: urbr: [seq:464]	
00006889	53.96357727	usbip_vhci: store_urbr_submit: urbr: [seq:464], func:URB_FUNCTION_CLASS_INTERFACE	
00006890	53.96358490	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00006891	53.96367264	usbip_vhci: vhci_read: Enter	
00006892	53.96368790	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00006893	53.97518539	usbip_vhci: vhci_write: Enter	
00006894	53.97520828	usbip_vhci: process_urb_res: urbr:[seq:464], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB	
00006895	53.97521973	usbip_vhci: store_urb_data: Function code:27	
00006896	53.97522736	usbip_vhci: vhci_write: Leave: STATUS_SUCCESS	
00006897	53.97524643	usbip_vhci: vhci_internal_ioctl: Enter	
00006898	53.97524643	usbip_vhci: ioctl code: IOCTL_INTERNAL_USB_SUBMIT_URB	
00006899	53.97525024	usbip_vhci: process_irp_urb_req: function: URB_FUNCTION_CLASS_INTERFACE	
00006900	53.97525787	usbip_vhci: store_urbr: urbr: [seq:465]	
00006901	53.97525787	usbip_vhci: store_urbr_submit: urbr: [seq:465]	
00006902	53.97526550	usbip_vhci: store_urbr_submit: urbr: [seq:465], func:URB_FUNCTION_CLASS_INTERFACE	
00006903	53.97526550	usbip_vhci: vhci_internal_ioctl: Leave: STATUS_PENDING	
00006904	53.97539902	usbip_vhci: vhci_read: Enter	
00006905	53.97540665	usbip_vhci: store_urbr_partial: urbr: [seq:465]	
00006906	53.97541809	usbip_vhci: vhci_read: Leave: STATUS_SUCCESS	
00006907	53.97551346	usbip_vhci: vhci_read: Enter	
00006908	53.97552109	usbip_vhci: vhci_read: Leave: STATUS_PENDING	
00006909	53.98094177	usbip_vhci: vhci_write: Enter	
00006910	53.98095322	usbip_vhci: process_urb_res: urbr:[seq:465], ioctl:IOCTL_INTERNAL_USB_SUBMIT_URB	
[... 207 lines cut ...]
Dec 31 16:53:34 dky kernel: [ 8542.179806] usbip_core: usbip_recv:372: receiving....
Dec 31 16:53:34 dky kernel: [ 8542.179811] usbip-core00000000: 01000000 cd010000 07000100 01000000
Dec 31 16:53:34 dky kernel: [ 8542.179815] usbip-core00000010: 07000000 00000000 10000000 00000000
Dec 31 16:53:34 dky kernel: [ 8542.179820] usbip-core00000020: 00000000 08000000 00000000 00000000
Dec 31 16:53:34 dky kernel: [ 8542.179824] usbip_core: usbip_recv:375: received, osize 48 ret 48 size 0 total 48
Dec 31 16:53:34 dky kernel: [ 8542.179829] usbip_core: usbip_dump_header:289: BASE: cmd 1 seq 461 devid 65543 dir 1 ep 7
Dec 31 16:53:34 dky kernel: [ 8542.179833] usbip_core: usbip_dump_header:298: USBIP_CMD_SUBMIT: x_flags 0 x_len 16 sf 0 #p 0 iv 8
Dec 31 16:53:34 dky kernel: [ 8542.179837] wangcan number_of_packets:0
Dec 31 16:53:34 dky kernel: [ 8542.179842] usbip_host: stub_recv_cmd_submit:516: submit urb ok, seqnum 461
Dec 31 16:53:34 dky kernel: [ 8542.179845] usbip_host: stub_recv_cmd_submit:529: Leave
Dec 31 16:53:34 dky kernel: [ 8542.179848] usbip_host: stub_rx_pdu:540: Enter
Dec 31 16:53:34 dky kernel: [ 8542.179851] usbip_core: usbip_recv:336: enter
Dec 31 16:53:34 dky kernel: [ 8542.179874] usbip_core: usbip_recv:368: stub_rx   :
Dec 31 16:53:34 dky kernel: [ 8542.179876] usbip_core: usbip_recv:372: receiving....
Dec 31 16:53:34 dky kernel: [ 8542.179881] usbip-core00000000: 01000000 ce010000 07000100 01000000
Dec 31 16:53:34 dky kernel: [ 8542.179886] usbip-core00000010: 07000000 00000000 10000000 00000000
Dec 31 16:53:34 dky kernel: [ 8542.179890] usbip-core00000020: 00000000 08000000 00000000 00000000
Dec 31 16:53:34 dky kernel: [ 8542.179894] usbip_core: usbip_recv:375: received, osize 48 ret 48 size 0 total 48
Dec 31 16:53:34 dky kernel: [ 8542.179898] usbip_core: usbip_dump_header:289: BASE: cmd 1 seq 462 devid 65543 dir 1 ep 7
Dec 31 16:53:34 dky kernel: [ 8542.179902] usbip_core: usbip_dump_header:298: USBIP_CMD_SUBMIT: x_flags 0 x_len 16 sf 0 #p 0 iv 8
Dec 31 16:53:34 dky kernel: [ 8542.179906] wangcan number_of_packets:0
Dec 31 16:53:34 dky kernel: [ 8542.179910] usbip_host: stub_recv_cmd_submit:516: submit urb ok, seqnum 462
Dec 31 16:53:34 dky kernel: [ 8542.179913] usbip_host: stub_recv_cmd_submit:529: Leave
Dec 31 16:53:34 dky kernel: [ 8542.179916] usbip_host: stub_rx_pdu:540: Enter
Dec 31 16:53:34 dky kernel: [ 8542.179919] usbip_core: usbip_recv:336: enter

koudis, 2019-12-31:
Hi,

I have exactly same problem. I work on bugfix as described in #74.

mankitty, 2020-01-02:
> Hi,
> 
> I have exactly same problem. I work on bugfix as described in #74.
> 
> > I am working on bugfix in usbip.exe and with composite devices. It seems that ABORT_PIPE, RESET_PIPE_AND_CLEAR_STALL and GET_DESCRIPTOR_FROM_INTERFACE is not implemented correctly. (it gets STALL_PID response sometimes)
> > Once the configuration of given USB device is not issued for first time the autorepair feature of USB is not able to repair failures due to non consistent implementation of URBs listed above.

Hi,
I'm really sorry, I just saw your message. excuse me, did you solve the problem I described? I looked at 74, but I don't seem to find any useful information. Could you please help me describe the process of solving the problem in detail? thank you

koudis, 2020-01-04:
@mankitty I will create PR soon.

I will be glad if you test my PR once it's created.

koudis, 2020-01-04:
Hello :),

@mankitty can you test my PR with your equipment?

 Thanks a lot :)

mankitty, 2020-01-05:
> Hello :),
> 
> @mankitty can you test my PR #90 with your equipment?
> 
> Thanks a lot :)

I just took a brief look at your code and it looks like it should solve my problem, but I have to test it on Monday, thank you very much

mankitty, 2020-01-06:
> Hello :),
> 
> @mankitty can you test my PR #90 with your equipment?
> 
> Thanks a lot :)

Thank you very much, this makes one of my devices available, but the other device (Jabra SPEAK 410 USB, speaker + microphone) still does not work, the program has been printing usbip_vhci: QueryBusTime called crazy and sometimes enter the vhci_ioctl_abort_pipe module, don't know what is your good idea? thank you very much

koudis, 2020-01-06:
Can you attach Win logs and screenshot of malfunction device in device Manager? Does the Jabra work fine under usbip linux client?

Is that error consistent? (you get always same result in DeviceManager?)

mankitty, 2020-01-07:
> Can you attach Win logs and Device Manager screenshot of malfunction device? Does the Jabra work fine under usbip linux client?
> 
> Is that error consistent? (you get always same result in DeviceManager?)
Hi,
The following is the kernel log under Windows and Linux. The attachment is a screenshot of the device manager mapped to Windows by usbip, thank you very much

[image]

Linux kenel log:
Jan  7 18:01:29 dky kernel: [  834.085074] usbip_core: usbip_dump_header:298: USBIP_CMD_SUBMIT: x_flags 1 x_len 0 sf 0 #p 0 iv 0
Jan  7 18:01:29 dky kernel: [  834.085081] usbip_host: tweak_set_interface_cmd:128: set_interface: inf 1 alt 0
Jan  7 18:01:29 dky kernel: [  834.087848] usbip-host 1-2: usb_set_interface done: inf 1 alt 0
Jan  7 18:01:29 dky kernel: [  834.087856] wangcan number_of_packets:0
Jan  7 18:01:29 dky kernel: [  834.087873] usbip_host: stub_recv_cmd_submit:516: submit urb ok, seqnum 104
Jan  7 18:01:29 dky kernel: [  834.087877] usbip_host: stub_recv_cmd_submit:529: Leave
Jan  7 18:01:29 dky kernel: [  834.087881] usbip_host: stub_rx_pdu:540: Enter
Jan  7 18:01:29 dky kernel: [  834.087885] usbip_core: usbip_recv:336: enter
Jan  7 18:01:29 dky kernel: [  834.089558] usbip_host: stub_complete:74: complete! status 0
Jan  7 18:01:29 dky kernel: [  834.089594] usbip_host: stub_send_ret_submit:198: setup txdata seqnum: 104 urb: ffff8800bb620540

Jan  7 18:01:58 dky kernel: [  862.485357] usbip_core: usbip_recv:368: stub_rx   :
Jan  7 18:01:58 dky kernel: [  862.485365] usbip_core: usbip_recv:372: receiving....
Jan  7 18:01:58 dky kernel: [  862.485377] usbip-core00000000: 01000000 69000000 03000100 00000000
Jan  7 18:01:58 dky kernel: [  862.485384] usbip-core00000010: 00000000 01000000 00000000 00000000
Jan  7 18:01:58 dky kernel: [  862.485389] usbip-core00000020: 00000000 00000000 00010b01 00000001
Jan  7 18:01:58 dky kernel: [  862.485395] usbip_core: usbip_recv:375: received, osize 48 ret 48 size 0 total 48
Jan  7 18:01:58 dky kernel: [  862.485402] usbip_core: usbip_dump_header:289: BASE: cmd 1 seq 105 devid 65539 dir 0 ep 0
Jan  7 18:01:58 dky kernel: [  862.485408] usbip_core: usbip_dump_header:298: USBIP_CMD_SUBMIT: x_flags 1 x_len 0 sf 0 #p 0 iv 0
Jan  7 18:01:58 dky kernel: [  862.485418] usbip_host: tweak_set_interface_cmd:128: set_interface: inf 1 alt 1
Jan  7 18:01:58 dky kernel: [  862.486902] usbip-host 1-2: usb_set_interface done: inf 1 alt 1
Jan  7 18:01:58 dky kernel: [  862.486911] wangcan number_of_packets:0
Jan  7 18:01:58 dky kernel: [  862.486927] usbip_host: stub_recv_cmd_submit:516: submit urb ok, seqnum 105
Jan  7 18:01:58 dky kernel: [  862.486932] usbip_host: stub_recv_cmd_submit:529: Leave
Jan  7 18:01:58 dky kernel: [  862.486936] usbip_host: stub_rx_pdu:540: Enter
Jan  7 18:01:58 dky kernel: [  862.486941] usbip_core: usbip_recv:336: enter
Jan  7 18:01:58 dky kernel: [  862.488093] usbip_host: stub_complete:74: complete! status 0
Jan  7 18:01:58 dky kernel: [  862.488136] usbip_host: stub_send_ret_submit:198: setup txdata seqnum: 105 urb: ffff8800bb620480
Jan  7 18:01:58 dky kernel: [  862.491460] usbip_core: usbip_recv:368: stub_rx   :
Jan  7 18:01:58 dky kernel: [  862.491468] usbip_core: usbip_recv:372: receiving....
Jan  7 18:01:58 dky kernel: [  862.491480] usbip-core00000000: 01000000 6a000000 03000100 00000000
Jan  7 18:01:58 dky kernel: [  862.491487] usbip-core00000010: 00000000 00000000 03000000 00000000
Jan  7 18:01:58 dky kernel: [  862.491493] usbip-core00000020: 00000000 00000000 01000122 00030003
Jan  7 18:01:58 dky kernel: [  862.491499] usbip_core: usbip_recv:375: received, osize 48 ret 48 size 0 total 48
Jan  7 18:01:58 dky kernel: [  862.491506] usbip_core: usbip_dump_header:289: BASE: cmd 1 seq 106 devid 65539 dir 0 ep 0
Jan  7 18:01:58 dky kernel: [  862.491512] usbip_core: usbip_dump_header:298: USBIP_CMD_SUBMIT: x_flags 0 x_len 3 sf 0 #p 0 iv 0
Jan  7 18:01:58 dky kernel: [  862.491520] usbip_core: usbip_recv:336: enter
Jan  7 18:01:58 dky kernel: [  862.491527] usbip_core: usbip_recv:368: stub_rx   :
Jan  7 18:01:58 dky kernel: [  862.491531] usbip_core: usbip_recv:372: receiving....
Jan  7 18:01:58 dky kernel: [  862.491535] usbip-core00000000: 80 bb 00
Jan  7 18:01:58 dky kernel: [  862.491540] usbip_core: usbip_recv:375: received, osize 3 ret 3 size 0 total 3
Jan  7 18:01:58 dky kernel: [  862.491545] usbip_host: tweak_special_requests:209: no need to tweak
Jan  7 18:01:58 dky kernel: [  862.491548] wangcan number_of_packets:0
Jan  7 18:01:58 dky kernel: [  862.491563] usbip_host: stub_recv_cmd_submit:516: submit urb ok, seqnum 106
Jan  7 18:01:58 dky kernel: [  862.491566] usbip_host: stub_recv_cmd_submit:529: Leave
Jan  7 18:01:58 dky kernel: [  862.491570] usbip_host: stub_rx_pdu:540: Enter
Jan  7 18:01:58 dky kernel: [  862.491574] usbip_core: usbip_recv:336: enter
Jan  7 18:01:58 dky kernel: [  862.494635] usbip_host: stub_complete:74: complete! status 0
Jan  7 18:01:58 dky kernel: [  862.494691] usbip_host: stub_send_ret_submit:198: setup txdata seqnum: 106 urb: ffff8800bb6200c0
Jan  7 18:01:58 dky kernel: [  862.822349] usbip_core: usbip_recv:368: stub_rx   :
Jan  7 18:01:58 dky kernel: [  862.822356] usbip_core: usbip_recv:372: receiving....
Jan  7 18:01:58 dky kernel: [  862.822366] usbip-core00000000: 01000000 6b000000 03000100 00000000
Jan  7 18:01:58 dky kernel: [  862.822372] usbip-core00000010: 03000000 02000000 80070000 00000000
[... cut ...]
00000192	17.46266937	usbip_vhci: vhci_read: Enter	
00000193	17.46267319	usbip_vhci: process_read_irp: Enter	
00000194	17.46268082	usbip_vhci: store_urbr_partial: urbr: [seq:107]	
00000195	17.46268845	usbip_vhci: store_urbr_partial: status: 0	
00000196	17.46269226	usbip_vhci: vhci_read: Leave: STATUS_SUCCESS	
00000197	17.46273041	usbip_vhci: QueryBusTime called	
00000198	17.46273804	usbip_vhci: QueryBusTime called	
00000199	17.46274948	usbip_vhci: QueryBusTime called	
00000200	17.46275520	usbip_vhci: QueryBusTime called
.
.
.

koudis, 2020-01-07:
In USBIP-Win there is a function called "QueryBusTime".

According to [`documentation`](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/usbbusif/nc-usbbusif-pusb_busiffn_query_bus_time)

It must return current frame number. Parameters of the function is currently unused. Maybe try to ser frame number to 0?

mankitty, 2020-01-07:
> In USBIP-Win there is a function called "QueryBusTime".
>
> According to [...]

No, the parameters of this function have already been used, the following code, where the function is used and where it is defined, this function is called crazy, I do n’t know how to do it

-------------------------------------------------------------------------------------------------------------
Code here：

in function vhci_QueryInterface_vpdo

.
.
.
switch (version) {
	case USB_BUSIF_USBDI_VERSION_1:
		bus_intf->IsDeviceHighSpeed = IsDeviceHighSpeed;
		/* passthrough */
	case USB_BUSIF_USBDI_VERSION_0:
		bus_intf->QueryBusInformation = QueryBusInformation;
		bus_intf->SubmitIsoOutUrb = SubmitIsoOutUrb;
		bus_intf->QueryBusTime = QueryBusTime;
		bus_intf->GetUSBDIVersion = GetUSBDIVersion;
		bus_intf->InterfaceReference   = InterfaceReference;
		bus_intf->InterfaceDereference = InterfaceDereference;
		bus_intf->BusContext = vpdo;
		break;
	default:
		DBGE(DBG_GENERAL, "never go here\n");
		return STATUS_INVALID_PARAMETER;
	}
.
.
.

static NTSTATUS USB_BUSIFFN
QueryBusTime(IN PVOID context, IN OUT PULONG currentusbframe)
{
	UNREFERENCED_PARAMETER(context);
	UNREFERENCED_PARAMETER(currentusbframe);

	DBGI(DBG_GENERAL, "QueryBusTime called\n");
	return STATUS_UNSUCCESSFUL;
}

koudis, 2020-01-07:
Try to apply and test this patch pls [test_patch.txt](https://github.com/cezanne/usbip-win/files/4030247/test_patch.txt).

koudis, 2020-01-07:
Try to apply and test this patch pls [test_patch.txt](https://github.com/cezanne/usbip-win/files/4030254/test_patch.txt)

koudis, 2020-01-07:
and if it does not work try this please :) [test_patch.txt](https://github.com/cezanne/usbip-win/files/4030298/test_patch.txt).

mankitty, 2020-01-08:
> and if it does not work try this please :)
> [test_patch.txt](https://github.com/cezanne/usbip-win/files/4030300/test_patch.txt)

Thank you for your answer. However, when I open the music, I still print the information as shown in the picture crazy, and the music software cannot be used normally.

[image]

koudis, 2020-01-08:
@mankitty Sorry but I do not know where is problem.
I do not have any USB sound device. I will try borrow from someone but it takes a time.

(QueryBusTime called multiple times have not to be a problem. Maybe we need to return proper value from QueryBusTime...)

mankitty, 2020-01-08:
> @mankitty Sorry but I do not know where is problem.
> I do not have any USB sound device. I will try borrow from someone but it takes a time.
> 
> (QueryBusTime called multiple times have not to be a problem. Maybe we need to return proper value from QueryBusTime...)

Thank you for helping me so much, I will do my best to solve this problem by myself, thank you very much

mankitty, 2020-01-10:
> and if it does not work try this please :)
> [test_patch.txt](https://github.com/cezanne/usbip-win/files/4030300/test_patch.txt)

[debug_pdu.log](https://github.com/cezanne/usbip-win/files/4044467/debug_pdu.log)
Hello, sorry to bother you, this problem still makes me baffled, debug_pdu.log is my log file, I hope you can take a look when you have time, thank you very much

koudis, 2020-01-24:
@mankitty sorry for late answer.

I think that audio devices needs to obtain proper timing from USB Bus.

Wee need to update QueryBusTIme function to return proper bus time/frame.

mankitty, 2020-01-26:
> @mankitty sorry for late answer.
> 
> I think that audio devices needs to obtain proper timing from USB Bus.
> 
> We need to update QueryBusTIme function to return proper bus time/frame.

I think so, and I am working in this direction, thank you very much.

mankitty, 2020-02-20:
> > @mankitty sorry for late answer.
> > I think that audio devices needs to obtain proper timing from USB Bus.
> > We need to update QueryBusTIme function to return proper bus time/frame.
> 
> I think so, and I am working in this direction, thank you very much.

Sorry to disturb you again.
However, I still haven't been able to solve this problem. Among them, I improved the QueryBusTime function by referring to the https://github.com/OpenXT/xc-vusb.git project, but there was no printing other than crazy printing QueryBusTime called , Including upper-layer applications and vhci drivers.

The final print under linux is as follows

Feb 20 11:24:43 dky kernel: [ 6370.748089] usbip_core: usbip_recv:375: received, osize 160 ret 160 size 0 total 160
Feb 20 11:24:43 dky kernel: [ 6370.748109] usbip_host: stub_recv_cmd_submit:518: submit urb ok, seqnum 36
Feb 20 11:24:43 dky kernel: [ 6370.748113] usbip_host: stub_recv_cmd_submit:531: Leave
Feb 20 11:24:43 dky kernel: [ 6370.748117] usbip_host: stub_rx_pdu:542: Enter
Feb 20 11:24:43 dky kernel: [ 6370.748120] usbip_core: usbip_recv:336: enter
Feb 20 11:24:43 dky kernel: [ 6370.751134] usbip_host: stub_complete:74: complete! status 0
Feb 20 11:24:43 dky kernel: [ 6370.751168] usbip_host: stub_send_ret_submit:198: setup txdata seqnum: 35 urb: ffff88012a477600
Feb 20 11:24:43 dky kernel: [ 6370.761205] usbip_host: stub_complete:74: complete! status 0
Feb 20 11:24:43 dky kernel: [ 6370.761216] usbip_host: stub_send_ret_submit:198: setup txdata seqnum: 36 urb: ffff8800bb6cbc00

As for windows
QueryBusTime** called Printing is crazy.

I really have no other way, I look forward to your free time to look at it, thank you very much.

mankitty, 2020-02-29:
nothing

closed 2020-02-29 by mankitty

koudis, 2020-02-29:
Sorry for late answer.
I do not known :(. This is out of my scope.

qiuliankui, 2021-03-05:
@mankitty 请问这个问题最后怎么解决的，USB IP支持声音设备输出声音吗？