cezanne/usbip-win #91: Use it to share a USB device between a Windows 7 host and a Windows 7 guest VM

Zero3K, 2020-01-05:
How do I do that?

Zero3K, 2020-01-07:
Maybe you all could make a wiki that includes how to do that.

MegaZeroX10, 2020-03-06:
I was able to get it built for a x64 platform of Windows 7 but not for a x86 platform of it. Please check https://github.com/cezanne/usbip-win/issues/95 for more info.

cezanne, 2020-03-06:
@MegaZeroX10 : Well, indeed, x86 platform is in a low priority. But @cantops in an old issue(#17) reported successful deployment in x86.

MegaZeroX10, 2020-03-06:
I don't know how he managed to get it compiled.

cezanne, 2020-03-13:
@MegaZeroX10 : I just uploaded a new branch to build usbip-win against x86 platform. Please check [build_x86 branch](https://github.com/cezanne/usbip-win/tree/build_x86). I confirmed compilation but have not tested because there's no x86 machine. Can you check ? I attached x86 binaries( [usbip-win-x86.zip](https://github.com/cezanne/usbip-win/files/4329597/usbip-win-x86.zip) )

Dimazhan, 2020-03-27:
An error occurs while installing the vhci driver:

> The driver for this device could not be loaded. The driver may be corrupted or missing. (Code 39)

Stub driver installed without errors, after editing architecture in inf file

cezanne, 2020-03-30:
@Dimazhan : Thanks for your response. Well, it is fortunate that the stub works. :cry: 

x86 support is not a major concern. I'll just merge x86 build patch into the master branch.

closed 2020-04-02 by cezanne