cezanne/usbip-win #95: Errors when compiling a release version of it for 32-bit Windows 7

Zero3K, 2020-01-06:
I am getting the errors in the attached text file when compiling a release version of it for 32-bit Windows 7 using Visual Studio 2019 Community Edition:

[usbip-win-windows7-x86-release-compilation-errors.txt](https://github.com/cezanne/usbip-win/files/4024394/usbip-win-windows7-x86-release-compilation-errors.txt)

koudis, 2020-01-07:
Sorry, we does not support Windows 7. Only windows 10 x64.

Zero3K, 2020-01-07:
I was able to compile a x64 build for Windows 7 and it works with no issues.

koudis, 2020-01-08:
Nice, can you close this issue please? :)

Zero3K, 2020-01-08:
No, another person has managed to get it compiled for a 32-bit OS.

koudis, 2020-01-09:
Windows 7 is not supported any more. We do not support Windows 7/10 x86 too.

If you want to add support for windows 10 x86, create PR please :)..

Zero3K, 2020-01-09:
Why is that?

Zero3K, 2020-01-10:
According to a guy on ReactOS' Mattermost Server, its not able to find the header files.

cezanne, 2020-03-30:
@Zero3K : x86 build support has been added. Please check 2f82656d80d78f4d186b6bf25be108551c13a67e.

closed 2020-03-30 by cezanne