cezanne/usbip-win #96: Old usbip project does not work in the project

windowsair, 2020-01-31:
Hi! I have a project that works fine under the old version of usbip windows, but it doesn't work under the current project.
The current project can handle requests such as `OP_REQ_DEVLIST`, `OP_REQ_IMPORT`, but there seems to be a problem with processing `USBIP_CMD_SUBMIT`.
Specifically, when processing such as `GET CONFIGURATION DESCRIPTOR` or `GET DEVICE DESCRIPTOR`, it throws the following error:
```bash
usbip: debug: D:\Desktop\USB\usbip-win-master\userspace\src\usbip\usbip_attach.c:57:[import_device] got free port 1
usbip: error: unknown command in pdu header: 117768450
usbip: error: unknown command in pdu header: 301994497
```
The following figure is the specific transmission process:
[image]

It can be seen that usbip parsed URB data incorrectly.

The above process works normally in older versions of usbip. Is this the different protocols reason?Hope to get some instructions from you. Thank you!

koudis, 2020-01-31:
Hello,

can you try usbip from my branch?
https://github.com/cezanne/usbip-win/pull/90

It solves some problems about descriptors and others.

windowsair, 2020-01-31:
> Hello,
> 
> can you try usbip from my branch (#90)?
> 
> It solves some problems about descriptors and others.

Thank you for your reply! I have tried your branch, but the problem remains.

koudis, 2020-01-31:
That numbers you reported - It seems that there can be problem with network-system byte order.
Which USBIP server do you use? Windows or Linux? Which version of Linux kernel (in case of Linux server)?

windowsair, 2020-01-31:
`unknown command in pdu header: 117768450` 
you should be able to see that the data here is actually in URB Data.
Is there annetwork-system byte order problem in urb?

cezanne, 2020-03-05:
@windowsair : Which version did you tested? You might use different versions of usbip_vhci.sys and usbip.exe.

windowsair, 2020-03-08:
> @windowsair：您测试了哪个版本？您可能使用了不同版本的usbip_vhci.sys和usbip.exe。

0.0.7. The driver installation experience is terrible, I am not sure if it is a problem

windowsair, 2020-03-08:
> @windowsair : Which version did you tested? You might use different versions of usbip_vhci.sys and usbip.exe.

0.0.7 The driver installation experience is terrible, I am not sure if it is a problem.

cezanne, 2020-03-14:
@windowsair : It's weird.. Can you share logs of a forwarder routine in usbip.exe? Getting forwarder logs are explained in [README.md](https://github.com/cezanne/usbip-win#how-to-get-usbip-forwarder-log)

cezanne, 2020-04-02:
@windowsair : I'm going to close. Reopen if you want.

windowsair, 2020-05-01:
> @windowsair : It's weird.. Can you share logs of a forwarder routine in usbip.exe? Getting forwarder logs are explained in [README.md](https://github.com/cezanne/usbip-win#how-to-get-usbip-forwarder-log)

The following is the log content, I am not sure whether it contains useful information 😥:
```
DUMP: CMD_SUBMIT,seq:1,devid:10001,dir:in,ep:0
  flags:0,len:12,sf:0,#p:0,intv:0
  setup: 8006000100001200
DUMP DONE-------
```

cezanne, 2020-05-01:
@windowsair : Ok, your forwarder log is valid. However, is that all logs? If your forwarder log has only 1 entry, a function driver did not invoke any usbip request. Did you install a function driver? What is your USB device ? Please retry an up-to-date version (v0.1.0).

windowsair, 2020-05-01:
@cezanne yes, now i am testing under v0.1.0. After waiting for a while, i received a UNLINK log, and there are no extra logs. The device is a USB HID and no additional driver is required.

cezanne, 2020-05-01:
@windowsair : No way.. USB HID is the best working device.. 😢

windowsair, 2020-05-02:
@cezanne Ok... i will show you the code to reproduce this issue asap

windowsair, 2020-05-03:
I found that this behavior is different from the old version. When the old version usbip processed Standard USB request `Get Configuration Descriptor`, it would judge the length according to `transfer_buffer_length`  in the packet header and send the corresponding data.
The pseudo code is as follows:
```c
if(header->u.cmd_submit.transfer_buffer_length == sizeof(usb_config_descriptor)){
    send_header(header, header->u.cmd_submit.transfer_buffer_length);
    send_data(kUSBConfigDescriptor, sizeof(kUSBd0ConfigDescriptor));
}
else{
    send_header(header, header->u.cmd_submit.data_length);
    send(kUSBConfigDescriptor, sizeof(kUSBd0ConfigDescriptor));
    send(kUSBInterfaceDescriptor, sizeof(kUSBd0InterfaceDescriptor));
}
```

In the new version, the value of `transfer_buffer_length` seems uncertain (i received a very large value). In the Response packet, `transfer_buffer_length` is returned as it is in the Request Packet, which causes an error.

cezanne, 2020-05-03:
@windowsair :
> I found that this behavior is different from the old version. When the old version usbip processed

The old version means 0.0.7?

> The pseudo code is as follows:

Which S/W does your pseudo code represent?

windowsair, 2020-05-03:
The old version does not refer to your repo. It is an earlier version of usbip-win.

This code is used to communicate with usbip to simulate a usb device.

windowsair, 2020-05-03:
Now, i found that  `usbip_wudev.c` throw a exception:
```
Run-Time Check Failure #2 - Stack around the variable 'buf' was corrupted.
```

cezanne, 2020-05-03:
@windowsair : usbip-win is not compatible with the old version. I recommend you to use the same version of usbip.exe and a windows kernel(vhci) driver in usbip-win.

windowsair, 2020-05-03:
Yes, as you said, we can use `usbip.exe` and` usbipd.exe` to share USB devices between different hosts. But sometimes I also want to simulate a USB device (my own program communicates with usbip.exe via socket), so I want to find the difference in communication between this and the old version of the project.

For the exception mentioned above (seems to be an array overflow in usbip?). Do you have any ideas？Thank you!

windowsair, 2020-05-03:
@cezanne There seems to be a problem with `buf` in` supplement_with_interface`. I just increased the size of `buf`, and now the program works perfectly, it is the same as the old version of usbip.😄

windowsair, 2020-05-03:
~~There seems to be a problem with `buf` in` supplement_with_interface`. I just increased the size of buf~~, and now the program works perfectly, it is the same as the old version of usbip.😄

In fact, `buf [18]` contains Config Descriptor (9 bytes) + Interface Descriptor (9 bytes)
`Interface Descriptor` only contains 9 bytes described by Standard Interface Descriptor, excluding HID Descriptor and Endpoint Descriptor

cezanne, 2020-05-04:
@windowsair : Good job. I'm going to close this issue.

closed 2020-05-04 by cezanne